MarlinFirmware/Marlin #9720: Filament change M125, M600, M602, M701, M702

mylife4aiurr, 2018-02-19:
Normally I use filament sensor (endstop) attached to pi3 running Octoprint. So I had all the move commands and logic handled by Octoprint.
filament reloaded pluggin -
Out of filament GCODE:
M600 X-84 Y-48 Z50 ; Delta printer, moves head up and out of the way

Before print job is resumed:
M602 ;

So I would never hard code a park position in the firmware. Until now. I needed to do a filament change, from lcd of printer. So I should change?
#define NOZZLE_PARK_POINT { (X_MIN_POS + 10), (Y_MAX_POS - 10), 20 }
to
#define NOZZLE_PARK_POINT { X-84, Y-48, Z50 }

Question 1:
If the lcd menu is used to initiate the filament change, what will happened if the nozzle is close to Z max. This is delta printer, max height 296. So if the head is at Z280 for instance. I have the park point defined { X-84, Y-48, Z50 }, the Z50 is relative from height of model I believe, but will it respect the max height of 296. Will it move the x and y and some z or will it ignor endstops because 280+50 = 330?

Question 2:
gcode initiated filament change. I've used M600 (M600 X-84 Y-48 Z50). Same question - will it respect the Z max if printer is already close to Zmax. I ask because I see M125 has option:
* The "P" parameter controls the action applied to the Z axis:
 *
 *    P0  (Default) If Z is below park Z raise the nozzle.
 *    P1  Raise the nozzle always to Z-park height.
 *    P2  Raise the nozzle by Z-park amount, limited to Z_MAX_POS.  <<<-------

Should I be using M125 P2, to limit Z movement to Z_MAX_POS?? or does also M600 respect Zmax?? 

park position already defined so should I use something like this in a gcode initiated filament change :
M125 P2 L20 ; 
M600 ;

or is 

M600 ; is enough??

Question 3:
#define ADVANCED_PAUSE_FEATURE
#define FILAMENT_LOAD_UNLOAD_GCODES       // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.

M602, M701, M702 - Have no documention @ http://marlinfw.org/meta/gcode/

AnHardt, 2018-02-19:
Sometimes i wonder about people investing 5 minutes for writing 3 questions, requesting some hundred reader to invest a minute for reading, instead of testing themselves in 3 minutes.

Of cause the lift will work, as long the software endstops are on. If not report back.

mylife4aiurr, 2018-02-19:
@AnHardt when I posted a question about skew correction - u acted like the answer was obvious. Turns out the code needed to be re-written to show more decimal places to user:
 0.0 to user means off. however 0.0034232 means its doing something.

Now I ask another question you say "try your self and see and report back" 

These questions help other users wondering same thing. Only stupid question is the one not asked, why are you always so belittling. I'm not asking questions to troll you.

AnHardt, 2018-02-19:
Sorry for hitting you again.

thinkyhead, 2018-02-25:
@AnHardt — I have learned to take a deep breath and count to 10. Especially this past 2 weeks going through 900 issues. Certain users are definitely trying to waste our time. I'm sure we both know their names. They waste the time of maintainers on other projects too.

Anyways… as far as this issue goes, yes, it is a bit **tl;dr** and I'm only here to code right now and get bugs fixed. We're actually going to start putting the kibosh on these kinds of support questions and refer users to the RepRap forums, Facebook, and other support-oriented resources so that we can focus on our core job here.

That said, of course we want to hear about bugs and things that are broken so we can fix them. And we hope we'll get more time and help on documentation too, so that we can refer users to that. Questions like these tell us what we need to explain better on the Marlin.org website.

mylife4aiurr, 2018-03-19:
U can close this. I have figured out it myself @AnHardt

My problem with Octoprint filament sensor reloaded pluggin is being addressed by the author. (It caches/logs sensor triggers during filament changes which will trigger another filament after the current filament change -> get stuck in a loop) But I still wanted to know about respecting endtops for filament run-outs/changes near xyx max.

Any way the consecutive triggering of filament change/run-outs with the pluggin  forced me to learn how to wire the endstop directly to the ramps board, with an added capacitor as researched thru reading previous support questions. I also read 1.1.9 will account for fail positives b4 triggering the m600 command. I look forward to trying it.

closed 2018-03-19 by mylife4aiurr