MarlinFirmware/Marlin #9723: 1.1.x-bugfix New UBL_Z_RAISE_WHEN_OFF_MESH option problems

gloomyandy, 2018-02-19:
Hi, I've been experimenting with the UBL code and as part of that recently updated (last night) to 1.1.x-bugfix. As part of that I enabled the new option UBL_Z_RAISE_WHEN_OFF_MESH to try and handle any off mesh moves. However I noticed that with this option enabled virtually all points on the mesh end up being classed as off mesh and so return the "default value" which I have set to 2.5. I had a quick look at the code in ubl.h which currently looks like:
`          if (!WITHIN(rx0, 0, GRID_MAX_POINTS_X - 1) || !WITHIN(ry0, 0, GRID_MAX_POINTS_Y - 1))
            return UBL_Z_RAISE_WHEN_OFF_MESH;
`

I do not think this is correct as rx0 and ry0 are print co-ordinates not an index into the mesh array. Instead I think the following test should be used:
` 
          if (!WITHIN(rx0, MESH_MIN_X, MESH_MAX_X) || !WITHIN(ry0, MESH_MIN_Y, MESH_MAX_Y))
            return UBL_Z_RAISE_WHEN_OFF_MESH;
`
Possibly a +1/-1 needs to be applied to the limits but I'm not 100% sure about that. But with the above change applied the code seems to work as intended.

Let me know if you need any further information.

gloomyandy, 2018-02-20:
I should probably have supplied a little more information. The problem mentioned above is with the function "get_z_correction(const float &rx0, const float &ry0)". I saw the problem when testing the G29 J command, which uses this function to compare the corrected value with the probed result. When I used it I ended up with a crazy mesh, it turned out that this problem was causing part of this issue. I'm not sure how widely this function is used.

Roxy-3D, 2018-02-20:
Yikes!   I have to run out the door.   But I'll dig in on this when I get back home.

Roxy-3D, 2018-02-20:
There is also a spelling error in that function:

           return UBL_Z_RAISE_WHEN_OFF_MESHH;

I'll get things patched as soon as I can!!!!

THANK YOU FOR FINDING THIS!!!!

gloomyandy, 2018-02-20:
No problem, not sure if my current fix is the best especially unsure if the range needs +/-1 adding to it to clamp things safely. Let me know if you need any more information.

Roxy-3D, 2018-02-20:
I'm thinking it would be the least burden on the CPU to change
```
if (!WITHIN(rx0, 0, GRID_MAX_POINTS_X - 1) || !WITHIN(ry0, 0, GRID_MAX_POINTS_Y - 1)) return UBL_Z_RAISE_WHEN_OFF_MESH;
```
to
```
if (!WITHIN(rx0, X_MIN_POS, X_MAX_POS) || !WITHIN(ry0, Y_MIN_POS, Y_MAX_POS)) return UBL_Z_RAISE_WHEN_OFF_MESH;
```

If you are going to run with modified code...   It would be very helpful to have that change checked out and confirmed as working.

gloomyandy, 2018-02-20:
Doesn't it need to check the position against the mesh, rather then the full range of the machine? The change I gave above to:
`if (!WITHIN(rx0, MESH_MIN_X, MESH_MAX_X) || !WITHIN(ry0, MESH_MIN_Y, MESH_MAX_Y)) return UBL_Z_RAISE_WHEN_OFF_MESH;`
has been working fine for me (I had previously had a "Z axis rises to sky" incident when running G26 but with the above code I've not had another. My system has a MESH_INSET of 10, so in my case the mesh is smaller than the possible range. I think that MESH_MIN_X etc. should be resolved at compile time so using them should not be any more expensive.

Roxy-3D, 2018-02-20:
You are correct!!!!

Roxy-3D, 2018-02-21:
@gloomyandy   I moved your suggestions to bugfix-2.0.0 and bugfix-1.1.x

Thank You!!!

closed 2018-02-23 by thinkyhead