MarlinFirmware/Marlin #9724: Problem with Z_PROBE_OFFSET_FROM_EXTRUDER

robertbba, 2018-02-19:
Hi, does somebody have issue with Z_PROBE_OFFSET_FROM_EXTRUDER?
I'm trying to set up my printer Z probe but this setting doesn't respond. Can anyone help me with this ???

hern4ndes, 2018-02-24:
whats happen?

thinkyhead, 2018-02-24:
If EEPROM is enabled, you'll need to use `M502` followed by `M500` to make sure your new Configuration settings are applied.

dosono, 2018-02-25:
It does not seemed to be stable. Sometimes you can just save the new offset and sometimes you need to reset with M502. The problem is that if you already fine tuned your UBL mesh, re configuring it is really annoying. If I could download the UBL map for a backup and load it later on after a hard reset M502, that could be much less of a problem. Any Idea?

thinkyhead, 2018-02-25:
@dosono Unfortunately UBL has not yet fully implemented its `M503`. If it had, you could simply copy the G-code output from that command and play it back to re-create your mesh. To work around this, for now use `M420 V` and copy the output. With some clever text editing you could turn it into G-code to restore the mesh values.

thinkyhead, 2018-02-25:
@dosono — For UBL the command to get the grid in a form you can restore later is: `G29 S-1`. See http://marlinfw.org/docs/gcode/G029-ubl.html

closed 2018-02-25 by thinkyhead

dosono, 2018-02-25:
Good idea. The question is why the M851 is not stable and you need to reset the EEPROM. 
Best would be just to be able to set your offset from the first place. Is it a bug ?

thinkyhead, 2018-02-25:
@dosono — Are you trolling us?

dosono, 2018-02-25:
How did you come to that sad idea :(  
I just became one of your top 5 supporters in Patreon last week.
I am currently straggling much with configuring the Z offset to my printer. 
Only slicer offset is working but Z_PROBE_OFFSET_FROM_EXTRUDER reconfigured with M851 is just not stable enough, sometimes it works and sometimes not and need a hard reset with M502.
I am really sorry if this post is disturbing anyone, I had the idea that this is the right place for placing it.

thinkyhead, 2018-02-25:
We are experiencing a large amount of troll activity right now. You chimed in on an issue not your own that may or may not be related, only adding noise. The developer team finds this kind of behavior very annoying, and we must repeatedly remind visitors that this is not a support forum, a community space, or a substitute for reading documentation and doing proper research.

This is our workplace.

I appreciate your patronage, but I'm not the only developer here and there's a reason we have procedures. They exist so that we can obtain the details necessary to help you, and to help us track down real issues.

You should create a new issue and fill in the requested details. This would help us to help you, and expedite the process. We can speculate all day, but data is the currency of the realm here.

Support forums exist for new users having difficulties with configuration and features. There are several Marlin and 3D-printer-oriented Facebook groups, a very active forum at http://forums.reprap.org, and loads of resources for getting set up and going which can be found by searching Google and YouTube. I urge you to take advantage of these more appropriate venues.

Apologies for being terse, but I have hundreds of things to go through and am trying to get a release out. Thanks for understanding, and I look forward to helping you if you cannot find your answers elsewhere.

All the best…