MarlinFirmware/Marlin #9725: Idea da aggiungere nel menu

philfire, 2018-02-19:
Il mio, non e un vero problema, ma uno spunto per una aggiunta nel menu, del marlin. L'idea, mi e venuta, quando ho sostituito i pololu 8825 con i tmc2100, che richiedono l'inversione dei motori. Perché, avendo la EPROM attiva, non inserire la possibilità di invertire la marcia e salvare l'impostazione in EPROM, direttamente da display? Sarebbe comodo.
 you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

GMagician, 2018-02-19:
"Visto che tutti scrivono in Inglese sarebbe una cortesia non usare la tua lingua"
As everybody write in English it should be a courtesy not to use your own language

In my opinion that will increase code size and slow down execution for something that has to be changed only once, when you prepare your machine.. I may be completly wrong though

thinkyhead, 2018-02-25:
@GMagician — Agreed.

Una sola parola: efficienza. Le costanti producono codice molto più veloce delle variabili.

closed 2018-02-25 by thinkyhead

GMagician, 2018-02-25:
@thinkyhead do you know Italian? wow you astonish me!
This seems not a google translation, too correct

thinkyhead, 2018-02-25:
I can't speak Italian to save my life. But I can tell a bad translation from a good one in most Romantic languages.