MarlinFirmware/Marlin #9727: Anycubic Kossel freezing up few minutes after activation.

jar0fair, 2018-02-19:
### Bug Report
- Description: Printer is freezing up and stops all fans and motion after first layer or two of printing. LCD screen is unresponsive, requires restart.
- Expected behaviour: Complete the print 
- Actual behaviour: print aborted, machine unresponsive.
- Steps to reproduce: “Print from SD”

dennisbun, 2018-02-23:
How much did you put into DELTA_SEGMENTS_PER_SECOND? Try using 100 or less, also try to unplug the usb while printing. Last but not least, check if your power supply is grounded properly.

thinkyhead, 2018-02-23:
emilio1625/Anycubic-Kossel-Marlin#4

thinkyhead, 2018-02-23:
Also: If you're using 32x micro-stepping, the MCU may not be able to keep up, causing a reboot. You may need to lower your max feedrates and/or `DELTA_SEGMENTS_PER_SECOND` as suggested by …… @dennisbun …….

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

jar0fair, 2018-02-23:
@dennisbun Delta segments have always been set at 100. The USB is not plugged in while printing. I am not sure how to tell if the PSU is grounded or not. But, it worked before...

@thinkyhead I do not see any options to adjust "micro-stepping" but it is probably at the default, as I have never adjusted that.

dennisbun, 2018-02-23:
To adjust to microstepping you can reposition the jumpers beneath the drivers. Try using lower delta segments too. Also try using older firmware, sometimes it works. If you can, measure the voltage from the power supply.

thinkyhead, 2018-02-24:
@jar0fair — Please ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop them into your next reply. We'll check them over and see if anything is amiss.

jar0fair, 2018-02-24:
@thinkyhead Sure thing. Thanks for taking a look at these. I am pretty new to this. I know enough to have got this working on the machine. But not enough to fix it when it broke :3 I have also asked the manufacturer for a replacement PSU, under suspicion that it may be the cause.
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/1754377/Configuration.files.zip)

thinkyhead, 2018-02-28:
@jar0fair — Be sure to take advantage of the user support forums at http://forums.reprap.org . Since we are highly focused on development and tending the next release, we don't have a lot of time to help new users to figure out things which are not due to bugs. That is what users-helping-users is all about. You should always go there first, and only when everyone is stumped and unable to help you and thinks it might be a bug should you come here.

I am personally too busy for the rest of the week to delve into this, but I can tell you that freezes are always weird and hard to track down. If Marlin is over-stressed, it doesn't freeze, it usually reboots.

Please follow the earlier advice and see where it leads. And please, please take advantage of the community to help you. We are a half-dozen volunteers here, and we have a lot on our plates.

atransell, 2018-07-09:
@jar0fair , i am new here i was just searching for why my anycubic kossel linear plus was randomly freezing,  few times on the same job. its a mission to get it to start where it crashed  //screen unresponsive(frozen), fans still going heat still on\\ i need to pull the plug basically.  my problem was a faulty SD card i tried a different one and it did not happen again. hope this helps somebody!

closed 2018-11-05 by thinkyhead