MarlinFirmware/Marlin #9728: 1.1.8 Aceleration and jerk not work in no-printing movement

fbarcenas, 2018-02-19:
------------------------------------------------------------
### Bug Report 1.1.8 version
- Description: When move manually Aceleration and jerk not work (in no-printing movement)
- Steps to reproduce:
  -  when you move 100mm 2-3times continuously the stepper sound wrong (like if have a big acceleration or jerk).

While is printing not problem happen, is only when you move manually.

In version 1.1.3 not fail, another version or bug fix today not tested

------------------------------------------------------------

Studiodyne, 2018-02-20:
Yes and the same , when you move on UBL mesh edit ,  all is brutality in movements

thinkyhead, 2018-02-20:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

fbarcenas, 2018-02-21:
All absolutly solved with actual Bugfix of today.

Talking in general terms, I think than the actual bug fix must be 1.1.9 immediately. 1.1.8 have fails than need solved urgently, and already is solved in the bug fix.

thinkyhead, 2018-02-23:
> the actual bug fix must be 1.1.9 immediately

1.1.9 will be released very soon.

fbarcenas, 2018-03-02:
I want reopen this problem, after do some test i have the follow conclusion.
- Actual today bug fix: no problem in movements of NO-printing.
                                      Yes problem in movements of printing
- 1.1.8: Problem in movements of NO-printing.
            No problem in movements of printing

Same machine, same config, same file.

Sebastianv650, 2018-03-02:
- Do you use any kind of bed leveling?
- Does this also happens when you disable bed leveling?
- If it's not bed leveling related, can you create a short gcode file which contains a set of G1 commands that reproduces the issue so I can test it?

fbarcenas, 2018-03-02:
- Not
- Always is disabled
- You can use the piece than you want, usually the fail is when do the "prime pillar" for clean the nozzle. a square of 25x25mm. Speed 50mm/s and movements in 45º X/Y

Sebastianv650, 2018-03-02:
I checked the planner only a short time ago,it was fine on my machine back then. I will recheck with latest rcbugfix tomorrow.

Please tell me your jerk values so i can see if there is something related to them in special. 

It's only a noise that leads to you conclusion of having a problem isn't it?

fbarcenas, 2018-03-02:
Acceleration: 500
Jerk: tested with 1.0 and 0.5

Sorry i don't understand your question, could you rewrite again? sorry for my english

Sebastianv650, 2018-03-02:
1.0 and 0.5 for xjerk, yjerk and ejerk?

My question was: you wrote you have problems with the moves, as it would crash (=a hard sound). Is this the only problem, or do you get also shifted layers for example?

fbarcenas, 2018-03-02:
Yes X and Y
Sound crash yes and also you can see like the layers also is shifted/moved (like around 3 or 4mm)

Sebastianv650, 2018-03-03:
I can't reproduce something like a stop with too high jerk. In fact, I'm seeing quite the oposite without having a real answer at the moment.
I wrote a test gcode file containing 3 loops around a 30x30mm square. First one without extrusion, second loop like a print move and third loop with retractions. With this, I set XY jerk to 5 and E to 1 first, then switching the values.
While the calculated max junction speed looks fine, the set initial and final step rates are nearly always MINIMAL_STEP_RATE (=120):

[image]

I have to dig a little bit deeper..

Sebastianv650, 2018-03-03:
@fbarcenas please add your Configuration.h file here for download, I have maybe fixed a problem but I guess there is also a problem in your configuration.

fbarcenas, 2018-03-03:
Well, i do new test... and this is horrible. I think than the best is i try to explain all from the first.

I do more than 30 prints with 1.1.8, i only have problem in no-printing movements. 
Never i have a problem with printing movement.

I upload bug fix version 28feb (and i check than the no-printing movements are fixed!) and i do a new print:
- In the first hour i can see two crash noises and two fails in the layers, 
  First fail: In axis Y+ move 3mm (aprox) the layer shifted and after 20minutes happen again.
I repeat the same print and fail again.... shifted layer in Y+ again...

So i upload the 1.1.8, and i start the same file gcode.
Now the fail happen!! but in the axis X-, the layer is shifted more than 100mm!!!

Problem of file? Problem of jerk? What happen?
I upload:
- Configuration file 1.1.8
- Configuration file bugfix 28feb
- File gcode
[Archivo comprimido.zip](https://github.com/MarlinFirmware/Marlin/files/1777844/Archivo.comprimido.zip)

Sebastianv650, 2018-03-03:
That's a big printer! I guess the mass of this printer might be the reason why a small jerk error count lead to such high shifts, but it's realy strange nevertheless. I'm aware of reports about hard movements with bed leveling enabled, but no one with this issue with disabled ABL.

Maybe #9914 will give us some knowledge, but in fact it's more in the oposite direction as I don't see any jerk without that PR in current bugfix branch.

thinkyhead, 2018-03-05:
With Sebastian's PR merged (and feeling extra confidence because it's more like the 1.1.5 planner code) please give the latest `bugfix-1.1.x` another go and see if we now have good movements in both LCD/manual and regular G1/printjob movement. I suspect it will be good.

fbarcenas, 2018-03-08:
I am sorry to say than the problem not is fixed.
With 1.1.8 i have shifted layers in X axis
With Bugfix 28Feb and 8 March i have shifted layers in Y axis.

Same file, same machine.

I don't know than say....

thinkyhead, 2018-03-09:
Try setting `MINIMUM_STEPPER_PULSE` to 2 and see if it helps.

fbarcenas, 2018-03-10:
I test with bug fix today than include this, but fail...
I see and the fail is always in the same points...
When do a curve non-hard, is like apply wrong acceleration or jerk value and the stepper don't do acceleration....And i can see like must do a big movement without acceleration.
I attach a example of gcode.

In this, fail a lot, is continuously. In the inner perimeter especially
[test.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/1799733/test.gcode.zip)

If i use a weird value of jerk like 0.20 seem than work...

Some ideas?

fbarcenas, 2018-03-10:
Is like: instead of do a curve, do a lot of little lines without acceleration...

thinkyhead, 2018-03-11:
That G-code looks pretty screwy. You should provide a simpler example that's not composed of small, disconnected segments and way off-center.

fbarcenas, 2018-03-11:
I attach new gcode, more easy and simple.
[testeasy.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/1800470/testeasy.gcode.zip)

is ok now?

thinkyhead, 2018-03-11:
The segments in that G-code are quite small; as small as 0.1mm for long runs, which is problematic. Try modifying your slicer settings or altering the model so that the facets are a bit longer (at least 0.5mm), or slow down your printing speed so that the planner won't starve.

fbarcenas, 2018-03-13:
Answer of support software slicer:
_"It sounds like what you are attempting is a reduction in the complexity of the mesh to minimize the amount of small moves. This can be accomplished using the mesh reduction tool found in the Mesh menu in the top toolbar but it is surprising that movements of the above mentioned .1mm length are causing issues. I would also please keep in mind that .5mm is a fairly large distance for curved objects and will result in faceting of the part."_

thinkyhead, 2018-03-14:
I seriously doubt a segment length of 0.5mm will result in any obvious faceting.

Anyway, we still hope you will try the things we suggest to see if it makes any difference. Your testing helps us to narrow down the issue.

VanessaE, 2018-04-09:
I seem to be having this issue also (at commit de5f69b2852e0e2fadd02f3c5a8d19f41ab2194e)

A typical print move is fine, but travels frequently sound like they're being slammed up to speed without acceleration, which causes lots of stalls/skipped steps.  Speeds are usually 60 mm/s coarse features, 30 mm/s on fine details, and 200 mm/s travels, but even 240 mm/s print moves don't stall -- only travel moves are affected.

My X/Y acceleration is at 2000 mm/s², jerk is at 10.  The hardware's mechanical limit is around 350 mm/s, 6000 mm/s², and somewhere north of 40 for jerk.

Sebastianv650, 2018-04-09:
Maybe you can isolate one occurrence in the gcode? It might help to set print and travel acceleration to a very low number like 200. This way it makes nice sounds during acceleration and deceleration so it's easy to hear when an acceleration is missing. A sample gcode section with the travel move +-2 lines or so would be helpful.

Sebastianv650, 2018-04-09:
> It seems to be pretty random, but I have a hunch that it's related to the bit of LIN_ADVANCE that intentionally reduces X/Y acceleration when needed.

This could only affect print moves and it's also only able to reduce accelerations. I'm more on the side to think it's another hickup of the jerk calculation (#10341).

It can take one day or two until I can do the test print, but I will do it if necessary.

VanessaE, 2018-04-09:
That (10431) sounds plausible, yeah.  I'll leave related comments on that issue in a bit.

As for that test print, there's no big rush; dropping to 1500 mm/s² is a usable workaround (besides, when print speeds are down in the 30/60 range, increasing acceleration gives diminishing returns).

boelle, 2019-02-20:
@fbarcenas still having issues with latest bugfix 2.0?

boelle, 2019-03-17:
@fbarcenas i think your issue is that you try to print faster than the hardware can

i had similar issues and when i lowered speeds it started to work

closed 2019-05-16 by thinkyhead