MarlinFirmware/Marlin #9730: bed maxtemp triggered, but hardware ok.

mamama1, 2018-02-20:
Marlin 1.1.8

AM8, standard A8 hotbed, MKS gen 1.4 board (atmega2560).

many features and buffers enabled, 85% RAM used, arduino warning about high memory usage and possible instabilities triggered.

can the bed maxtemp error be triggered by an error due to high mem usage of 85%?

the hardware is ok, bed temp was at 75 degrees, maxtemp is at 85 degrees.

also: maxtemp is set to 85 degrees but i cant set more than 65 through the display. through gcode and octoprint i can set higher temps. this issue persisted already when various features weren't enabled and therefore low memory condition didn't exist.

thinkyhead, 2018-02-20:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

mamama1, 2018-02-20:
thanks but if the current release is not going to be supported any further, an announcement of the next stable release would be really great. it feels like every issue gets the same answer, "use the bugfix release".

AnHardt, 2018-02-20:
@mamama1 
`bugfix-1.1.x` **is** the branch we support (fix bugs) the current release in (, plus doing some innovations).

mamama1, 2018-02-20:
so bugfix-1.1.x is considered stable/more stable than the latest release?

GMagician, 2018-02-20:
> so bugfix-1.1.x is considered stable/more stable than the latest release? always?
as stated by @AnHardt  this is bugfix and innovation. From bugfix side, yes it stabilizes bugs, from innovation side it may add some new bugs

AnHardt, 2018-02-20:
Ouer 'releases' are stable in the sense of 'they will not be changed'. That does not say anything about having more or less errors. Usually bugfix is the preferable branch 3-4 days after a release. Usually the release gets a bit more attention/downloads when they are fresh. Then lots of issues come in. For some fixing is simple and it's done quite fast. This could be the ideal time to download a 'bugfix' branch to get a firmware with relative few bugs. After that most of the developers return to the funny side of programming - developing and improving features.

mamama1, 2018-02-20:
i'd like to contribute before criticisibg but unfortunately my c coding skills are not good enough for this. but maybe this can be taken as constructive criticism anyway: why not concentrate solely on bugfixing in the bugfix branch and only introduce new features in some development or nightly build branch? new features often bring new bugs (also outside of the new features) and so the "user" can potentially be left with the choice between two buggy versions. either the old final release with a known fixed bug in the bugfix branch or the bugfix branch with a newly introduced, showstopping bug...

thinkyhead, 2018-02-25:
Thanks. We are on the same page.

batata004, 2018-02-26:
Guys I dont know why are you arguing about the bufix and such. To me it's clear: there is a bug. And the bug surelly is related to something very wrong.

I watched today the hotend temperature and it is very strange, it updates every 1s and sometimes it goes from 190 to 215, then 230 to 210... all in 1 sec. Also I already opened another thread reporting this same error the OP reported here, errors like MAX TEMP and such.

Finally I AM PRETTTY SURE there is something overloading arduino cause sometimes when I turn the knob it freezes for like 1 or 2 seconds and then the menu scrolls very fast to a random option. Maybe the  RAM is being used too much (I already saw other threads of people reporting too much RAM and some "smart" people said it's not a big probem having too much ram being used except if it's close to 100%).

Anyway, now with this thread opened by @mamama1 it's clear to me that something is wrong with this version. I was working fine with last version and had nos issues like this!

mamama1, 2018-02-26:
the arguing was because many issues were commented with „please try latest bugfix version“, which basically leaves the current version of the stable branch kind of unsupported...

anyways, regarding to ram usage, i too observed weird behaviour when hitting about 80% of ram usage. according to my knowledge you can use up almost any ram if you are not doing things like malloc without checking free ram first. also ram will get fragmented when using malloc which will lead to more issues because it may happen that at one point in time malloc wont find a free block in the memory which is large enough and unfragmented to accomodate the needs of the malloc call and then everythig may go boom...

batata004, 2018-02-26:
@mamama1 when I said what I said I was totally agreeing with you! 

> I dont know why are you arguing about the bufix and such

Exactly that. The bug happens with the latest stable version, so it should be fixed in that version, not in the bugfix or any other version. Makes no sense asking people to test the bug with the bugfix branch or the 2015 branch.

I am not a C programmer and I cant help with that, but what I am experience is random issues which I never had before so many times.

thinkyhead, 2018-02-27:
@batata004 You are only adding noise. Please step away.

[image]

AnHardt, 2018-02-27:
@mamama1 
> also: maxtemp is set to 85 degrees but i cant set more than 65 through the display. through gcode and octoprint i can set higher temps.

The limit is 15°C not 20.
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/ultralcd.cpp#L1303

This is to avoid maxtemp errors caused by overshot and (not yet) implemented for other ways to set the temperature.
Maxtemp should be set at least 15°C above the temperature you want to print with. It's mainly thought as a protection against shorted thermistor wires. Protection of the nozzle is a side effect.

mamama1, 2018-02-27:
that makes sense, thanks!

mylife4aiurr, 2018-06-19:
bugfix 1.1.x 78410b2
Reporting same strange behavior. Thermal run away reported, printer halted. 
but seems as though reporting the temp is being delayed but to cpu overload??? idk. 

Gonna try latest build and also double check wiring, if problems I'll report back.

boelle, 2019-02-20:
@mamama1 still having issues with latest bugfix 2.0?

closed 2019-05-16 by thinkyhead