MarlinFirmware/Marlin #9732: [1.1.7+] [REGRESSION] home offsets broken when homing to max

Lenbok, 2018-02-20:
### Bug Report
[copied from comments in #9689, since this looks like a different bug]
#### Description: 
I have a Mendel90, which homes Z to MAX (Z_HOME_DIR = 1). Since the distance between the Z max endstop and the bed is not static (e.g. varies according to films applied to bed, length of extruder nozzle for the current extruder), we need to use a home offset to adjust Z_MAX_POS.  I suspect there's a bad interaction between a Z offset and homing to max Z as the Mendel90 does -- when you home to max Z, setting a Z offset should result in a change in the total height, rather than a simple shift in coordinate. The resulting behaviour is quite screwy, see below...

In 1.1.6 and prior, M206 Znnn would correctly adjust the position that the machine reports immediately after it has homed. in 1.1.7+, the position reported on the LCD differs from that reported by M114, and the Z axis moves in a buggy manner.

#### Steps exhibiting the problem:
I recompiled (release 1.1.8), with my Z_MAX_POS set to 180.
M206 is initially set to Z0. Homing and manually moving the z axis works as it should. 
Now I manually jogged to ~8mm above my bed so the nozzle touched a piece of 8mm rod (actually 7.96mm). The LCD said ~3.1mm, so I issued M206 Z4.86. However M114 says Z is now at 7.96 (which it is) but the LCD hasn't updated.  Why doesn't the LCD display the correct position?

Now I homed all axes, which homes Z to max as it should, however, after homing the LCD Z says it is at 180 (it should say 184.86, which we have now determined is the real distance from the bed to the Z MAX endstop). The printer serial output does think it is at 184.86 though:
````
Send: G28
Recv: X:100.00 Y:100.00 Z:184.86 E:0.00 Count X:8000 Y:8000 Z:576000
````

Now I jog down 100mm in octoprint and check the coordinates:
````
Send: G91
Send: G1 Z-100 F200
Send: G90
Send: M114
Recv: X:100.00 Y:100.00 Z:80.00 E:0.00 Count X:8000 Y:8000 Z:240448
````
(LCD reports Z is 75.14). What?  At this point Z should be 84.86 on both M114 and LCD. Something is screwy.

Now I jog Z *up* by 1mm - however the machine physically moves *down*!:
````
Send: G91
Send: G1 Z1 F200
Send: G90
Send: M114
Recv: X:100.00 Y:100.00 Z:76.14 E:0.00 Count X:8000 Y:8000 Z:228096
````
(LCD reports Z is 71.28).

1) LCD and m114 coordinates should be in sync
2) M206 home offsets should work correctly with machines homing to MAX.
3) Jogging should move the axis in the direction you tell it!!!

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1739070/configuration.zip)

thinkyhead, 2018-02-20:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

Lenbok, 2018-02-20:
I just tested on bugfix-1.1.x (as at 35800ee). Jogging now works in the correct direction. 

This just leaves the LCD being out of sync with M114 and not showing the correct position (i.e. when the nozzle is actually at the bed, the LCD reads Z-4.85).

thinkyhead, 2018-02-23:
See if this change to `Marlin.h` helps:
```diff
- #if HAS_POSITION_SHIFT || DISABLED(DELTA)
+ #if HAS_WORKSPACE_OFFSET
    #define LOGICAL_X_POSITION(POS)   NATIVE_TO_LOGICAL(POS, X_AXIS)
    #define LOGICAL_Y_POSITION(POS)   NATIVE_TO_LOGICAL(POS, Y_AXIS)
    #define RAW_X_POSITION(POS)       LOGICAL_TO_NATIVE(POS, X_AXIS)
    #define RAW_Y_POSITION(POS)       LOGICAL_TO_NATIVE(POS, Y_AXIS)
  #else
    #define LOGICAL_X_POSITION(POS)   (POS)
    #define LOGICAL_Y_POSITION(POS)   (POS)
    #define RAW_X_POSITION(POS)       (POS)
    #define RAW_Y_POSITION(POS)       (POS)
  #endif
```

thinkyhead, 2018-02-23:
Thanks again for the report. This is now patched!

closed 2018-02-23 by thinkyhead

Lenbok, 2018-02-23:
@thinkyhead Umm, nope, it isn't fixed! I just checked out 9c65890 (just after your patch), and M114 still disagrees with what is displayed on the LCD! Please reopen.

thinkyhead, 2018-02-23:
@Lenbok That's too old. Get the latest.

thinkyhead, 2018-02-23:
Whoops. You didn't catch my typo? Hold on…

Lenbok, 2018-02-23:
Sweet, all sorted now, thanks.

foosel, 2018-02-23:
Is it really though? I just ran into something similar here with a printer homing to minimum and 1.1.8. 

After a `M206 Z2` a *relative* jog of the Z axis by +0.1mm no longer works, instead it moves down 1.9mm. It looks like the reason for that is [this bit](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L3263-L3264) which applies the offset even to relative moves (which unless I understand the whole workspace offset concept completely wrong it definitely shouldn't). I'm just about to test a patch I prepared for this behaviour but maybe I'm barking up the wrong tree here...

foosel, 2018-02-23:
Ah, never mind, I was checking the wrong branch, the issue indeed is fixed ;)

thinkyhead, 2018-02-24:
No problem. We are also knives-out for bugs right now!

commit 2c977531 (2018-02-23): Fix Z position on HD44780

Fix #9732
PR #9770 (2018-02-23): [1.1.x] Fix Z value on HD44780
Fix #9732

Oops! Forgot to apply the `LOGICAL_Z_POSITION` macro to the HD44780 coordinate display.
commit 4ea05826 (2018-02-23): Fix #9732 again