MarlinFirmware/Marlin #9734: [1.1.x/2.0] UBL z_fade_height menu ' easy and for comfort'

Studiodyne, 2018-02-20:
@Roxy-3D  @thinkyhead 

FOR LCD USER , we have three problems ,  
-Need gcode/host just to enable fade height 😢 
-The z_fade_height value never visible
-Impossible to change by lcd

Suggestion
-Add the Z_Fade_height lcd menu ' in tune and in prepare '  , to set it during a print 👍 
With this , we see the value everytime, we initialise every time it's needed , and possibility to change when we want !
Thks

Roxy-3D, 2018-02-20:
This is not that hard to do.   And in fact, the code is already written and in the manual mesh leveling section.   It needs to be cleanly extracted and made available to all the bed leveling systems (that support it).

Studiodyne, 2018-02-20:
I have already made it for myself.  
Just tell here because will be made and merged immediately .

Roxy-3D, 2018-02-20:
OK!   I'll check out and merge the Pull Requests when they show up.

Studiodyne, 2018-02-20:
Thks 😎

closed 2018-02-24 by thinkyhead