MarlinFirmware/Marlin #9736: Missing  jerk steps and other setting in lcd menu rc bugfix downloaded yesterday and today

bogaskilop, 2018-02-20:
Hi  yesterday i tried to compile new fw rcbugfix  for tmc2130 and all gos smooth and printer is working  but 
in LCD menu there are missing settings like steps  jerks accelleration   i got only Z offset in this menu tree 

i downloaded rcbugfix today compiled and still the same situation.

only this option is available in Motion  menu 
Z Offset | M851 Z | HAS_BED_PROBE (with BABYSTEP_ZPROBE_OFFSET it babysteps)

missing : 

Feedrate » | Feedrate settings |  
Acceleration » | Acceleration settings |  
Jerk » | Jerk settings |  
Steps/mm » | Steps/mm for XYZ axes and extruders

#

Sebastianv650, 2018-02-20:
Do you accidently enabled #define SLIM_LCD_MENUS in Configuration_adv.h?

bogaskilop, 2018-02-20:
yes i always used that option in older FW versions

bogaskilop, 2018-02-20:
Ok thx   this solve the problem

closed 2018-02-20 by bogaskilop