MarlinFirmware/Marlin #9742: [Concept] TMC Sensorless Homing in M120/M121?

thinkyhead, 2018-02-21:
The `M120` and `M121` commands don't currently support sensorless homing. Should s.h. be enabled and disabled by these commands too?

I would extend `M120` and `M121` to enable/disable of sensorless homing per-axis, while only the real endstops would be enabled/disabled using `M120`/`M121` with no parameters.

The main use-case for `M120`/`M121` is to enable endstops for the sake of the SD-abort-on-endstop-hit option. Outside of that it can be troublesome. If `M120` is activated and a `G1` move hits an endstop the current position becomes "unknown." Since we don't try to correct the position based on the steps count, this makes `M120` slightly dangerous. It is feasible to recover the position from the stepper counts (and we could add another G-code to do that).

With sensorless homing it also becomes feasible to determine the position for any bump that stops the axis. But where and how to apply such a correction is not straightforward, especially for a print in progress. A position recovery re-home (as on Prusa MK3) makes the most sense.

teemuatlut, 2018-02-21:
The current implementation only enables the DIAG pin when homing and as far as Marlin knows, it's just a signal on a pin. So at the moment sensorless homing will not work with any abort on endstop hit feature.
But as we move to developing the skipped step detection and recovery system, we can no longer disable the DIAG pin outside of homing and so it makes sense to have gcodes that can enable/disable triggering.

thinkyhead, 2018-02-23:
I modified `quick_home_xy` to use sensorless homing too. I hope I got that right!

boelle, 2019-10-24:
@thinkyhead still need this one?

boelle, 2019-11-24:
this one has gone stale too and debate could move to discord

Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-11-24 by boelle