MarlinFirmware/Marlin #9747: Servo settings not required with new BLTouch settings?

Careless, 2018-02-21:
Was just editing the latest bugfix as of tonight, and see that there is a #define BL_TOUCH... is it treated separately from the #define NUM_SERVOS, #define SERVO_DELAY, #define SERVO_ANGLES and whatnot (I am not sure if I got them all correct, but they are servo related). 

Can I just ignore the servo settings completely, and enabling the BLTouch will make the required changes to accommodate the sensor?

GMagician, 2018-02-21:
It depends on your machine configuration.
For sure angles are never used. BLTOUCH use its own parameters to work but other parameters are related to your configuration.
By default firmware assign servo 0 and SERVO_DELAY = 50 if you don't have other servo you don't need anything else.
If you just have another servo then NUM_SERVOS and SERVO_DELAY must be initialized correctly (2 values and usually first is 50 if servo 0 is still BLTOUCH)

Careless, 2018-02-21:
The printer does not have any other servos. So I should not need to mess around with SERVO settings if all I am doing is enabling BL Touch sensor, correct? I can leave NUM_SERVOS 0 and SERVO_DELAY at default and not worry about them?

I am asking because the old BL Touch recommended settings for Marlin 1.0.7 that is posted on their website makes use of those settings in regard to BL Touch, and not in regard to other servos.

GMagician, 2018-02-21:
Yes, with one servo system, if SERVO_DELAY=50 has no issues (and it shouldn't), you need to enable only BLTOUCH in configuration.h

Careless, 2018-02-21:
Ok, thanks for clearing that up for me.

GMagician, 2018-02-21:
Don't remember default value for NUM_SERVOS... if commented by // no problem otherwise it has to be forced to 1

thinkyhead, 2018-02-23:
> Can I just ignore the servo settings completely,

Yes.

closed 2018-02-23 by thinkyhead