MarlinFirmware/Marlin #9748: [FR]  Additional motor moving in the background independently

JLatasa, 2018-02-21:
I need to control a motor independently (not syncronized) so that I can run other comands while it is running. 
Is it possible to allow the rest of the commands to load normally from the buffer while a stepper is moving in the background?

AnHardt, 2018-02-21:
No. Not currently and not in the near future.

thinkyhead, 2018-02-23:
I concur with @AnHardt — This would not be simple. Basically you would have to hack the stepper ISR to add this support, in part emulating the technique that `LIN_ADVANCE` uses to make E semi-independent of the other axes. Or, you would need to add a whole new ISR to control the extra stepper, and this would steal too much time from the main stepper ISR.

If the extra stepper doesn't need to move at a high rate of speed, you could work from a crude Arduino sketch for controlling steppers and hack it into the temperature ISR, where `BABYSTEPPING` is managed. But this may not move smoothly or continuously because that interrupt is slow and low priority.

AnHardt, 2018-02-23:
Additionally the 'g-code-language' does not allow asynchronous moves. G0/G1 can't handle that. It needs another G??? to handle that additional asynchronous axis.

boelle, 2019-02-20:
@thinkyhead maybe add the feature request label? assuming that @JLatasa still have interest in this

boelle, 2019-06-22:
missing label

sjasonsmith, 2020-11-10:
This appears to have been more of a support question "can Marlin do this" rather than a fully formed feature request. Given the apparent lack of interest and compatibility with existing Marlin architectures I will go ahead and close it out.

closed 2020-11-10 by sjasonsmith