MarlinFirmware/Marlin #9749: 1.1.x (All versions) Z homing gone crazy

charles2690, 2018-02-21:
I will start by letting you know that I am using an inductive sensor on the Z axis. Using M119, it register open and triggered normally when I use a metallic object to check it (or when I lower it manually).

X and Y axis are homing fine. However, when I attempt to home the Z axis, it never works. It will go up a bit, then down (as expected), but will stop in mid air and never reaches the bed (the light on the sensor will not light up). Then it will go up and down slower (as expected). It could stop higher or lower than the first probe, but never at the bed's height.

I never had this problem before. The endstop is working properly (light and status with M119)

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1744778/Configuration.zip)

charles2690, 2018-02-21:
I should add that even when tested with a multimeter, the sensor does not appears to be triggered

M48 results:

Finished!
Mean: 26.301061 Min: -4.208 Max: 46.796 Range: 51.004
Standard Deviation: 16.685581

Sebastianv650, 2018-02-21:
Maybe a pullup is needed but not set? Sounds like a pin which value is floating around.

charles2690, 2018-02-21:
It is already done

// coarse Endstop Settings
#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

thinkyhead, 2018-02-23:
With 1600 steps on your Z axis you are limited in your Max Feedrate for Z. If you exceed it then the board will never have a chance to reset the watchdog, and it causes the board to reboot. You should lower your Z Max Feedrate with `M203 Znnn` until you find a value where your Z axis can handle it without resetting the board.

closed 2018-03-07 by thinkyhead