MarlinFirmware/Marlin #9750: G2 Small Holes Distorted

rmisko11, 2018-02-21:
------------------------------------------------------------
### Bug Report
- Description: --- when milling a small hole it is sometimes distorted
- Expected behaviour: --- clean well formed hole
- Actual behaviour: --- hole circumference is distorted where parts are outside the specified circle
- Steps to reproduce:
  
Note: the tool diameter is 3 mm and hole diameter is 4.5 mm

G0 X140.00 Y140.00 F1500
G91             ( Switch to Relative )
G0 X-0.75 F1500   ( Offset X for correct hole position )
G0 Z-5 F100          ( Go to ground level )
G2 I0.75 Z-10.00   ( Spiral down to hole depth )
**G2 I0.75 Z0         ( Repeat at depth )   ;  if I remove this it seems to work okay **
G2 I0.75 Z10.00    ( Spiral back up )
G0 Z5 F1500         ( Go to safe height of 5mm )
G0 X0.75               ( Remove X offset )
G90             ( Switch to Absolute )

AnHardt, 2018-02-21:
Did you play with this parameters?

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L730-L739

1mm seems not to be very suitable for that small holes.

dennisbun, 2018-02-22:
Try using "R"  maybe?

rmisko11, 2018-02-22:
AnHardt,

I changed line 735 to:

#define MM_PER_ARC_SEGMENT 0.5 // Length of each arc segment

which appears to work quite well.

Thank you I it is very unlikely that I would have figured that out without
your help.

closed 2018-02-22 by thinkyhead

AnHardt, 2018-02-22:
If your holes are going smaller and smaller one time you will hit the next border.
Depending on your steps/mm you may have to adjust
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L751-L752

rmisko11, 2018-02-22:
> 1.x/Marlin/Configuration_adv.h#L751-L752

Thanks for the tip but mostly use drill bits for tiny holes, in this case I
didn't have a chuck that would fit my 5mm bit so I was improvising.

thinkyhead, 2018-02-23:
Proposing that…

Very small circles should override `MM_PER_ARC_SEGMENT`. Basically, the number of degrees covered by a segment in a full circle should be <= 45, and probably <= 30 so there are at least 8-12 facets. If a circle is so small that `MM_PER_ARC_SEGMENT` is covering more than that number of degrees, it should do a finer sub-division. If the total arc is also very small, like under 45 degrees, then it should be split into at least 2 or 3 parts to give a curve.

Adding this logic would slightly increase load, but not too badly.

TehBigA, 2018-03-08:
I actually just came across this issue on my printer and was going to re-flash to adjust the config. I may have missed the conversation in another thread, but why isn't the number of segments calculated based off radius?

thinkyhead, 2018-03-08:
> why isn't the number of segments calculated based off radius?

Because no one has so far contributed code making that modification.