MarlinFirmware/Marlin #9751: [FR] Receive progress when printing from SD Card

marcboivin, 2018-02-21:
It would be of great use to be able to receive print progress when printing from the SD card. 

Feature request in OctoPrint to have proper progression when starting a print from the SD card. 

A plugin could be made to pool the printer with M27 at a fixed interval, but the much cleaner solution IMHO would be to have a custom G-Code that would enable progress report when printing from the printer's SD card.

My general knowledge of G-Code is quite limited and I couldn't find general design principle on how to correctly implement this feature. I'm opening it here in the hope that the community finds the feature appealing enough to implement.

Feature request from OctoPrint https://github.com/foosel/OctoPrint/issues/2335

Regards,

Marc

houseofbugs, 2018-02-21:
https://github.com/MarlinFirmware/Marlin/issues/9636

TheSFReader, 2018-02-22:
I would like to code the feature, but wonder if it's better as an extension (through an additional / optional parameter) to M27 or as a new M Code (in which case I'll have to find out how they are allocated.

I'd add it just like the M155 temp auto report (including the additional capability report).

TheSFReader, 2018-02-22:
First implementation pushed to my fork, but since I'm on vacation till sunday, I have no way to test it 
https://github.com/TheSFReader/Marlin/tree/Report_SDCardProgress
If anyone is able to check it before Sunday, I'll make a pull request.
The AUTO_REPORT_SD_STATUS option is enabled in the configuration_adv.h, and used with the M27 command : 
M27 S5 will (if it's working correctly) auto-report the status every 5 seconds, 
M27 S0 should disable the autoreport.

marcboivin, 2018-02-22:
Same here, can't test until Sunday. I asked people from Octoprint if they can test it before that. 

Thank you!

TheSFReader, 2018-02-26:
Tested it, corrected it, and sent a pul request #9832

thinkyhead, 2018-02-26:
Merged into `bugfix-2.0.x`.

closed 2018-02-26 by thinkyhead