MarlinFirmware/Marlin #9754: CR-10 Mini - NO SDCARD PRESENT - ERROR

ldkraemer, 2018-02-22:
### Bug Report
- Description:  Creality CR-10 Mini  
- Software:  Marlin-bugfix-2.0.x/

Software builds properly, loads properly, and functions properly, EXCEPT that the SDCard
ALWAYS shows NO SDCARD PRESENT, even after Power Off for for 5 minutes.

These are the Components in my CR-10 Mini
https://www.ebay.ie/itm/Creality-3D-CR-10-3D-Printer-Control-Mainboard-Upgrade-Circuit-Board-Control-/253193655402
https://www.aliexpress.com/item/Universal-LCD-12864-3D-Printer-Disp

Searches of Issues & Pull Request for bugFix 2.0.x haven't located any solutions.

Any help or ideas would be appreciated.

Larry

[CR-10Mini-Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1746213/CR-10Mini-Configs.zip)

thinkyhead, 2018-02-23:
First, try doing a "low level" format of the SD card to make sure it isn't just very slow.

If that doesn't help then try changing this line in `pins_RAMPS.h`:
```diff
-     #define SD_DETECT_PIN     49
+     #define SD_DETECT_PIN     -1
```
…and see if the SD card will respond to the "Init SD Card" item in the LCD menu.

ldkraemer, 2018-02-23:
Scott,
I've tried with these changes and without the changes.

I have defined CR10_STOCKDISPLAY in my configuration.

357    #if ENABLED(CR10_STOCKDISPLAY)

      #define LCD_PINS_RS       27
      #define LCD_PINS_ENABLE   29
      #define LCD_PINS_D4       25
     * #define SD_DETECT_PIN     -1*

      #if DISABLED(NEWPANEL)
        #define BEEPER_PIN      37
      #endif

368    #else

414      #if ENABLED(CR10_STOCKDISPLAY)
        #define BTN_EN1         17
        #define BTN_EN2         23
        *#define SD_DETECT_PIN   -1*
      #else
        #define BTN_EN1         31
        #define BTN_EN2         33
       * #define SD_DETECT_PIN   49*
422      #endif

I am still getting the NO SDCARD PRESENT on bugFix version 2.0.x

With or without the changes it does not init my SDCard.  All information is
still on the card and I can not
access it.  For about 3 seconds I cannot change the menu, so it is acting
like it's busy formatting the SDCard.
After that I can scroll through the menu.

If I load Version bugFix 1.1.x it works fine.  All files are present and I
can select one for printing.

THANKS.

Larry

thinkyhead, 2018-02-24:
Duplicate of #8746 

Let's move discussion to that issue.

closed 2018-02-24 by ldkraemer

ldkraemer, 2018-02-24:
Scott,
OK.  I've looked over the differences in bugfix2.0.x and bugfix1.1.x, and
there are so many it's almost impossible
to know what could be causing the problem.

What is a good way to debug the code?  I'm not that familiar with the
Arduino IDE?

Larry

AnHardt, 2018-02-24:
Debugging with the ArduinoIDE is hopeless.
What you need, is a real text editor of your choice, being able to search in a directory tree. (Notepad++ works for me (Windows)). A program or plugin to compare 2 versions of a file is also very helpful.

thinkyhead, 2018-02-25:
If it works with 1.1.x then stick with that. Eventually we'll find and fix the issue.

The best tool for building 2.0.x is Visual Studio Code + PlatformIO, without a doubt.