MarlinFirmware/Marlin #9758: G91 not working with M206

rmisko11, 2018-02-22:
I am using Marlin 1.1.7 where I am trying to use relative positioning (G91) with M206 but the M206 offset appears to be added to the relative position request, e.g.

Start at absolute position X0
G90    ( Absolute positioning )
G0 X0 ( No move as expected )
G91  ( Switch to relative positioning )
G0 X0 ( No move as expected )
G90 ( Go Back to Absolute )
M206 X-20  ( Reset home position to X20 )
G0 X0  ( Moves 20 units along X-axis as expected )
G0 X0 ( Moves 20 units more along X-axis which is NOT what I expected – I expected NO MOVE )
G0 X0 ( Moves 20 units more along X-axis which is NOT what I expected  – I expected NO MOVE )
.
.
.

How do I get relative positioning in my adjusted coordinate system?

AnHardt, 2018-02-22:
Let's have a look at `bugfix-1.1.x`.
Global search for "G91" -> sets `relative_mode` only.
Global search for `relative_mode` -> the relevant use is in [gcode_get_destination()](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L3159-L3190)
Removing any irelevances it comes down to
```cpp
destination[i] = (relative_mode)
        ? current_position[i] + v
        : LOGICAL_TO_NATIVE(v, i);
```
with "v" being the value.
Because the non relative mode seems to work try
```diff
- current_position[i] + v
+ LOGICAL_TO_NATIVE(current_position[i] + v, i)
```

AnHardt, 2018-02-22:
`1.1.7` is looking different
```cpp
void gcode_get_destination() {
  LOOP_XYZE(i) {
    if (parser.seen(axis_codes[i])) {
      const float v = parser.value_axis_units((AxisEnum)i) + (axis_relative_modes[i] || relative_mode ? current_position[i] : 0);
      destination[i] = i == E_AXIS ? v : LOGICAL_TO_NATIVE(v, i);
    }
    else
      destination[i] = current_position[i];
  }
```
so maybe already fixed in `bugfix-1.1.x`.

There is good reason for requesting [here](http://marlinfw.org/docs/basics/reporting_bugs.html)
```
Include information about:
* The Marlin version (please test [the Development branch] before submitting reports)
```

rmisko11, 2018-02-22:
Sorry, I misspoke I am using 1.1.8.

rmisko11, 2018-02-22:
For 1.1.8 if I do this:

//      destination[i] = i == E_AXIS ? v : LOGICAL_TO_NATIVE(v, i);

destination[i] = (relative_mode)
        ?  v                                               // Note: I don't
use current_position[i]
        : LOGICAL_TO_NATIVE(v, i);

It seems to fix my problem but I have no idea if that causes problems
elsewhere.

thinkyhead, 2018-02-23:
Duplicate of #9424. Already patched.

closed 2018-02-23 by thinkyhead

rmisko11, 2018-02-23:
AnHardt,

Took a while but am beginning to understand.  I did search the bug list
except I was focusing on M206 and totalling missed the existing G91 one,
howver, going through that I didn't understand what was said.  That helped
get my code working a patch I shared in another message.

 Eventually I found bugfix-1.1.x gcode_get_destination in 1.1.8 with the
one in bugfix-1.1.x which appears to work well.

Anyway sorry for any inconvenience I caused but thank you for helping me
find the solution to my problem.