MarlinFirmware/Marlin #9759: [2.0] SHOW_CUSTOM_BOOTSCREEN & CUSTOM_STATUS_SCREEN_IMAGE compile error

fisero, 2018-02-22:
### Compile Error
When I compile with SHOW_CUSTOM_BOOTSCREEN & CUSTOM_STATUS_SCREEN_IMAGE  I get an error:
```
from C:\TEMP\sketch\src\lcd\ultralcd.cpp:112:
C:\TEMP\sketch\src\lcd\dogm/dogm_bitmaps.h:40:36: fatal error: /../../_Bootscreen.h: No such file or directory
     #include "../../_Bootscreen.h"

                 from C:\TEMP\sketch\src\lcd\ultralcd.cpp:112:
C:\TEMP\sketch\src\lcd\dogm/dogm_bitmaps.h:135:36: fatal error: /../../_Statusscreen.h: No such file or directory
   #include "_Statusscreen.h"
```
In file **dogm_bitmaps.h** are wrong paths in includes. There should be "**../../../**_xxxscreen.h"
Thus according to comments in Configuration.h those files should be placed in Marlin source root folder.

thinkyhead, 2018-02-23:
Thanks! Now patched.

closed 2018-02-23 by thinkyhead

fisero, 2018-02-23:
Also on line 137 path is wrog  #include "_Statusscreen.h" ;) It should be #include "../../../_Statusscreen.h"

thinkyhead, 2018-02-23:
Thanks. Trying to do 5 issues at a time when I should do 1 at a time…

commit 0542b037 (2018-02-23): Fix bootscreen include path

Fix #9759