MarlinFirmware/Marlin #9765: Compile-error with #NO_VOLUMETRICS enabled

Master92, 2018-02-22:
### Compile Error
When I compile with `NO_VOLUMETRICS` I get an error:
```
sketch/ultralcd.cpp: In function 'void lcd_control_filament_menu()':
ultralcd.cpp:3638: error: 'class Planner' has no member named 'filament_size'
         MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);
                                                                                 ^
sketch/ultralcd.cpp:290:30: note: in definition of macro '_MENU_ITEM_PART_2'
         menu_action_ ## TYPE(__VA_ARGS__); \
                              ^
sketch/ultralcd.cpp:345:66: note: in expansion of macro 'MENU_MULTIPLIER_ITEM'
     #define MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(type, label, ...) MENU_MULTIPLIER_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)
                                                                  ^
sketch/ultralcd.cpp:3638:9: note: in expansion of macro 'MENU_MULTIPLIER_ITEM_EDIT_CALLBACK'
         MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);
         ^
ultralcd.cpp:3638: error: 'class Planner' has no member named 'filament_size'
         MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);
                                                                                 ^
sketch/ultralcd.cpp:294:102: note: in definition of macro '_MENU_ITEM_PART_2'
         lcd_implementation_drawmenu_ ## TYPE(encoderLine == _thisItemNr, _lcdLineNr, PSTR(LABEL), ## __VA_ARGS__); \
                                                                                                      ^
sketch/ultralcd.cpp:345:66: note: in expansion of macro 'MENU_MULTIPLIER_ITEM'
     #define MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(type, label, ...) MENU_MULTIPLIER_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)
                                                                  ^
sketch/ultralcd.cpp:3638:9: note: in expansion of macro 'MENU_MULTIPLIER_ITEM_EDIT_CALLBACK'
         MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);
         ^
exit status 1
'class Planner' has no member named 'filament_size'
```

When I comment lines 3634-3653 in ultralcp.cpp, the compilation finishes without errors, so there seems to be an error in the check against ```parser.volumetric_enabled ```.

Using a RAMPS1.4 with ```REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER``` and Marlin 1.1.8.

AnHardt, 2018-02-22:
#9083

thinkyhead, 2018-02-23:
Duplicate of #9083

closed 2018-02-23 by thinkyhead