MarlinFirmware/Marlin #9766: [ThinkyHead] - Filament run COMPILE ERROR ' read explanation'

Studiodyne, 2018-02-22:
In the class , all is private by default , and all your functions are used in extern 
I have added ' public: ' for all functions , and now all compile  and works

I have several warning about , some forbids declarations but code works 

```
class FilamentRunoutSensor {
public :
  FilamentRunoutSensor() {}

  static bool filament_ran_out;
  static void setup();
```

teemuatlut, 2018-02-22:
https://github.com/MarlinFirmware/Marlin/pull/9769

Studiodyne, 2018-02-22:
Yes 👍 
But you scarify too much the thinkyhead style loll
He will never love you re write so many lines

Thinkyheadisation ! Is the first rule here lollllllllll

thinkyhead, 2018-02-23:
Love you man, but this kind of stuff will get you banned.

closed 2018-02-23 by thinkyhead

Studiodyne, 2018-02-23:
Please  your are develloper , don't try to be a menace for society 👍  you are not good in this job
and , please , inform this issue of what you have decided ,  if you have fixed something , if issue have a happy end 
Thks

thinkyhead, 2018-02-23:
I'm much happier now.