MarlinFirmware/Marlin #9767: Issue with 1.1.8 firmware and Anet A6 with RAMPS 1.4 DRV8825 with 1/32 steps

sidbyron, 2018-02-22:
I have an issue with Marlin 1.1.8. Recently I changed the anet's mainboard with Ramps 1.4 and DRV8825. In the beggining I tried tu used them with 1/16 and I flashed marlin 1.1.8 really I was working for 2 days becuase when I printed the cube calibration I got an issue with the corners. I 've working with Simplify 3D so I tried to improve the profile S3D and it never worked so I return to 1.1.6 (it was working very well with Anet's mainboard) and I ajusted the settings for work with RAMPS 1.4 (steps for work with 1/32 and adjust the PIDS and some ither options) and I put all the jumpers for work with 1/32 so the issue disappear and it works like a charm I know that S3D ahve some isssue about this and all google you can find how to fix this with some options in Simplify but I think some part of this issue it's related with the firmware.

 Obviusly I checked the current and my anet's motors are working at 0.55 v .

 Today I flashed again Marlin 1.1.8 and I compare all the settings with 1.1.6 version and all settings it's the same (feedrates, accelerations, steps, PIDS) and the issue only happens with 1.1.8

This issue is more visible with S3D with Cura and I cannot any information about this. After each flash firmware I always put M502 and M500. 

Somebody knows what happen or how to solve it? I think the issue is the new version of marlin. Im sure this issue not is related with Mechanical issue because Why is Marlin working well with 1.1.6 version?

Here is the picture:

[image]

thinkyhead, 2018-02-23:
Have you ever used `LIN_ADVANCE`?

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

sidbyron, 2018-02-25:
I 'd been investigating about LIN_ADVANCE in this case the feature isn't active and I never have used. Do you recommend the activation of this feature? if I read well I need to put the K value by each type of material ( one for PLA, one for ABS, one for PETG, bla,bla )

Regards.

thinkyhead, 2018-02-28:
Duplicate of #9494

thinkyhead, 2018-02-28:
Continue discussion on that topic. A deeper investigation will begin in earnest soon.

closed 2018-02-28 by thinkyhead

peekpt, 2018-02-28:
Sorry for the low input but I can confirm that was using an older fw 1.1.x bugfix, then I updated my printer to 2.0.x bugfix and despite the noise, I was having extremely good corners with that older K . I tried to tune new K pattern and find the perfect extruded line and notice although the fast->slow transition was ok, the slow->fast transition had a strange blob in every calibration lines. 
IMO a strange pause/lag in occurring in that transition, when it changes acceleration from slow to fast. Exactly what is happening to the pictures that has been posted. In the corners, the travel movement slows down, changes direction and starts to accelerate. In the middle of the process makes that strange blob

thinkyhead, 2018-03-01:
> IMO a strange pause/lag in occurring in that transition, when it changes acceleration from slow to fast.

It is a sound theory. We're delaying the release of 1.1.9 until we figure this out.