MarlinFirmware/Marlin #9771: work MKS Robin on STM32

userosos, 2018-02-23:
Hi everyone!
Will marlin work with MKS Robin?

thinkyhead, 2018-02-23:
We'll add it to the "to do" list. Thanks for visiting.

washeree, 2018-05-24:
Cant wait for it. I think so many people stuck with their fw. Thank you.

vivian-ng, 2018-06-17:
I found this reverse engineered pinout for the MKS Robin while trying to find a solution, wonder if it will be of help to those trying to write a version of Marlin for it.
[https://docs.google.com/spreadsheets/d/1xn-nZW4KOSEYWt05clBIl1fS5pS82zgXlpCp8Voftiw/edit#gid=0](url)

Got it from this thread on RepRap.org:
[https://reprap.org/forum/read.php?13,716774,page=3](url)

alexxy, 2018-06-19:
I can create pinout files for MKS Robin, but i cannot test it. 

So if someone with board can test i can do it

washeree, 2018-06-19:
i already flashed my board while i was trying to dump fw so i can be victim no problem 👍  Why do you think display wont work?

alexxy, 2018-06-19:
because this type of display seems not supported by marlin.

what type of stm32 is on Robin?

washeree, 2018-06-20:
May be in future display can work. I think so its a TFT32 with ribbon connector. Its a F103ZET6.

alexxy, 2018-06-20:
yep, but display code running on same controller (not on separate one as on TFT24/32)

washeree, 2018-06-20:
Can we assign different display such as RepRapDiscount Full Graphic Smart Controller?

vivian-ng, 2018-07-27:
Has there been any luck on getting Marlin to work with the MKS Robin?

gerleimarci, 2018-08-25:
And what about MKS Robin Lite?

pen25, 2018-09-18:
i ordered two and will be shipping 1 to scott. just got them in and will ship 1 to scott sometime this week. will probably wait until friday since he is headed to makerfair nyc this weekend.

imrj, 2018-11-13:
hey guys any update on this? just got me an MKS robin board, would love to try it out with Marlin!!

jmz52, 2018-12-17:
@washeree,
Original bootloader for MKS Robin (STM32F103ZET6) can be downloaded from https://reprap.org/forum/read.php?13,716774,839792#msg-839792

You can connect 'RepRapDiscount Smart Controller' or 'MKS MINI 12864' to MKS Robin, but the 'RepRapDiscount Full Graphic Smart Controller' is not supported for STM32F1 HAL yet.
http://3dtoday.ru/upload/main/bba/bba66777a72d90df6aeaf7ff0ea514ba.jpg

pen25, 2018-12-17:
I can say Scott just moved. So don't know what the status is. I figured he would get to it when he got to it. I still haven't tried it in its stock form yet. But I'm lazy

imrj, 2019-01-04:
So can I just flash mine using Marlin 2.0 and platform.io and use the STM32F1 Hall? I can take care of the pins assignment from a spreadsheet I have that has all the pins definition for Robin....

vivian-ng, 2019-01-04:
@imrj How about creating your own pins file for the MKS Robin to give it a try?
I think you can follow the idea [here](http://marlinfw.org/docs/hardware/boards.html#adding-a-new-board). If it works, you can then send a pull request for it to be added into the 2.0 branch.

jmz52, 2019-01-04:
@imrj, MKS Robin has it's own build environment in platformio.ini file - "mks_robin", it has the right HAL and encryption feature to use Marlin with original MKS bootloader (i.e. install/update Markin from SD card)
Problem is that recent changes messed with configuration examples and STM32F1 HAL, so right now bugfix-2.0.x does not work on MKS Robin. 
Also these is an open pull request https://github.com/MarlinFirmware/Marlin/pull/12782 to add support for SD card on SDIO interface used on Robin board.

There is no need to write your own pins file - Marlin/src/pins/pin_MKS_robin.h already contains ping definition for MKS Robin board. Add `#define MOTHERBOARD BOARD_MKS_ROBIN` in your Configuration.h to use it.

If you want to try Marlin 2.0 you can use code from [Implement-SDIO-support-for-STM32F1](https://github.com/jmz52/Marlin/tree/Implement-SDIO-support-for-STM32F1)
and replace files 
Marlin/src/HAL/HAL_STM32F1/watchdog_STM32F1.cpp
Marlin/src/HAL/HAL_STM32F1/watchdog_STM32F1.h
by
[watchdog_STM32F1.cpp](https://github.com/jmz52/Marlin/blob/MKS-Robin-Touch-POC/Marlin/src/HAL/HAL_STM32F1/watchdog_STM32F1.cpp)
[watchdog_STM32F1.h](https://github.com/jmz52/Marlin/blob/MKS-Robin-Touch-POC/Marlin/src/HAL/HAL_STM32F1/watchdog_STM32F1.h)

Configuration.h and Configuration_adv.h should be copied from Marlin/src/config/examples/Mks/Robin

This will get you the funtionality Marlin will have once SDIO code is merged and STM32F1 HAL is fixed.
MKS Robin TFT is used to display standard graphics 12864 UI - https://ibb.co/0sbJKXT
Please note that you will need encoder to get access to menu.
Once Marlin is compiled, look for Robin.bin in .pioenvs\mks_robin directory. You need to copy this file to SD card for firmware update. And yes, you can go back to original firmware after you've installed Marlin from SD card.

Alternatively you can use code from [MKS-Robin-Touch-POC](https://github.com/jmz52/Marlin/tree/MKS-Robin-Touch-POC)
It is already had Configuration.h and Configuration_adv.h setup for MKS Robin.
Along with support of on-board SD card this code provides you three on-screen buttons to access menu, and colored screen that indicates print progress (PRINTER_EVENT_LEDS feature). Or you can use M150 to set screens' color.
https://ibb.co/p1Z9Q2K
Screen's color here is actually green - an indication of completed print from SD card.

If you have wiped your bootloader you can either flash bootloader using file linked in my earlier comment, or remove the "extra_scripts" line from "[env:mks_robin]" section of platform.ini. Without "extra_scripts" option the marlin firmware will be compiled to use with no bootloader, just flash the firmware.bin using STLink.

jmz52, 2019-01-08:
@Sneakers82, you need to update STM32 packages used by PlatformIO.
One way to do it is to remove .platformio\platforms\ststm32 from your user's environment directory to force PlatformIO to download latest version.

Also you need to update file .platformio\packages\framework-arduinoststm32\STM32F1\libraries\SPI\src\SPI.cpp - change `SPIClass SPI(1);` to `SPIClass SPI(2);`
MKS Robin uses SPI2 to access touch screen controller.

What version of MKS Robin board and MKS Robin TFT you have?
Mine are v2.4 and TFT_v2.0, so this is what have been tested so far. Robin v2.3 with TFT_v1.1 has different (meaning - incompatible) LCD controller and different Touch Screen connection.
I am working on this but code is not ready to be pushed to GitHub yet.

washeree, 2019-01-11:
@jmz52 nice work really, thanks a lot. It was a big news for everyone i think so. My board is v2.3 and tft as you said v1.1 so bootloader is different or same? In my opinion they changed some pins?

jmz52, 2019-01-11:
@washeree, bootloader v2.3 *might* be different, but pins are the same - all revisions of MKS Robin use the same firmware. There should not be any problems with older versions of bootloader. And I have a confirmation that v2.4 bootloader is working on v2.3 board.

[MKS-Robin-Touch-POC](https://github.com/jmz52/Marlin/tree/MKS-Robin-Touch-POC) updated to support ili9328 controller found in TFT v1.1.
There are two sets of touch screen calibration parameters in Marlin\src\HAL\HAL_STM32F1\xpt2046.h 
one for TFT v2.0 (default) and another one for TFT v1.1.

You might calibrate your touch screen using this ["calibration tool"](https://github.com/jmz52/MKS-Robin-toolkit/blob/master/lcd_and_touch/Robin.bin)
It will also show you what LCD controller is installed in your display module.
Please let me know if controller is not the ili9328.

washeree, 2019-01-11:
Thanks a lot for your help jmz52, some how i couldnt compile marlin via vs code with pio. It gaves lots of errors. Im installing atom now lets see if it make differences.

washeree, 2019-01-11:
@jmz52 my screen is 8239ili

jmz52, 2019-01-11:
@washeree, can you please make a screenshot for me? Looks like x coordinate is inverted.

washeree, 2019-01-11:
Looks like yes ;)

jmz52, 2019-01-11:
@washeree, should be working now.

washeree, 2019-01-12:
@jmz52, so i will setup marlin and then use your calibration tool or still no chance v1.1 lcd working on marlin? Btw thanks a lot for your hard work again.

jmz52, 2019-01-12:
@washeree, [MKS-Robin-Touch-POC](https://github.com/jmz52/Marlin/tree/MKS-Robin-Touch-POC) supports MKS Robin TFT v1.1. You can try calibration parameters for v1.1 that are already present in Marlin\src\HAL\HAL_STM32F1\xpt2046.h (you need to comment v1.1 set of parameters). If these will not be accurate enough for you, use calibration tool to get values for your screen.

Marlin bugfix-2.0.x does not support v1.1 yet. There are several code changes I want to make before submitting pull request.

washeree, 2019-01-12:
@jmz52 thank you. I made it work, your values on Y axis little bit different then mine. Next week i will try with my p3steel. I wish i can code these things too.

jmz52, 2019-01-12:
@washeree the bed sensor works fine for me and some other folks who are using Marlin on their Robin boards (v2.4 and v2.3). Are 230 and 238 degrees measured on cold (i.e. room temperature) bed or heated one?
What type of sensor is configured in your Configuration.h file? 
Please check if you get -14 degrees when sensor is not connected and TEMP_SENSOR_BED is set to 1.

PhantomNix, 2019-01-13:
@jmz52 , I've spent the last two days wanting to throw this board against the wall. I'm having trouble compiling, going over and over the steps you took to make it work... Is there possibly something that could go wrong? I've completely reinstalled everything and started with fresh files, without editing them so not sure where I'm messing up.

Got the Robin V2.4 wit V2 TFT

washeree, 2019-01-13:
@PhantomNix hello, i was having problem while compiling then i tried Atom IDE and all was fine some how with vs code couldnt compile it. I hope it helps.

washeree, 2019-01-13:
@jmz52 one last thing robin use which pins while compiling? pins_MKS_ROBIN.h ?

PhantomNix, 2019-01-13:
Yep, that's the one

jmz52, 2019-01-14:
@PhantomNix I am using PlatformIO with Atom. Were you able to compile Marlin with these?

@washeree, it looks like  hardware problems with MCU. Disconnected temperature sensor with type 1 in Configuration.h should give you a -14 degrees. Different Reading you are getting indicate that there is either electrical problems with board (like failed resistors or capacitors or conductive dust) or failed MCU. Since you are getting same reading on all three temperature inputs and you have same reading ion both Marlin and MKS firmware, I suspect that MCU is the the one to blame.

washeree, 2019-01-15:
tahnk you @jmz52, i thought same and i checked in front of pin cap and res all are fine then i think MCU pin fried or dust with compressor i clean it and i tried same results. I will look for stm32f1 from china, will swap and reflash then lets see. Tahnk you again.

regtoy, 2019-01-25:
mksrobin TFT V1.0  and calibration @jmz52 
/* MKS Robin TFT v1.0 */
#define XPT2046_X_CALIBRATION    -12545
#define XPT2046_X_OFFSET           347
#define XPT2046_Y_CALIBRATION     9282
#define XPT2046_Y_OFFSET           -26

regtoy, 2019-01-26:
Hi, anybody who can help? I have a question
Is it possible to add uart port for TMC2208?
If you have, which port can be used for this job

youprint, 2019-02-18:
Now it works ! Thanks so much jmz52 for your bootloader 👍 
https://ibb.co/x86mvxp
Now I'm going to compile Marlin 2.0 and try...

youprint, 2019-02-18:
Compilation done with success on Atom/plaformio with your repo jmz52, thanks again ! LCD support great on your repo (POC) !
https://ibb.co/Q6NvMdY
I'm going to try it tomorrow with motors and TMC 2208.
Thanks again jmz !

boelle, 2019-02-20:
@userosos 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

pinchies, 2019-02-25:
Hi all,

I'm trying to configure the JGAurora A5S/A1 motherboard using the same setup as the MKS Robin, as they have the same CPU. We've got the pinout pins.h sorted, but I'm having problems with the device freezing when it tries to read the bed temp. For anyone who successfully used the MKS Robin with Marlin 2.0, did you have to change the ADC configuration? **Would anyone be happy to share a zipped build folder (.platformio and zipped project folder?) so I can see the working setup for the MKS Robin?** 

I'm just wondering if I need to change the ADC initialisation somehow in  ~\.platformio\packages\framework-arduinoststm32\STM32F1\variants\generic_stm32f103z\board.cpp

Cheers.

MS1987, 2019-05-15:
Found the schematic diagram of MKS Robin has been open source:
https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/Hardware
And also the MKS Robin2, which using the STM32F407 power mcu:
https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin2/Hardware

nanokarlo, 2019-05-21:
Hello, is Marlin supported for STM32F103VET6 MKS Robin Mini ???

jmz52, 2019-05-21:
There is a fork of Marlin that supports MKS Robin Mini - https://github.com/berghauz/Marlin-bugfix-2.0.x-FLSUN-QQ-S
More information can be found at https://www.facebook.com/groups/120961628750040?view=permalink&id=404428707069996&anchor_composer=false

spaivaras, 2019-05-31:
Can someone clarify what is the status with mks robin and TFT in Marlin 2.x i see configuration files and the board seems to be supported out of the box.  But TFT doesnt work at all. u8g_com_stm32duino_fsmc.cpp doesnt have any dma code. TFT works with fsmc.cpp from jmz52 fork and lcd 320x240_upscale class tho. So I guess its still not merged? Thank you

1kemanciii, 2019-06-10:
@jmz52 tft v1.1 touch scrren ok.Mks robin v 2.2 board
SD card problem. No SD Card

What should I do for sd card

github051, 2019-06-19:
can anybody is having detailed step by step procedure to flashing of mks robin v2.4 board? Thank you

jmz52, 2019-06-19:
@github051 
- compile firmware
- copy 'Robin.bin' to SD card
- insert SD card to MKS Robin board
- power on or reset MKS Robin board

BigDi, 2019-06-19:
Sorry for the questions. Is it possible to port this to a Tronxy Mainboard CXY (aka Chitu f mini) first view looks like same hardware.

github051, 2019-06-19:
Hi JmZ52,
Can you please explain the procedure for firmware compilation and settings.
I am new to 3d printing world.

pinchies, 2019-06-19:
I wrote this for another STM32 printer. Hope this [marlin 2.0 compile guide](https://jgauroraforum.com/discussion/730/how-to-compile-your-own-custom-marlin-2-0-firmware-for-jgaurora-a5s-and-a1#latest) helps you get started:

0bey, 2019-06-19:
@pinchies thanks for helping, but i can't compile without error
i have take configuration.h and configuration_adv.h here : https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware/Marlin2.0-MKS-Robin/Marlin
i use your source and tuto
platformio.ini change to environement "mks_robin"
`Marlin\src\HAL\HAL_STM32F1\u8g_com_stm32duino_fsmc.cpp:35:10: fatal error: U8glib.h: No such file or directory`
(monochrome 8b lib, necessary ?)
sorry i'm newbie in firmaware environement and programming (and french...)

jmz52, 2019-06-19:
@0bey 
You need to remove `U8glib-HAL` from `lib_ignore` in platformio.ini file. It really should not be there for MKS Robin, but this line keeps getting back over and over again.

Configuration files for MKS Robin are located in https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/Mks/Robin.
Configuration files you were trying to use are for different code base - https://github.com/jmz52/Marlin/tree/MKS-Robin-Touch-POC. This code is a bit outdated proof-of-concept and is not part of main Marlin repository. Configuration files from this branch are not fully compatible with main Marlin repository.

github051, 2019-06-20:
Hi jmz52,
i was downloaded the firmware ( https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin ) and uploaded via SD card. Now i am able to test the g code, but for the Z-Axis i need two motors. How to configure the firmware for dual motors ?

0bey, 2019-06-20:
Hey @jmz52

I use officiel source, use exemple configuration.h et configuration_adv.h for mks_robin and U8glib-HAL
Compilation it's OK :) 

Thx for help

Now, I learn to configure before try on my alphawise U20

AndroWolf, 2019-06-22:
Hi, sorry i have absolutly no idea what i do here :-) but i need only Marlin 2.0 on my Robin 2.3 Board :-)
I Hope the community here can help me out to get it worked:

- Atom and PlatformIO installed. (got a Welcome to PlatformIO)
- Clang 3.9.1 for Windows (64-bit) installed with select “Add LLVM to the system PATH” option on the installation step
- https://github.com/makerbase-mks/MKS-Robin -> clone or download button to get master zip file
- extract/copy out "MKS-Robin/MKS Robin/firmware/Marlin2.0-MKS-Robin/" to a new Folder on Dektop as "RobinSDCARD" 
- Open in Atom "Welcome to PlatformIO Home" with Open project Folder "Desktop\RobinSDCARD\Marlin2.0-MKS-Robin"
- press PlatformIO:Build an got error with U8glib-HAL (many of them) 
- klick on platformio.ini and strg+f "U8glib-HAL" found 5 what now?

`lib_ignore  =
  U8glib
  LiquidCrystal_I2C
  LiquidCrystal
  NewliquidCrystal
  LiquidTWI2
  Adafruit NeoPixel
  TMCStepper
  Servo(STM32F1)
  TMC26XStepper
 # U8glib-HAL
  c1921b4`

at 347 shlould i delete only this one? i tried it and got after save same U8Glib-Hal fails.

0bey, 2019-06-22:
This morning I use same source without problem
@AndroWolf did you try without U8glib-HAL from platformio.ini like say jmz52 ?

AndroWolf, 2019-06-22:
only find and delete? "U8glib-HAL"

AndroWolf, 2019-06-22:
> 
> 
> This morning I use same source without problem
> @AndroWolf did you try without U8glib-HAL from platformio.ini like say jmz52 ?

i think i have something not installed or missed? sad that there not a list what tools or programm i need to build this robin.bin xD

[12 more replies]

closed 2019-08-05 by thinkyhead