MarlinFirmware/Marlin #9772: Prints start outside the bed instead of center where part is located

locutus40, 2018-02-23:
ok i upgraded to .18 added all my stuff.. i have a ramps and a arduino mega.. with prusa i3 and a bl touch...
it homes fine g29 fine.. when it goes to print it goes back to home then starts oozing out filament without moving at the home position... any ideas? i placed my print in the middle of the bed on slicer.. i also made sure i have 200 mm x 200mm set in marlin and repieter host... ... maybe a bad arduino? or driver going? ive had it 5 years...

heres a video..
http://www.mediafire.com/file/8fa0aun62lv5ama/20180223_014853.mp4

thinkyhead, 2018-02-23:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

locutus40, 2018-02-23:
i am using bug fix 1.1x... it still does it.. and i cant use 2.0 because i use a mega which is not 32 bit.. i tried it again this time with slicer instead of prusa slicer.. same thing.. it shoots out filament like a faucet ... and just sits in home position...

thinkyhead, 2018-02-23:
Apologies if the "Issue" form was not prominent enough. We ask for certain details up-front so we don't just chase our tails. Please fill out the following form and one of our helpful volunteers will try to help solve your problem.

---
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

thinkyhead, 2018-02-23:
You'll have to re-post your video to YouTube. We have ethical issues with MediaFire.

locutus40, 2018-02-23:
ethical issues? what did they do? im trying to figure out how to upload to youtube

locutus40, 2018-02-23:
ok its uploading to youtube link shortly..

thinkyhead, 2018-02-23:
> ethical issues? what did they do?

They're spammy, with hidden popups and the rest.
Meanwhile, with YouTube I don't have to download a 90MB file to see your issue.

locutus40, 2018-02-23:
yea theres popups.. when you say ethical i thought your talking about them killing baby kittens or something.. pop ups are everywhere even on youtube.. but heres the youtube link..
i was also thinking i had a corruption of the boot loader so maybe its a failing arduino mega..
https://youtu.be/KgZIyzyFfxs

ive been sick for a month.. and life.... everywhere you look? ive posted an issue here and tried helping someone out about an hour ago...

i dont come on here often...reason being i havent had issues for a long time till i tried updating.. then it all went to hell fast

thinkyhead, 2018-02-23:
> then it all went to hell fast

Ain't that always the way?

thinkyhead, 2018-02-23:
So, to reiterate: ***Most importantly, we need to see your configuration files.***

Meanwhile, try turning off `SEGMENT_LEVELED_MOVES`. And reset your EEPROM with `M502` / `M500` just to be sure there's no funky old data lying around.

locutus40, 2018-02-23:
where is segment leveled moves in config h? ill try all that now

thinkyhead, 2018-02-23:
Where? Please use the search function of your text editor.

thinkyhead, 2018-02-23:
You can ZIP them up, as it explains in the template posted above.

locutus40, 2018-02-23:
ok might be tomorrow by the time i get all this figured out.. its 3 am here getting tired.. i m502 and m500 just now... im and im using arduino ide i see no text search function?

locutus40, 2018-02-23:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1751425/Configuration.zip)

hope this worked

thinkyhead, 2018-02-23:
Try commenting out the lines `#define NUM_SERVOS 3` and `#define SERVO_DELAY { 300,300,300 }`. See if that helps at all.

locutus40, 2018-02-23:
ok ill do that now.... i still cant fine this.. im using arduino ide theres no search function.. and debug turn on where is that in the config h file so i can turn it on?

locutus40, 2018-02-23:
im dowloading notepad now.. ive disabled the servos still need to find that feature and go from there

locutus40, 2018-02-23:
ok text editor is soo nice i disabled servos enabled debugging and disabled that other thing im going to upload heat up my bed and try a print stand by

AnHardt, 2018-02-23:
Like in all cases where an axis is not moving, check the endstop states with M119.

locutus40, 2018-02-23:
end stops work it homes and bed levels fine it just starts printing at home postiion and shoots out a ton of filament...  standbye heating now

thinkyhead, 2018-02-23:
If the problem persists, also zip up the G-code file you're printing so we can see if any odd settings are being applied.

locutus40, 2018-02-23:
ok its heating getting ready to home etc

locutus40, 2018-02-23:
still did it.. its shooting out filament on the home position like mad so much so it cant keep up with it entering and its grinding.. ill upload a 200mb video to youtube its going to take a while so ill come back tomorrow and post it... here... im thinking its the arduino. but heres the new config.h file to look over to prove i did what you said i dont know how to get the debug thing so tell me how to do that..

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1751495/Configuration.zip)

theres the new config.. i show in the video i took where its printing in the exact center of the plate...
[velosynth-spoke-magnet-v1 (repaired).zip](https://github.com/MarlinFirmware/Marlin/files/1751499/velosynth-spoke-magnet-v1.repaired.zip)

theres the file im trying to print.. its a magnet holder for a bicycle spedo ive repaired it.. etc..

im at a loss for words guys... like i said it might be the arduinio its self seeing how the boot loader got corupted.. i had to upload a new bootloader via another arduino.. so it might be the chip its self...

take note i had to change sides of my x home to the left instead of the right when i used right after the error happened with the boot loader the x woulndt move no matter what i did so i placed it on the left and voila it works.. its all seeming like a hardware problem.. 5 years old.. something bound to break.. i just dont have another one to test or any money right now.. so if i get another one in a little bit a week or so ill re do it and test.. if no one else has any new input?

thinkyhead, 2018-02-23:
I can't do anything with an STL file. We need to see the G-code, please.

locutus40, 2018-02-23:
the g code hmm hold on let me get it

thinkyhead, 2018-02-23:
Printing from a host or from SD Card?

thinkyhead, 2018-02-23:
We'd like to see your `Configuration_adv.h` too, please.

locutus40, 2018-02-23:
printing from sd  card or my pc it acts the same spews out filament etc.. im uploading the video to drop box 10 minute to go then i have to upload it to youtube it will take an hour for that.. . i did give you my config heres it again 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1751525/Configuration.zip)

locutus40, 2018-02-23:
oh wait advanced sorry lol

locutus40, 2018-02-23:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1751528/Configuration_adv.zip)

thinkyhead, 2018-02-23:
So, your homing position is where? On the front-left or the back-right?

locutus40, 2018-02-23:
when looking at the machine aka video to the left and bed back towards me.... heres the gcode
[gcodeorigianl.zip](https://github.com/MarlinFirmware/Marlin/files/1751539/gcodeorigianl.zip)

thinkyhead, 2018-02-23:
That G-code would really help too, whenever you can get a hold of it…

thinkyhead, 2018-02-23:
That file only contains the very last bit of the G-code. Please export all the G-code from your slicer or host and send us that.

locutus40, 2018-02-23:
i tried thats all that it gives me... let me try again

thinkyhead, 2018-02-23:
Which Slicer or host?

locutus40, 2018-02-23:
[gcodefinal.zip](https://github.com/MarlinFirmware/Marlin/files/1751565/gcodefinal.zip)

slic3er and repiteir host.. i dont like pronterface because if you click a move button it can go way past your end stops hanvent figured out how to stop that so i just use repiteier all are up to date

locutus40, 2018-02-23:
in about 20 minutes ill have a new youtube video of it doing the same thingwith the stuff you told me to turn off and on

thinkyhead, 2018-02-23:
No need, thanks.

thinkyhead, 2018-02-23:
Here's the first move in that G-code file. See anything interesting?
```gcode
G1 X-66.903 Y-66.983
```

locutus40, 2018-02-23:
ok well its uploading...

so any thoughts? im like figuring a new board... or a new 32 bit board... ive never seen this problem before its strange... funny thing is its a genuine mega board

locutus40, 2018-02-23:
yea its placing the first move at the start position.. heres my work flow..

get part or design.. place in repiterhost.. then slice.. then hit print..  slicer is the one thats doing this then?

thinkyhead, 2018-02-23:
Yes, fix your slicer settings. -66, -66 is way off the bed. Tell the slicer that your bed goes from 0,0 to 200,200 and be careful to put the model in the middle before you export the g-code or print from it.

If your homing position is at the front-left corner of the bed, moving Y+ should bring the bed forward and moving X+ should move the nozzle to the right. If the bed home is at the back-right of the bed, your  Y+ should move the bed backward and X+ should move the nozzle left. If the machine is set up with the home in the front-right or the back-left of the bed, your handedness is backwards and objects will be printed mirrored.

locutus40, 2018-02-23:
ok where in slicer do i make the changes you suggest?

thinkyhead, 2018-02-23:
That's really beyond the scope of what we can do for you. Please continue with Google or http://forums.reprap.org/

closed 2018-02-23 by thinkyhead

locutus40, 2018-02-23:
in bed shape i have x155 y155
origins x 0 y 0.. 

do i just put where y is 200,200? im freaking lost

[newscreenshot.zip](https://github.com/MarlinFirmware/Marlin/files/1751632/newscreenshot.zip)
see what i mean here..
you suggest telling my slicer where the bed is but cant tell me where to change the settings?

locutus40, 2018-02-23:
https://youtu.be/6g9qFN5L2Eo heres what it does the new video same but with the whole process

locutus40, 2018-02-23:
ok i just put in 200x 200 instead of 155x 155... and sliced it it still says -66 -66 as the first line.. so it did nothing... im going to delete slicer and try another program or download the new version if there is one... or just toss my printer and pc out the window ll

locutus40, 2018-02-23:
yea im using 1.30 a dev version stable is 1.2.9 im gunna try that maybe thats why

locutus40, 2018-02-23:
wth you closed it.. its not resolved... what if i do get a new board and its still doing it with fresh software etc then it is a bug.. leave it open

thinkyhead, 2018-02-23:
It's not a bug. The problem is obvious to us with years in the field.

locutus40, 2018-02-23:
ok its obvious to you.. you told me to do 0,0 200 200.. 

where ? in slicer whre do i do that?
btw i deleted everything downloaded fresh stuff
to make sure..

locutus40, 2018-02-23:
nevermind im going to figure this out on my own...

thinkyhead, 2018-02-23:
Thank you. We have many things to do, and this project repository is not a support site. This is a hub for Marlin development. For help from other users, http://forums.reprap.org/

locutus40, 2018-02-23:
not a support site lol yet theres 500 issues with the firmware.. thats support or asking for help which i was doing.. i came up with a issue... and i belive someone else had a issue i was talking to similar like 2 days ago .. so to say you have many things to do.. implies your tooo important or members are too important to help little old me.. .. i came to you with a genuine issue.. ... if i do all this stuff and believe me ive got fresh copies of repiter host and the last stable version of slicer and it gives me the same issue then its a hardware... or your firmware and if i buy a new board and it still exsists its a firmware issue.. hence why i came here in the first place.. you know this is how you diag stuff is ask questions.. im a ase certified tech ive been working on electric vehicles an 5th wheels cars trucks you name it for over 20 years.. i ask questions still with all that knowledge i even have a college degree in my feild.. went though a physics class to get my degree in engineering and repair... so when im treated like this.. for asking questions it rolls me.. ive had a issue about ohh a few months ago hence why you asked me where i been because 3 of you got on my case  about it.. same treatment... i come here for help seems i end up fixing my own issues on my own.. and no help comes only critizizim god i suck at spelling that and its 5 am an di havent sleep in 24 hours...

so cut me some slack.. its a genuin issue and im trying to figure out what it is ..... thats all.. so dont chastize me for asking for alittle help or outside knowledge...

thinkyhead, 2018-02-23:
Best of luck to you. Glad I could help solve the issue. Your happiness is important to us.

AnHardt, 2018-02-23:
> not a support site lol yet theres 500 issues with the firmware.. th - bla bla bla

We (being able to read Marlins source) are able to find answers about everything you can ask about Marlin. Sometimes it needs a while, but in theory, we can.

The problem with your last question is - it's not about Marlin, but about your host program.

For RH: Ope4n 'Printer Settings', 'Printer Shape', check 'Classic Printer', - edit to your needs, 'OK'.

For Slic3r. Open 'Printer Settings' tab, click 'Bed shape:' 'Set...' , Edit Origin, 'OK',

locutus40, 2018-02-24:
// @section homing

// The center of the bed is at (X=0, Y=0)
//#define BED_CENTER_AT_0_0

what about this... if my bed is 200 mm x200mm what do i put in the above?

AnHardt, 2018-02-24:
You can't put something in there. You can activate it, or not.
Sorry for forgetting about that.

[58 more replies]