MarlinFirmware/Marlin #9773: Marlin prints too small

blackdevil8975, 2018-02-23:
I have an issue with my Marlin software. My bed size 6430x7500.

If I start a printing job the printer only prints in a size of 2x2cm.

When I manually move the printer I can use the full bed size.

My configuration.h is attached below.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1752156/Configuration.zip)

revilor, 2018-02-23:
Wow, that's amazing. The bed size of your printer is larger than my whole apartment!

So, when you say

>  the printer only prints in a size of 2x2cm

how big would you expect the printed object to be? With the little information you gave, this could be wrong steps per unit in Configuration.h or an issue with your slicer software.

teemuatlut, 2018-02-23:
Would be cool to see a picture of the machine.
Please also attach an example of gcode file that doesn't work.

xC0000005, 2018-02-23:
I can repro this using only gcode commands. My bed x is 120mm.

G28X homes the X axis correctly.
G1 X120 moves the X carriage to the half-way point instead of right at the edge of the far side.
This is with no other changes than syncing my enlistment with bugfix-2.0.x

blackdevil8975, 2018-02-23:
So, I do not know why but I have to set the bed size in steps and not mm. So its not as big as your arpartment :D 

I think there is something wrong with the gcode file.
[right_handed_cutter.zip](https://github.com/MarlinFirmware/Marlin/files/1752438/right_handed_cutter.zip)

Any suggestions why I have to set my bed size in steps?

thinkyhead, 2018-02-23:
@blackdevil8975 You should set your bed size in mm, and then make sure you have the correct settings for `DEFAULT_AXIS_STEPS_PER_UNIT`. Your XY values are currently 80. This indicates 80 steps per millimeter, which is equivalent to 1.8° steppers with 16x microstepping and GT2 belts. If that's your hardware then those values are correct.

If you have 32x microstepping on XY, double it. If you have 0.9° steppers, double it again.

Since your X only goes halfway it sounds like your values should be 160.

After modifying your settings be sure to use `M502` and `M500` to used the new values and set them in EEPROM for the next boot.

Somehow your G-code has the object too large and way off the edge of the bed. Your slicer may be drunk and lacking sleep!

[image]

```
The print goes:
- from 0.00 mm to 238.61 mm in X and is 238.61 mm wide
- from 0.00 mm to 255.41 mm in Y and is 255.41 mm deep
- from 0.00 mm to 15.00 mm in Z and is 15.00 mm high
```
I've seen this kind of thing before. You'll need to make sure your slicer is configured properly and make sure the object is centered on the bed when you slice it.

xC0000005, 2018-02-23:
I think your reply has told me what's wrong with my build as well (though I didn't change it that I know of).

Thanks.

blackdevil8975, 2018-02-23:
Thanks for the answer.

The problem with the x going halfway was @xC0000005 problem, but not mine :)

So I got 1.8° Stepper and 16x microstepping and im using a M8 threaded road so I set the steps to 2560. I took the information from this site: http://reprap.org/wiki/Step_rates/de

The Z axis isnt currently in use.

Now I have another question. I set the jumpers on the Ramps board for x16 microstepping. 
Do I need any configuration change in Marlin to activate microstepping?

thinkyhead, 2018-02-23:
> Do I need any configuration change in Marlin to activate microstepping?

Nope! With 8x microstepping each step sent by Marlin is 1/8 of a step automatically, and with 16x microstepping each step sent by Marlin is 1/16 of a step, etc.

closed 2018-02-23 by thinkyhead

blackdevil8975, 2018-02-23:
Do my Steps/mm seem ok for my setup?

Do you have a explanation why my stepper motors arent as strong anymore and arent that fast?
I mean they cant even move the whole thing now

thinkyhead, 2018-02-23:
There are standard things to check. If they lack torque it's probably the stepper driver current. Search YouTube for videos on how to adjust it.

If the steppers won't move as fast as you command them, then possibly the max feedrate is holding them back. Try increasing it from the LCD "Control" menu.

Remember that feedrates are expressed in mm/m, so `G1 X100 F3000` means 50mm/s.

Do they now move 100mm when you tell them to move 100mm? Then your steps-per-mm is ok.

If they try to move but sit and grind, possibly your feedrate or acceleration is too high.

Adjust, calibrate, adjust, tighten, and adjust and calibrate some more.

blackdevil8975, 2018-02-23:
The feedrate was too high, thanks for the answer :)