MarlinFirmware/Marlin #9777: K-factor calibration generates bad G-code

v1ru3, 2018-02-23:
[image]

bed is set to 300x300mm

Sineos, 2018-02-23:
I cannot reproduce this. Can you paste here the complete settings block from top of the gcode file please.

v1ru3, 2018-02-24:
; ### Marlin K-Factor Calibration Pattern ###
; -------------------------------------------
;
; Created: Fri Feb 23 2018 11:38:13 GMT-0500 (Eastern Standard Time)
;
; Settings Printer:
; Filament Diameter = 1.75 mm
; Nozzle Diameter = 0.6 mm
; Nozzle Temperature = 220 Â°C
; Bed Temperature = 40 Â°C
; Retraction Distance = 5 mm
; Layer Height = 0.2 mm
; Z-axis Offset = 0 mm
;
; Settings Print Bed:
; Bed Shape = Rect
; Bed Size X = 300 mm
; Bed Size Y = 300 mm
; Origin Bed Center = false
;
; Settings Speed:
; Slow Printing Speed = 2400 mm/min
; Fast Printing Speed = 7200 mm/min
; Movement Speed = 7200 mm/min
; Printing Acceleration = 500 mm/s^2
; Jerk X-axis = 15
; Jerk Y-axis = 15
; Jerk Z-axis = -1
; Jerk Extruder = -1
;
; Settings Pattern:
; Linear Advance Version = 1.5
; Starting Value Factor = 250
; Ending Value Factor = 400
; Factor Stepping = 0.2
; Test Line Spacing = 5 mm
; Test Line Length Slow = 40 mm
; Test Line Length Fast = 120 mm
; Print Pattern = Standard
; Print Frame = false
; Number Lines = true
; Print Size X = 210 mm
; Print Size Y = 3775 mm
; Print Rotation = 0 degree
;
; Settings Advance:
; Nozzle / Line Ratio = 1.05
; Use BL = false
; Extrusion Multiplier = 1
; Prime Nozzle = true
; Prime Extrusion Multiplier = 2.5
; Dwell Time = 2 s
;
; prepare printing
;

another thing i noticed is that if i dont define custom jerk settings and go with -1 seems to be ok but, when actual print starts the movement is slow and very segmented, like a line goes 0.3mm stops for a few nanoseconds and then moves again and so on.

Sineos, 2018-02-24:
[image]
This warning above is what I get when using your settings.

> ; Starting Value Factor = 250
> ; Ending Value Factor = 400
> ; Factor Stepping = 0.2
> ; Print Size X = 210 mm
> ; Print Size Y = 3775 mm

Also see the settings summary. With a K-Range of 400 - 250 = 150 and a stepping of 0.2 the script generates 750 test lines. This results in a Y print size of 3775mm

For `; Linear Advance Version = 1.5` you typically want to test a range between 0 and 2 with a stepping of 0.2. Maybe between 2 and 4 for Bowden setups.

`; Linear Advance Version = 1.5` is only applicable for the current development branches of Marline, i.e. bugfix-1.1.x. and bugfix-2.0.x.

> another thing i noticed is that if i dont define custom jerk settings and go with -1 seems to be ok but, when actual print starts the movement is slow and very segmented, like a line goes 0.3mm stops for a few nanoseconds and then moves again and so on.

This is a real bug and meanwhile fixed.

closed 2018-02-24 by v1ru3