MarlinFirmware/Marlin #9778: [bug-fix-2.0.x] TMC-CS Pin not found in configuration_adv.h

dembach, 2018-02-23:
cannot compile.
using delta default config and altered with : 
```
#define HAVE_TMC2130
#define X_IS_TMC2130
#define Y_IS_TMC2130
#define Z_IS_TMC2130
```

PINS are defined in my board file... (as usual)
but if i try to compile i encounter following error:
```
in file included from sketch/src/inc/MarlinConfig.h:34:0,
                 from sketch/src/Marlin.h:25,
                 from sketch/src/Marlin.cpp:31:
sketch/src/inc/SanityCheck.h:1529:6: error: #error "X_CS_PIN is required for X_IS_TMC2130. Define X_CS_PIN in Configuration_adv.h."
     #error "X_CS_PIN is required for X_IS_TMC2130. Define X_CS_PIN in Configuration_adv.h."
      ^
```

is it work in progress to alter the assignment of the CS-Pins ?

dembach, 2018-02-23:
definition file for pins is wrong... altered RAMPS_FD_v1 as example in commit 0809929

teemuatlut, 2018-02-23:
It should be `#ifndef X_CS_PIN`.
Basically a typo then.

dembach, 2018-02-23:
but there is no X_CS_PIN definition in the configuration files... right ?

teemuatlut, 2018-02-23:
No not yet.

dembach, 2018-02-23:
?_IS_TMC2130 is already there.... and we have to define it.... so why not use that ?

teemuatlut, 2018-02-23:
It was decided that the CS pin definitions will move to `configuration_adv.h` but with `pins_` files providing a board specific default value. It would cause a redefinition error if the `pins_` file would define a CS pin even when it was already defined in `configuration_adv.h`.

dembach, 2018-02-23:
But still you have got to alter the pins_ file if you need pins that are already in use...
i.e. RAMPS_FD file is using CS Pins that i cannot find on the board ;-) 
so i have to wait again to get things working completely....

teemuatlut, 2018-02-23:
No, `configuration_adv.h` will have a section for defining the CS pins. However, if you don't do that, Marlin will use the **default** values from the respective `pins_` file.

thinkyhead, 2018-02-23:
Patched.

commit e0227f21 (2018-02-23): Fix *_CS_PIN for RAMPS_FD

Fix #9778
closed 2018-02-23 by thinkyhead