MarlinFirmware/Marlin #9781: [bugfix-1.1.x] Case Light and Controller FAN not turning off

ZoCK3R1608, 2018-02-23:
Hi,

I enabled CASE Light, to digital pin 40, it switches a relay on and off. I disabled light on after power up.

When I power printer, lights turn on and off and on, if I manually disable light in lcd menu, light is turning off and then on again in just one second.

Second thing, after printing Controller FAN not turning off, insted it turns off in a second and then turning on again, this repeats infinite till I reset controller.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1752947/Marlin.zip)

thinkyhead, 2018-02-23:
First thoughts:

`CASE_LIGHT` requires a PWM pin. Can your relay handle that?

And, is pin 40 PWM-able on RAMPS? I forget.
Enable `PINS_DEBUGGING` and then use `M43` to get a pin map.

ZoCK3R1608, 2018-02-23:
Hi,

thanks for response. This configuration works well with stable Marlin 1.1.x. I used Bugfix Branch because I cannot get my TMC2130 working with stable Marlin.

I used AUX2 Pins on RAMPS1.4. I set PWM to 255, so only ON and OFF.

thinkyhead, 2018-02-24:
> I used Bugfix Branch because I cannot get my TMC2130 working with stable Marlin.

_If it's not one thing…_

Your pins sound sane, and it's good that it's known to work with 1.1.8. I could scan the relevant code looking for something to jump out, but that tends to be slow and laborious. Plus, the cause may be disconnected from these features.

Can you disable pretty much everything and do a flash, with just one of these features on at a time, just to see if any other enabled features could be interacting with these pins or with the timers associated with these pins? That might lead to a signpost.

ZoCK3R1608, 2018-02-25:
Hello thanks for response.

I redownloaded Marlin-1.1.x (stable) and reconfigured everything only neccessary things. X and Y are fine. Z makes very loud noise, move more mm than it should. Sometimes Firmware stopped working, if I move Z often. It showed same message on LCD Display as
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1756587/Marlin.zip)
 I pressed killswitch near Encoder knob.

ZoCK3R1608, 2018-02-25:
I measured Z-Axis offset and recalculated steps per mm and adjusted... now its working properly and smooth with stable Marlin.

It is strange I need to reduce steps per mm from 1600 to 800. I am using Marlin with this printer since 2014 and never need to adjust steps per mm in various versions.

thinkyhead, 2018-02-28:
A change from 1600 to 800 without any change to hardware indicates that most likely your micro-steps for Z have been reduced by 1/2. It could indicate a short between the jumper pins underneath the stepper driver, or some other fault in the stepper driver. Hard to say.

- What is your Z micro-stepping?
- What kind of stepper motor? (1.8° or 0.9° per full-step aka 200 or 400 steps-per-revolution)
- What is the pitch of your Z leadscrew?

closed 2018-02-28 by thinkyhead

ZoCK3R1608, 2018-02-28:
Hi there,

pitch is about 1mm, I use TMC2130 stepper driver with SPI connection. 16µSteps with interpolation to 256 steps.
Stepper motors has 1.8° aka 200 steps per revolution.
I imported my config with a few changes from bugfix-1.1.x to stable marlin with a features enabled I wanted and now I must revert to 1600 steps, but now for Y axis I need reduse steps from 80 to 40... Its very weird, I leave it now, because printer working properly.

On my Anet A8 with Trigorilla Board I have similar setup (4x TMC2130 with SPI, 16->256 steps) and with many features activated too and there is all working properly without modifying steps per axis.

thinkyhead, 2018-02-28:
For your Z the correct value should be 200 * 16 = 3200 step-pulses-per-revolution. If the pitch of the Z rod is 1mm that gives you **3200** steps-per-mm. If the pitch of the Z rod is 2mm you get **1600** steps-per-mm. And if the pitch of the Z rod is 4mm you get **800** steps-per-mm.

If you rotate the Z rods one full revolution, the Z axis should raise by 4mm given the number of steps-per-mm you've entered into the firmware.

Just to be sure that your configuration is taking effect, you should use `M502` and `M500` to apply your changes and write them to the EEPROM.

ZoCK3R1608, 2018-02-28:
I know eeprom settings to read write and so, I used them many times :-(

Z must then a 2mm pitch thing, I ordered it from a china shop. Description must be wrong. I measured all my axis with a calliper to be sure. Test print cube is near exactly 20mm, so I think setting is correct.

thinkyhead, 2018-02-28:
How many "starts" are there on the leadscrew?
[image]

ZoCK3R1608, 2018-02-28:
I dont want to put out this leadscrew, but I can see only one, not sure.

thinkyhead, 2018-03-01:
You can test by using a magic marker. Stick the end into one of the threads and rotate the rod. If there are multiple starts, there will be gaps where the threads have been marked.