MarlinFirmware/Marlin #9782: [bugfix_2.0.x]Due not starting up anymore

dembach, 2018-02-23:
unchanged config from 04.02.2018 but something happened that my due is not starting with the bugfix_2.0.x of today anymore.
disabled all features that may interfere with advanced features like TMC2130, SPI-EEPROM and so on... but still not starting up anymore, even without Shield.

attached the config
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1753052/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1753054/Configuration.txt)

thinkyhead, 2018-02-23:
Before each build, delete `.pioenvs` and `.piolibdeps` beforehand. Try the current `bugfix-2.0.x` first. Then try some earlier ones, starting with this from a day or two ago:

https://github.com/MarlinFirmware/Marlin/archive/a6b7e721cfe8876cd4ff262cb325983e8be1274b.zip

dembach, 2018-02-23:
@thinkyhead : 
tested both..... not working....
reverted back to 04.02.2018, working.... no config changes...working

cjsoong, 2018-02-24:
@dembach 
I use DUE + RAMPS-FD V2 + marlin 2.0x 20-02-2018 in delta mode no problem.
Do you have to re-edit the new config file? Maybe the config file version is too old
After upload firmwrec need to turn off the power re-boot, because sometimes i2c bus abnormal boot will be delayed for a long time.
process::
1. upload firmware
2. Power off
3. re power on
4. init EEPROM,
5. Restore failsafe
6. store setting

thinkyhead, 2018-02-24:
The second commit I linked above was from 21-02-2018 so it is remotely possible that a version from the 19th or 20th will be ok and something did get broken. Let's try one from the 19th and see if we end up with a working Due.

https://github.com/MarlinFirmware/Marlin/archive/6e932f979d57350125d79d76e525cd005ba142f0.zip

dembach, 2018-02-24:
@thinkyhead : that´s 1.1 you have sent in your last post...

Bob-the-Kuhn, 2018-02-24:
It works if you disable **I2C_EEPROM** in **pins_RAMPS_FD_V2.h** and **PRINTCOUNTER** in **Configuration.h**.

I'll let you know when I know more.

dembach, 2018-02-24:
what a crap..... but thanks @Bob-the-Kuhn

Bob-the-Kuhn, 2018-02-24:
It's just PRINTCOUNTER.  

Seems to be hanging in the TWI bus routines within the DUE Platformio tool chain.  But that could be just because I don't have an I2C EEPROM.  Please post your working Marlin & I'll see if it hangs in the same place.  If it doesn't then I can do some more investigation.

dembach, 2018-02-24:
one minute.....

Bob-the-Kuhn, 2018-02-24:
As is it ran.  When I enabled **I2C_EEPROM** in **pins_RAMPS_FD_V2.h** it hung in the same place.

Should **I2C_EEPROM** be enabled?

thinkyhead, 2018-02-25:
I guess it should only be enabled if there is a real I2C EEPROM. The pins file implies it's a built-in feature. Good possibility there's a bug in our handling. Remotely possible it's a bug in the toolchain.

thinkyhead, 2018-02-25:
Is it dying right away on `Wire.begin` or later, on `eeprom_update_block`?

thinkyhead, 2018-02-25:
Try this change to `I2cEeprom.cpp`:
```diff
- for (byte c = 0; c < n; c++) {
-   if (Wire.available()) eeprom_temp[c] = Wire.read();
-   flag |= (eeprom_temp[c] ^ *((uint8_t*)pos + c));
- }
+ uint8_t *ptr = (uint8_t*)pos;
+ for (byte c = 0; c < n && Wire.available(); c++)
+   flag |= Wire.read() ^ ptr[c];
```
It looks like the current implementation is trying to de-reference the value at `pos + c` as if it was a pointer, rather than just taking the byte.

Bob-the-Kuhn, 2018-02-25:
I know for sure that it's dying in TWI_ConfigureMaster(twi, twiClock, VARIANT_MCK); in Wire.begin

That's proably because I don't have an I2C EEPROM to test with.

thinkyhead, 2018-02-25:
Do we get a prize if we find a bug in the library?

dembach, 2018-02-25:
so there was a price for @Bob-the-Kuhn already... but he decided to get a RAMPS_FD 2.1 for himself, though there was the invitation to a 2.2 by me ;-) 
So i guess there will be more effort for @Bob-the-Kuhn to get it running, due to the RAMPS_FD 2.1 has got I2C EEPROM ....
so it is the same again like #9484 ... we should close one of these..... just wondering why the version of 04.02.2018 is booting ...

Bob-the-Kuhn, 2018-02-25:
@dembach - I have a couple if I2C EEPROMS coming.  I expect it'll be late this week before I can debug this problem.

dembach, 2018-02-25:
@Bob-the-Kuhn where are you from ?

Bob-the-Kuhn, 2018-02-25:
I'm in Austin Texas.

dembach, 2018-02-25:
ahh... Digikey delivers within one day to US land....  ;-)

got also I2C Eproms here.... very easy to install.... one side is only GND , other side the communication pins. so just the eeprom itself is needed... no other components.... just too easy...

thinkyhead, 2018-02-26:
Duplicate of #9484

closed 2018-02-26 by thinkyhead

Bob-the-Kuhn, 2018-03-04:
PR #9937 fixes this issue.