MarlinFirmware/Marlin #9787: Temp ERR>>>>

S1CAR1US, 2018-02-24:
I am getting tired of getting the thermal runaway error or min temp error for no reason and everytime it just kills the print!  How can there not be a solution by now for resuming a print when there is not even a power loss?  Every time I have to re home, then re slice the file but have it start at the current z pos.  set z with G92 and G90... its ridiculous.  All of this because of false errors.  Here is my Temp Curve to show that there is indeed never a thermal runaway... Just like in other instances, there is never a min temp reached.
[image]
[image]

thinkyhead, 2018-02-24:
Try relaxing the settings until they stop triggering.

psavva, 2018-02-25:
@thinkyhead, we see this all too often, and always with the same outcome to relax the thermal runaway settings.  Perhaps we should again consider relaxing them in the default configs...

thinkyhead, 2018-02-25:
It appears the majority of users don't experience this, so the settings are good for most setups. Plus we do repeat that if you get lots of runaway errors it's ok to relax (but not disable) these settings.

But let's keep collecting data from people experiencing problems and see what range of values makes the most sense.

There are also some who report that lowering the `MAX_PID` values for the nozzle and bed by a little bit (and then re-tuning with `M303 U`) are getting more stable temperatures.

thinkyhead, 2018-02-26:
Duplicate of #9730

closed 2018-02-26 by thinkyhead

batata004, 2018-02-26:
I am having exactly this same issue,  @thinkyhead  suggstion will probably not work, relaxing settings will cause not effect cause I already relaxed it a lot and randomly I still get thermal errors (once every 6 or 7 prints).

AnHardt, 2018-02-27:
The usual reasons for 'Thermal Runaways' and 'Heating failed' problems are
* underpowerd heaters
* part cooling fans, cooling the nozzle or bed
* noisy sensors
* untuned or bad tuned parameters of the safety system and/or the PIDs

In the second picture you can see the heater puting 'all in' for ~40s, but the temperature still dropping.

batata004, 2018-02-27:
@thinkyhead  I already said, I am not a C programmer. Also I am not demanding anything, I am just saying that marlin is getting on each version more unstable and slow. Anyone who uses marlin in a day to day basis with arduino can easily see the instability and have lots of frustration.

I WILL BE CLEAR: sometimes I turn on the printer and EVEN TO SPIN THE KNOB to control de LCD gets slow! I turn it off, turn on again and I can spin the KNOB very fast with no problem.

The temperature problem is the same: sometimes I turn on the printer and start heating it and it works fine. Other times the temperature spikes 10 or 15 degrees in a matter of 1 second (OF COURSE IN REALITY THE TEMPERATURE IS NOT SPIKING 10 or 15 DEGREES, it has sometihng to do with MARLIN being too slow, busy...whih causes the reading of the sensor to be wrong).

@AnHardt I am having none of those problems,

thinkyhead, 2018-02-28:
@batata004 To help you with an issue, you must fill out the form provided in the New Issue window, follow our instructions to collect data, and help narrow down the issue. We aren't here to hold hands, we are here to track down and fix things that are within our control. We have never established with you where the cause lies.

We are not here to entertain complaints. Complain on your blog or to your co-workers. We are all doing the best we can, and we work hard to make improvements. If you want bug-free software, you will find it underneath a log behind the shed, just left of the big rock.

thinkyhead, 2018-02-28:
@batata004 — Don't try and get familiar with me. I'm not your buddy.

The only correct answer is to apologize and stop adding noise. Yes, I'm having a tough day dealing with a handful of obnoxious people like yourself who don't seem to have any self-awareness or know when to stop. Keep it up and you will be banned from this project. This is hardly the first time we have noticed your tendencies.

batata004, 2018-03-01:
@thinkyhead FUCK YOU BITCH. You are being very disrespectful. I NEVER OFFENDED YOU. I NEVER TOLD YOU TO DO ANYTHING.

You on the other hand keeping acting like if I were your bitch. If you want to be respected, RESPECT OTHERS. I never, EVER, disrespected you. 

If I get banned I dont care, what will happen? I wont be able to report any bug anymore? You are saying that as if you ever solved a bug/problem that I reported - which you never did. So yeah, go ahead, show the bitch you are.

emartinez167, 2018-03-01:
Wow, talk about some people having an attitude of self entitlement...

Regards,

Ernesto.

emartinez167, 2018-03-01:
@batata004, piss off. Take your attitude somewhere else.

Regards,

Ernesto.

thinkyhead, 2018-03-01:
Well, given what we already know, that was expected. Good riddance.