MarlinFirmware/Marlin #9788: Marlin 1.1.8 + MKS Gen v1.4, PS_ON pin is inverted.

Pyro-Fox, 2018-02-24:
### Bug Report
- Description:Marlin 1.1.8 + MKS Gen v1.4, PS_ON pin is inverted.
- Expected behaviour: M80 should close the circuit/Turn PSU on, M81 should open the circuit/Turn PSU off
- Actual behaviour: M80 Turns off PSU, M81 Turns it on
- Steps to reproduce:
   -Plug ATX on pins to Pin 4 (physically labeled Servos2 , Pin6) and Ground
   -Turn on Physical power switch... Nothing happens
   -Send M80... nothing happens
   -Send M81... printer turns on.

I used my multimeter and confirmed the circuit is open when I send M80 and closed when I send M81.

All my firmware files are attached (so you can see the pin config files too):
[Pyro-Fox Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1754022/Pyro-Fox.Marlin.zip)

AnHardt, 2018-02-24:
Try
```
#define POWER_SUPPLY 2
```
This option is not ideally named, but it's purpose is to invert the pin.

thinkyhead, 2018-02-28:
As seen here. It's always been like this. Your "ATX" is just doing the opposite thing. Sometimes it means the wires are swapped. Sometimes it means the manufacturers are creative.
```cpp
  // Power Signal Control Definitions
  // By default use ATX definition
  #ifndef POWER_SUPPLY
    #define POWER_SUPPLY 1
  #endif
  #if (POWER_SUPPLY == 1)     // 1 = ATX
    #define PS_ON_AWAKE  LOW
    #define PS_ON_ASLEEP HIGH
  #elif (POWER_SUPPLY == 2)   // 2 = X-Box 360 203W
    #define PS_ON_AWAKE  HIGH
    #define PS_ON_ASLEEP LOW
  #endif
```

Pyro-Fox, 2018-02-28:
I haven't had a chance to get my printer out of it's enclosure to test AnHardt's suggestion yet, but I can tell you the behavior is reversed from what I experienced in 1.1.3. For ATX, closing the PS_ON pin to ground should turn on the power supply. However, M80 is opening the circuit instead.

thinkyhead, 2018-03-01:
This code hasn't changed in a dog's age. I look forward to seeing the results from your next round of testing!

thinkyhead, 2018-03-01:
I don't have an answer. The code appears to be perfectly fine, and we have not had a flood of reports from people whose PSUs are turning off when they should be turning on. For ATX we have always had it set so that the PSU is turned on with a LOW pin state.

#### Conditionals
```cpp
#if (POWER_SUPPLY == 1)     // 1 = ATX
  #define PS_ON_AWAKE  LOW
  #define PS_ON_ASLEEP HIGH
#else
   . . .
#endif
```
```cpp
#if ENABLED(AUTO_POWER_CONTROL)
  #define PSU_ON()  powerManager.power_on()
  #define PSU_OFF() powerManager.power_off()
#else
  #define PSU_ON()  OUT_WRITE(PS_ON_PIN, PS_ON_AWAKE)
  #define PSU_OFF() OUT_WRITE(PS_ON_PIN, PS_ON_ASLEEP)
#endif
```
#### M80
```cpp
  void GcodeSuite::M80() {
    . . .
    PSU_ON();
```
#### Power manager
```cpp
void Power::power_on() {
  lastPowerOn = millis();
  OUT_WRITE(PS_ON_PIN, PS_ON_AWAKE);
}

void Power::power_off() {
  OUT_WRITE(PS_ON_PIN, PS_ON_ASLEEP);
}
```

Pyro-Fox, 2018-03-02:
AnHardt's suggestion worked. Setting my PSU settings to Xbox 360 (Type 2) fixed the reversed logic issue. I guess my HP DPS-750RB power supply is just a special snowflake. Thanks for the help!

closed 2018-03-02 by Pyro-Fox

Pyro-Fox, 2018-04-26:
No modifications. Digital Pin 4 is defined in **pins_MKS_13.h** by default (in the latest versions of Marlin at least).

**Lines 44-52:**
```C++
//
// PSU / SERVO
//
// If POWER_SUPPLY is specified, always hijack Servo 3
//
#if POWER_SUPPLY > 0
  #define SERVO3_PIN      -1
  #define PS_ON_PIN        4
#endif
```

BSODOverride, 2018-04-26:
Right, I have that. Then below that It includes the "pins_RAMPS.H" file, which then has PS_ON_PIN set to 12. Do I need to change that to 4 to match?

Pyro-Fox, 2018-04-26:
No. The code in the **pins_RAMPS.h** file says:
```C++
#ifndef PS_ON_PIN
  #define PS_ON_PIN        12
#endif
````
"ifndef" means "If not defined".

That means, if no PS_ON pin has been defined, THEN use pin 12. Since it's already defined, those lines are skipped.