MarlinFirmware/Marlin #9791: TMC2130, Ramps 1.4, Marlin, pronterface-> X not homing

steverichab, 2018-02-24:
Hello from a newbie to 3D printing,

I have recently converted an old original Wanhao DI3 to Ramps 1.4 w/ LCD and it has been working fine with DRV8825 stepper drivers.

After watching Toms "Dolly Upgrade" video on TMC2130 stepper drivers I got 3 for my X/Y/Z axis's
I got the cheep Fysetc drivers but they appear to work fine after I soldered the proper jumpers.

I have gotten all axis working through Pronterface and even the LCD is fine via SPI.

X Axis limit and Y axis limit are wired to the diag pin on the TMC2130s but the Z limit is wired to a mechanical switch.
note: M119 reports back open always on X&Y and triggered on Z when it is home.

**All seems well but 2 things do not work 1) when I hit XHome on Pronterface the motor only shutters. 2) M122 returns nothing**. I also do not know if sensor less homing is working but first things first.

Here is a summary of my configuration.h and I have modified the configuration.adv.h and pins.ramps files

Please Help !!!

//**************************************************************************************************************************************

#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010100
#define SERIAL_PORT 0
#define BAUDRATE 250000

#ifndef MOTHERBOARD
#define MOTHERBOARD BOARD_RAMPS_14_EFB
#endif
#define CUSTOM_MACHINE_NAME "TMC2130 X Home Broken"
//===========================================================================
//======================== Extruder =======================
//===========================================================================
#define EXTRUDERS 1
#define POWER_SUPPLY 0
#define PREVENT_COLD_EXTRUSION
#define EXTRUDE_MINTEMP 170
// This option prevents a single extrusion longer than EXTRUDE_MAXLENGTH.
// Note that for Bowden Extruders a too-small value here may prevent loading.
#define PREVENT_LENGTHY_EXTRUDE
#define EXTRUDE_MAXLENGTH 200
#define INVERT_E0_DIR false
//===========================================================================
//======================== Thermal Runaway Protection =======================
//===========================================================================

#define THERMAL_PROTECTION_HOTENDS // Enable thermal protection for all extruders
#define THERMAL_PROTECTION_BED // Enable thermal protection for the heated bed

//===========================================================================
//============================== Endstop Settings ===========================
//===========================================================================
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define X_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the probe.
#define INVERT_X_DIR true 
#define INVERT_Y_DIR false 
#define INVERT_Z_DIR false
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

//=============================================================================
//============================== Movement Settings ============================
//=============================================================================
//************************************************************************************************************
//#define DEFAULT_AXIS_STEPS_PER_UNIT { 159.86, 159.8, 799.09, 190 }
// These settings are for DRV8825 stepper drivers at 1/32 steps/mm cut in half if A4988 drivers @ 1/16 steps/mm
#define DEFAULT_AXIS_STEPS_PER_UNIT { 79.95, 79.9, 399.55, 190 }
// These settings are for TMC2130 stepper drivers at 1/16, E0 DRV8825 drivers @ 1/32 steps/mm
//*************************************************************************************************************
//*************************************************************************************************************
/
// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 170
#define Y_MAX_POS 170
#define Z_MAX_POS 170
#define X_ENABLE_ON 0
#define Y_ENABLE_ON 0
#define Z_ENABLE_ON 0
#define E_ENABLE_ON 0 // For all extruders
```
One more thing if I use the correct DEFAULT_AXIS_STEPS_PER_UNIT { 79.95, 79.9, 399.55, 190 } the Y Home behaves the same way as the X Home 
If I use the original DEFAULT_AXIS_STEPS_PER_UNIT { 159.86, 159.8, 799.09, 190 } only the x home behaves that way

FYI:  G28 X0 in pronterface does not go home either.

teemuatlut, 2018-02-24:
M122 requires `TMC_DEBUG`
Try the bugfix-1.1.x branch as 1.1.8 has a bug that affects communication to the drivers.
`SENSORLESS_HOMING` may require some tuning with `M914` but post your `M122` output first so we can see if the communication is working alright.

steverichab, 2018-02-24:
Thank you I will try the bug fix branch.
Unfortunately there are  2 problems with my installation:
1) when I hit X Home on Pronterface the motor only shutters. 
2) M122 returns nothing.
I think there is a way to enable TMC_DEBUG in configuration.adv.h but I am not sure how to do it.

As soon as I get M122 working I will post its output

Thanks Again

steverichab, 2018-02-24:
I have gotten the bugfix-1.1.x loaded and the M122 is now operational and the X Home is working. 

Currently the SPI is the only thing not working.  When I go back to the non bug fix then the spi works again MOSI=> pin 51  MISO=> pin 50 SCK=> pin 52. 

Sensor less homing doesn't appear to work well so I would appreciate any advice. on both SPI and Sensor less homing 

SENDING:M122
		X	Y	Z
Enabled		true	true	false
Set current	800	800	800
RMS current	795	795	795
MAX current	1121	1121	1121
Run current	25/31	25/31	25/31
Hold current	12/31	12/31	12/31
CS actual		12/31	12/31	12/31
PWM scale	46	53	36
vsense		1=.18	1=.18	1=.18
stealthChop	true	true	true
msteps		16	16	16
tstep		1048575	1048575	1048575
pwm
threshold		0	0	0
[mm/s]		-	-	-
OT prewarn	false	false	false
OT prewarn has
been triggered	false	false	false
off time		5	5	5
blank time	24	24	24
hysterisis
-end		2	2	2
-start		3	3	3
Stallguard thrs	0	0	0
DRVSTATUS	X	Y	Z
stallguard
sg_result		0	0	0
fsactive
stst		X	X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:	X = 0x80:0C:00:00
	Y = 0x80:0C:00:00
	Z = 0x80:0C:00:00
>>> M119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open

teemuatlut, 2018-02-25:
Not entirely sure what you mean with "SPI not working" when you also show drivers getting correctly configured and read from through the use of SPI.

thinkyhead, 2018-02-25:
> M119 reports back open always on X&Y

`M119` doesn't apply to `SENSORLESS_HOMING` because the stallguard state is transient.

TimotF, 2018-02-25:
I've got the same issue here. I just replaced all my drivers with TMC2130 and they I can't get them working properly.

**What I can do :**

- communicate with every TMC2130
- have the X and the Y axis moving

**What I can't do :**

- have any other motor moving (they don't move, keeping their position whatever the current supplied)
- X, Y axis homing (when I run G28 X0 for example, the X motor doesn't move and does the exact same thing as the other motors (Z, E0, E1))

M122 result :
```
SENT: M122
READ: 		X	Y	Z	E0	E1
READ: Enabled		false	false	false	false	false
READ: Set current	800	800	800	800	800
READ: RMS current	795	795	795	795	795
READ: MAX current	1121	1121	1121	1121	1121
READ: Run current	25/31	25/31	25/31	25/31	25/31
READ: Hold current	12/31	12/31	12/31	12/31	12/31
READ: CS actual		12/31	12/31	12/31	12/31	12/31
READ: PWM scale	39	21	25	2	2
READ: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
READ: stealthChop	true	true	true	true	true
READ: msteps		16	16	16	16	16
READ: tstep		1048575	1048575	1048575	1048575	1048575
READ: pwm
READ: threshold		0	0	0	0	0
READ: [mm/s]		-	-	-	-	-
READ: OT prewarn	false	false	false	false	false
READ: OT prewarn has
READ: been triggered	false	false	false	false	false
READ: off time		5	5	5	5	5
READ: blank time	24	24	24	24	24
READ: hysterisis
READ: -end		2	2	2	2	2
READ: -start		3	3	3	3	3
READ: Stallguard thrs	8	8	8	0	0
READ: DRVSTATUS	X	Y	Z	E0	E1
READ: stallguard						
READ: sg_result		1023	38	391	0	0
READ: fsactive						
READ: stst		X	X	X	X	X
READ: olb						
READ: ola						
READ: s2gb						
READ: s2ga						
READ: otpw						
READ: ot						
READ: Driver registers:	X = 0x80:0C:03:FF
READ: 	Y = 0x80:0C:00:26
READ: 	Z = 0x80:0C:01:87
READ: 	E0 = 0x80:0C:00:00
READ: 	E1 = 0x80:0C:00:00
```
I have tested with version 1.1.8 and bugfix-1.1.x but the result seems to be the same...

thinkyhead, 2018-02-26:
@teemuatlut — Apparently TMC drivers are becoming popular stocking-stuffers. These issues are piling up and generally after a few helpful words (usually from you) everyone eventually just figures it out on their own. Should we write up a step-by-step guide to help get these things configured correctly and troubleshoot?

steverichab, 2018-02-26:
The bugfix-1.1.x branch did the trick on homing the x axis using Pronterface and/or the G28 X0 command, Thanks.  Unfortunately there are 2 issues to replace the non homing x axis issue.
- What  I mean by the SPI bus not working is that since I installed the bugfix-1.1x my 12864 LCD screen no longer works. If I return to the prior version the LCD works fine(without x homing).
- Sensor less homing appears not to be operating. When I hit the X or Y Home the print head travels to the end and keeps running for a few seconds and then finally decides to stop (very troubling). 

I love the step by step configuration idea @thinkyhead

thinkyhead, 2018-02-26:
Interesting. SPI is responsible for both the stallguard and the LCD. So the fact that they switch places in terms of what's working is … interesting.

teemuatlut, 2018-02-26:
Perhaps something like [this](https://github.com/teemuatlut/MarlinDocumentation/blob/TMC_section/_features/tmc_drivers.md)?

thinkyhead, 2018-02-26:
@teemuatlut — YES! Very nice.

teemuatlut, 2018-02-26:
@steverichab v1.1.8 definitely has a bug in it that affects SPI communication to the TMC2130 drivers on boot time.
If you don't have `MONITOR_DRIVER_STATUS` enabled, Marlin will **not** have any communication to the driver after initial configuration is pushed. Except for user commands of course.

Sensorless homing not working can be a multitude of things. Either the driver gets configured wrong for some reason. There's a hardware issue with either communication or the diag -> endstop pin connection. Or the most likely is that the threshold value isn't tuned. Try using `M914` to tune the threshold to work with your setup.
If that doesn't work, you can try running the stallGaurd sketch from the library examples. But please remove the belts first as the sketch will not respect hard stops or any end stop triggers. The example should output `0 <stallGuard_result> <actual_current>`. How stallGuard works should be very visible especially if you use the IDE plotter.

teemuatlut, 2018-02-26:
Einsy/Archim2 boards also use software SPI so they don't have to share any lines with other peripherals.

dcwalmsley, 2018-03-02:
Good Evening All,

I'm writing to this blog as I have run into similar issue with reference to X-Axis Home/Endstop.

Background:  I am a Noob when it comes to coding and compiling.  I have some experience in building custom printers and recently built CoreXY HyperCube which had been running the MiniRambo board on it until recent. I purchased the Ultimachine EINSY board (w/TMC 2130 drivers) and have it running to a point in place of the miniRambo.

1.  I have a Z-proximity Probe installed (Not PINDA)  and have it sensing the platform using "  #define Z_MIN_PROBE_PIN  10" in my "pins.EINSYRAMBO.h" section.  I also have successfully gotten Y Homing to work as well with "#define Y_MIN_PIN 69".  However, X Homing fails to move the extruder fully to the home position and stops short of where it should go.  At this point, I'm unclear as to what pin configuration I need it set to as I stumbled across someone's files and have it partially working.

I have zipped up my Config, Adv_Config, Pin.h, and pins.EINSYRAMBO.h files for you to review.  Appreciate any help working out this issue for me. 

2. I also found a couple files when dumped into the bugfix 1.1.8 files I'm currently attempting to use, seemed to fix the garbled LCD display problem I was having.  They are ultralcd_st7920_u8glib_rrd and ultralcd_st7565_u8glib_VIKI.  Not sure why they work, but hey, if they can help you all debug the LCD problem, have at it.

[Sample EINSYRAMBO Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/1773640/Sample.EINSYRAMBO.Configuration.Files.zip)

Appreciate any help with configuring the X-Min Pin number for this and any other tips.

Thanks for all you guys do.  I'm a hard fast Marlin fan!

R,
Doug W

thinkyhead, 2018-03-08:
Please the `bugfix-1.1.x` and `bugfix-2.0.x` branches using our `pins_EINSY_RAMBO.h` file and `BOARD_EINSY_RAMBO`. The 2.0.x branch also includes the LCD files, though currently they are commented out in `ultralcd_impl_DOGM.h`…
```cpp
/*
#if ENABLED(U8GLIB_ST7565_64128N)
  #include "dogm/ultralcd_st7565_u8glib_VIKI.h"
#elif ENABLED(U8GLIB_ST7920)
  #include "dogm/ultralcd_st7920_u8glib_rrd.h"
#endif
*/
```

dcwalmsley, 2018-03-09:
I must be lost.  I looked into the https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x/Marlin branch but was unable to find what you mentioned earlier (pins_EINSY_RAMBO.h) and BOARD_EINSY_RAMBO was not listed in "boards.h" file either.

thinkyhead, 2018-03-09:
Sorry, download `bugfix-1.1.x` again. The "Einsy" support had been added to the 2.0.x branch but was not yet added to 1.1.x.

closed 2019-07-06 by shitcreek