MarlinFirmware/Marlin #9792: [Configuration] Z-axis minimum ignored after homing

ArmyAg08, 2018-02-24:
### Bug Report
- Description: Printer is ignoring z axis minimum after homing

I have a FolgerTech 2020 aluminum kit I'm wanting to upgrade from their old firmware to the latest Marlin release so I can have access to new features as I upgrade my printer. I've attached the latest configuration.h that is giving me issues. The printer will home correctly to start, but as the nozzle moves to begin the print, the z-axis motors try to move the extruder below the minimum value even with the endstop being triggered. I've tried changing some settings with probing thinking that might be an issue (I don't have a probe installed yet). The latest was enabling Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN thinking the software would use the endstop as the probe and halt upon triggering it. I've also enabled MIN_SOFTWARE_ENDSTOPS but the printer seems to be ignoring that as well.

- Expected behaviour: The printer should not move below the z-minimum after homing.
- Actual behaviour: The printer is attempting to move below z-minimum after homing.
- Steps to reproduce:
  - Do this
  - Do that

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1754643/configuration.zip)

ArmyAg08, 2018-02-26:
I think I may have found the underlying issue. When issuing the G28 command to home all axes, X and Y home as expected. Z will also start to home correctly with the following steps.

1. Home quickly towards bed
2. Trigger endstop and stop movement
3. Probe moves away from bed
4. Fine tune movement towards bed
5. Trigger endstop and stop movement again
6. Probe moves away from bed again
7. M119 reports z-height as 13.4. Cura and Repetier show z-height as 0. Measured height is approximately 2mm as measured with calipers.

The next command is typically something like G1 Z5 to lift the nozzle to move the bed. Since it thinks the current height is 13.4, it moves down to 5 which crashes the nozzle into the bed.

thinkyhead, 2018-02-26:
What version of Marlin are you using?

ArmyAg08, 2018-02-26:
I am running 1.1.8. I will test with bugfix-1.1.x and report back.

ArmyAg08, 2018-02-26:
**bugfix-1.1.x**
After importing temperature settings and other machine specific variables, I was able to successfully home and complete a print. Success!

**1.1.8**
After I completed the home and test print above successfully, I ran a diff on the two configurations and moved all of those settings back to 1.1.8 to see what would happen. That version was also able to home and print a model successfully. I've attached the current working configuration.h and configuration_adv.h below, but the notable differences were the disabling of NUM_SERVOS, DEACTIVATE_SERVOS_AFTER_MOVE, NOZZLE_PARK_FEATURE, and ADVANCED_PAUSE_FEATURE. There could be an issue with something there.

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1760379/configuration.zip)

thinkyhead, 2018-02-26:
If there are issues with any of those features in 1.1.8 they will always be in 1.1.8. But fear not, 1.1.9 is due out very, very soon.

closed 2018-02-26 by thinkyhead