MarlinFirmware/Marlin #9794: [bugfix_2.0.x]Due/Radds Compile error

bwgeez02, 2018-02-24:
When compiling with Arduino 1.9.0 beta, to avoid filename ext error issue, I get this error.

Arduino: 1.9.0-beta (Windows 7), Board: "Arduino Due (Programming Port)"

In file included from sketch\src\module\thermistor/../../inc/../HAL/HAL_DUE/HAL_Due.h:36:0,

                 from sketch\src\module\thermistor/../../inc/../HAL/HAL.h:40,

                 from sketch\src\module\thermistor/../../inc/MarlinConfig.h:28,

                 from sketch\src\module\thermistor/thermistors.h:26,

                 from sketch\src\module\temperature.h:30,

                 from sketch\src\module\temperature.cpp:27:

sketch\src\module\thermistor/../../inc/../pins/pins_RADDS.h:143:31: error: pasting "/* BED*/" and "_WPORT" does not give a valid preprocessing token

 #define HEATER_BED_PIN      7 // BED

                               ^

sketch\src\module\thermistor/../../inc/../HAL/HAL_DUE/fastio_Due.h:75:33: note: in definition of macro '_WRITE'

   volatile Pio* port = (DIO ##  IO ## _WPORT); \

                                 ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:850:31: note: in expansion of macro 'WRITE'

   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)

                               ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:850:37: note: in expansion of macro 'HEATER_BED_PIN'

   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)

                                     ^

sketch\src\module\temperature.cpp:877:11: note: in expansion of macro 'WRITE_HEATER_BED'

           WRITE_HEATER_BED(LOW);

           ^

sketch\src\module\thermistor/../../inc/../pins/pins_RADDS.h:143:31: error: pasting "/* BED*/" and "_PIN" does not give a valid preprocessing token

 #define HEATER_BED_PIN      7 // BED

                               ^

sketch\src\module\thermistor/../../inc/../HAL/HAL_DUE/fastio_Due.h:76:31: note: in definition of macro '_WRITE'

   uint32_t mask = MASK(DIO ## IO ## _PIN); \

                               ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:850:31: note: in expansion of macro 'WRITE'

   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)

                               ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:850:37: note: in expansion of macro 'HEATER_BED_PIN'

   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)

                                     ^

[... 234 lines cut ...]

                               ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:850:37: note: in expansion of macro 'HEATER_BED_PIN'

   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)

                                     ^

sketch\src\module\temperature.cpp:1740:50: note: in expansion of macro 'WRITE_HEATER_BED'

         if (soft_pwm_count_BED <= pwm_count_tmp) WRITE_HEATER_BED(LOW);

                                                  ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1754761/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1754762/Configuration_adv.zip)
[pins_RADDS.zip](https://github.com/MarlinFirmware/Marlin/files/1754763/pins_RADDS.zip)

Bob-the-Kuhn, 2018-02-24:
Looks like Arduino doesn't like having comments after pin definitions on the non-analog pins.

Remove  **// BED** in your **pins_RADDS.h** file.

bwgeez02, 2018-02-24:
OMG. Thank you! Compiled fine after the deletion.

closed 2018-02-24 by bwgeez02

bwgeez02, 2018-02-25:
I'm pretty new to this. How do I report it?

thinkyhead, 2018-02-26:
Since it's a beta they're looking for bugs. Do what you did here, but at https://github.com/arduino/Arduino/issues

Say: "Comments at the end of "`#define`" lines are sometimes messing with the values. What's up with that? Show them your example from above. Paste a link to this issue.

Of course, search to see if an existing issue already exists.

bwgeez02, 2018-02-26:
Cool. Thanks!

thinkyhead, 2018-02-26:
Here's the relevant issue: https://github.com/arduino/Arduino/issues/5079