MarlinFirmware/Marlin #9796: ### Bug Report TMC2310 Driver Error Printer Haulted

GearDownForWhat, 2018-02-24:
I'm guilty of using this less as a bug report and more for help troubleshooting. I had my TMC2130's working fine, all I did was swap out my Z probe for a BL touch and now my printer just shows "Driver error Printer halted, please reset" 

I checked the wiring for the com's on the TMC2130s, that all looks good, so I tried disabling individually in the firmware, if I disable them all and have_tmc2130 it works, but the issue exists if any one of them in enabled. 

I can also get it to work if I comment out. "#define MONITOR_DRIVER_STATUS" but Im not sure what kind of side effects that is going to have

I turned on the diagnostics mode for the TMC2130's and got this, unfortunately it doesn't mean much to me, and theirs not a whole lot of info out there to help troubleshooting these new drivers.

------------------------- BEGINNING OF SERIAL MONITOR -----------------------------
start
echo:Marlin 1.1.7

echo: Last Updated: 2017-12-15 12:00 | Author: (none, default config)
echo:Compiled: Feb 24 2018
echo: Free Memory: 2572  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius

echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X200.00 Y200.00 Z800.00 E200.00
echo:Maximum feedrates (units/s):
echo:  M203 X100.00 Y100.00 Z3.00 E50.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z1000 E5000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.00 R3000.00 T1000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X8.00 Y8.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Auto Bed Leveling:
echo:echo:  M420 S0 Z10.00
echo:Material heatup parameters:
echo:  M145 S0 H190 B60 F255
echo:  M145 S1 H210 B90 F0
echo:PID settings:
echo:  M301 P19.31 I1.12 D81.35
echo:Z-Probe Offset (mm):
echo:  M851 Z-0.70
echo:Stepper driver current:
echo:  M906 X 400 Y 400 Z 400 E0 400
Error:STOP called because of BLTouch error - restart with M999
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
echo:SD init fail

Y driver error detected:
		X	Y	Z	E0
Enabled		false	false	false	false
Set current	400 	400 	400 	400
RMS current	397 	397  	397  	397
MAX current	560 	560 	560 	560
Run current	12/31	12/31	12/31	12/31
Hold current	0/31 	0/31 	0/31 	0/31
CS actual	0/31 	0/31 	0/31 	0/31
PWM scale	21    	26    	24	         6
vsense		1=.18 	1=.18	1=.18	1=.18
stealthChop	true 	true 	true 	true
msteps		32    	32    	32    	32
tstep		1048575	1048575	1048575	1048575
pwm
threshold		79 	79    	658	         263
[mm/s]		100.08	100.08	3.00	30.06
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time		5	5	5	5
blank time	24	24	24	24
hysterisis
-end		2	2	2	2
-start		3	3	3	3
Stallguard thrs	0	0	0	0
DRVSTATUS	X	Y	Z	E0
stallguard				X	
sg_result		0	0	0	0
fsactive					
stst		X	X	X	X
olb					
ola					
s2gb					
s2ga					
otpw					
ot					
Driver registers:
	X = 0x80:00:00:00
	Y = 0x80:00:00:00
	Z = 0x81:00:00:00
	E0 = 0x80:00:00:00

Error:Printer halted. kill() called!
 

------------------------- END OF SERIAL MONITOR -----------------------------

`Thank` you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuratio
[configurationzzz.zip](https://github.com/MarlinFirmware/Marlin/files/1755200/configurationzzz.zip)
n.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips

- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

teemuatlut, 2018-02-24:
First try checking if any of the pins conflict. BLTouch tests for Z_MIN_PIN or Z_MIN_PROBE_PIN.
You can also try if the same happens with bugfix-1.1.x.

Your TMC2130 communication seems to be working alright.

GearDownForWhat, 2018-02-24:
I checked both of those, don't see anything alarming, I did realize that I plugging the BLtouch servo into the wrong place, probably because the place that I was supposed to plug it in was being used for my chip select on my extruder0 TMC2130 but even that led me deeper into more problems 

Here it goes:

I went back to the drawing board and decided to reload the program that was know working, without the BL touch configured, that configuration gives me no error. 

Then I went ahead and re enabled the BL touch knowing that that configuration would cause a conflict on pin 11, because pin 11 is being used for my E0 stepper driver CS, but whats interesting is that I marlin tells me that the issue is with my Y axis stepper driver, but that driver isn't the one that should be having issues at all. 

I'm going to try to move the chip select pin for my E0 stepper driver and disable the BLtouch and see if I can get that config working right, then mess with the BLtouch. Ill get back to you.

GearDownForWhat, 2018-02-25:
Ok, I got everything working again!

Here is what I did wrong to save others from making the same mistake:

The BLtouch uses the same pins that I had configured my TMC2130's Chip Select to use, and just enabling the BL touch caused the TMC2130's to have issues.  Interesting enough, the message that I received on the serial interface was saying that there was an issue with the Y axis stepper driver, but the pin that the BL touch was interfering with was for the extruder, so that threw me for a loop.

The next issue was due to me literally plugging the BLtouch into the wrong spot on the ramps board, and getting unlucky enough to have it light up, causing me to think that I was plugged in properly, but just not configured right. This also caused me to not notice that the BL touch needed to use the pins assigned for my TMC2130.

I fixed the issue by going to pins_ramps and changing #define E0_CS_PIN   to a 2 instead of 11 , then plugging the TMC2130 CS wire into one of the end stop spots.  I had to change this configuration previously because I was using the LCD screen and the default pins that the TMC uses were in use by the screen. 

then I went and reenabled my BLtouch, plugged it into the correct spot, and all was well! 

also, there is a jumper that needs to be in place to power the needle thing so it can move up and down, I didn't know about that till the end.

Thanks For Your Help!

closed 2018-02-26 by thinkyhead