MarlinFirmware/Marlin #9799: [bugfix_2.0.x] LPC1768 BlinkM wrong I2C- Address

kAdonis, 2018-02-25:
### Bug Report
- Description: BlinkM not responding to M150 B255
- Expected behaviour: BlinkM  lights up blue

When I change i2C- Address in blinkm.cpp to 0x12 BlinkM is doing fine.

 Wire.beginTransmission(0x12)

So it seems to me, there is a Problem with 7 bit I2C- Adresses (like BlinkM with 0x09) wich are left shifted as stated in BlinkM Datasheet.
Just changing the Address to 0x12 breaks compatibility with Arduino, so i think  it has to done in the Hal
which is far beyond my skills

thinkyhead, 2018-02-25:
Did you know the address is configurable? https://thingm.com/fileadmin/thingm/downloads/BlinkM_datasheet.pdf

kAdonis, 2018-02-25:
Yes, I know that. But thats not the Point
The default Address 0x09 works with the Arduino mega, but with a ReArm and probably with the STMs its getting 0x12, because the 7bit Address ist left shifted.
this will affect all 7 bit Adresses on the I2C bus.

thinkyhead, 2018-02-25:
Ah, right. My hex math isn't working today. Thanks for clarifying.

So, you said if you give it 0x12 it works fine. That implies that we should be shifting all addresses our addresses left, so that the BlinkM will see the 0x09 it wants, correct?

Or is it the converse, where our usual 0x09 is being left-shifted on the way out, so that the BlinkM internally gets the 0x12 that it wants, but for STM and others we must do the left shift on its behalf?

thinkyhead, 2018-02-25:
Answered my own question with https://community.particle.io/t/i2c-7-or-8-bit-addresses-fixed-pull-request-submitted/1328/9

thinkyhead, 2018-02-25:
Our LPC HAL includes this… The address is being right-shifted before send.
```cpp
uint8_t TwoWire::endTransmission(void) {
  // transmit buffer (blocking)
  I2C_M_SETUP_Type transferMCfg;
  transferMCfg.sl_addr7bit = txAddress >> 1; // not sure about the right shift
	transferMCfg.tx_data = txBuffer;
```

kAdonis, 2018-02-25:
I dont know much about coding, i can´t help you there
but the BlinkM works with 0x12 and does not with 0x09

thinkyhead, 2018-02-25:
Well it's important that I state these things, because the core HAL developers will be reading this too. Thank you immensely for bringing this to our attention! Should be fixed later today.

kAdonis, 2018-02-25:
My pleasure!

boelle, 2019-02-20:
@thinkyhead was it fixed?

closed 2019-02-20 by thinkyhead

kAdonis, 2019-02-20:
@thinkyhead @boelle Sorry, this was **not** fixed. I know, it has a low priority and maybe I am the only one still using a BlinkM, but I still need to change I2C adress every time I download marlin

thinkyhead, 2019-02-20:
Hopefully this latest patch will take care of it!

kAdonis, 2019-02-20:
Dont know, if it's a unicorn, but it is working. Thanks a lot

commit a43e892f (2019-02-20): Use 8-bit i2c address on LPC platform

Fix #9799 — hopefully not a unicorn