MarlinFirmware/Marlin #9800: [BR] bugfix_2.0.x Radds LCD blank after upload

bwgeez02, 2018-02-25:
Expected behavior: Bootscreen then statuscreen

Actual behavior: LCD backlight turns on, first and third rows light up, but no bootscreen. No beeping if any buttons are pushed. Nothing happens if encoder is turned.

Steps to reproduce: Upload firmware with Arduino 1.9.0 beta (can't compile with any of the Arduino IDE's due to filename ext too long bug). LCD and printer operate normal after uploading known good firmware.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1755506/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1755507/Configuration_adv.zip)

bwgeez02, 2018-02-25:
Disabled PRINTCOUNTER per Bob-the-Kuhn advice on issue #9782. Boots normal now.

thinkyhead, 2018-02-25:
Hmm! Bug in `PRINTCOUNTER`?! Fun.

jamesarm97, 2018-04-20:
So is this still an issue? I just downloaded the latest as of today and get the two lines of blocks on the RADDS display. I tried Repetier with RADDs and the display works so must be something still going on in 2.0 with RADDS 4x20 display.

bwgeez02, 2018-04-21:
#9937 is supposed to fix the issue. I never recompiled with printcounter enabled, so I can't really answer your question. Did you try disabling printcounter?

bwgeez02, 2018-11-30:
Having this issue again with the latest build. I tried commenting sdsupport, printcounter is commented. I don't see any issues with pin configuration. 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2633388/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2633389/Configuration_adv.zip)

thinkyhead, 2018-11-30:
See if this is related: https://github.com/MarlinFirmware/Marlin/issues/12554

bwgeez02, 2018-11-30:
I read through that issue already. I'm using the standard Radds display and haven't had the ribbon cables off since the firmware update. It was working before, so I'm thinking it's related to the new firmware or I'm missing something in my config. I was using firmware from last May up til now.

thinkyhead, 2018-11-30:
Can you pinpoint the day when it stops working? Download a commit from halfway between May and now, then get a commit halfway between the working and non-working date, and repeat until you find the exact day whe it ceases to work.

bwgeez02, 2018-12-01:
It appears it was an issue with Arduino 1.9.0. Earlier commits had the same results, so I uninstalled all Arduino programs, files, and folders and installed a new copy. Machine is up and running again. Arghh.

closed 2018-12-01 by bwgeez02