MarlinFirmware/Marlin #9804: BUZZER no longer works on Anet A6

JohnnyTheOne, 2018-02-25:
BUZZER no longer works on Anet A6. are there others who also have it?
I use Bugfix 1.1.x and 2.0.x

thinkyhead, 2018-02-25:
When did it begin? It works ok in 1.1.8? New configs? `SPEAKER` enabled?

JohnnyTheOne, 2018-02-25:
I do not know when it started and yes speaker enabled. wanted to test the buzzer
I have tested version 1.1.0 does not work either. it must be in pins_ANET_10.h or my board is broken

JohnnyTheOne, 2018-02-25:
no sorry

thinkyhead, 2018-02-25:
Terribly sorry. We know the buzzer code works. You have a hardware problem.

JohnnyTheOne, 2018-02-25:
no hardware problem. I have tested pin 4 and it works
#define BEEPER_PIN 4 // 17
M300 S4000 P100

thinkyhead, 2018-02-25:
Ah, so you have solved a bug?

thinkyhead, 2018-02-25:
Can you verify that your display is either `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` or `ANET_FULL_GRAPHICS_LCD`? We incorporated the A8 configuration in advance of the A6, and received only one pins definition pertaining to both. Are you able to read the silkscreen on your board and see a version number?

JohnnyTheOne, 2018-02-25:
no only that pin 17 does not work

thinkyhead, 2018-02-25:
It's encouraging to know that Minimalism is alive and well. We'll follow up on this information and see what we can learn from other Anet A6 owners. Thanks much!

JohnnyTheOne, 2018-02-25:
my display is ANET_FULL_GRAPHICS_LCD and I have also tested REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER in config
and I have the Anet V1.0 from 2016

thinkyhead, 2018-02-25:
Bloody Pin 4! I hope this isn't one of those things I've heard before, meant to fix, and totally forgot.

JohnnyTheOne, 2018-02-25:
pin 4 is also not a good option it disturbs the FAN PIN

thinkyhead, 2018-02-26:
Isn't your buzzer hard-wired to the board or the LCD? How are you swapping pins around?

JohnnyTheOne, 2018-02-26:
buzzer is hard wired to the LCD. more gambling that science and I know is wrong

benlye, 2018-03-01:
According to the board schematics the buzzer on the A6 LCD is connected to pin 10 on the LCD connector.  Pin 10 on the LCD connector is in turn connected to digital pin 17 on the Atmega1284p.  I've confirmed this with a meter.

https://raw.githubusercontent.com/ralf-e/ANET-3D-Board-V1.0/master/ANET3D_Board_Schematic.png
https://raw.githubusercontent.com/ralf-e/ANET-3D-Board-V1.0/master/ANET3D_LCD12864_Board_Schematic.png

benlye, 2018-03-01:
I just confirmed that the buzzer on my Anet A6 LCD works correctly with the current pin definition (`#define BEEPER_PIN       17` in bugfix-2.0.x.

I guess your board or LCD has a hardware issue.

thinkyhead, 2018-03-02:
Thanks for the input, Ben. Now I know who to ask about A6 issues should there ever be any in the future. This is probably the last one we'll ever get, though.

closed 2018-03-02 by thinkyhead

keithluneau, 2018-05-02:
I just wanted to comment on this since it was helpful in diagnosing an issue with my printer. I have an A8 (same board as the A6), and I've installed the "Anet full graphics display" from the A6 onto it. I'm running Marlin 1.1.8. The beeper/buzzer on the display did not work. I was unable to generate any sounds using an M300 command, and had no feedback when selecting menu items. I landed here after searching a bit, and was happy to find the schematics posted for both the main board and LCD board. I worked through them and verified that they are correct (at least relevant to the buzzer) for the v1.5 Anet mainboard that I have. Looking at the code and the datasheet for the Atmega1284p, everything seems to be in order. I first verified that the Atmega was generating a signal on pin 20 (internal pin 17 as defined in the code), and then that the BC817 on the LCD board was closing and connecting the buzzer to ground as expected. 5V was present on the buzzer's positive pin. The next step was to remove the buzzer from the LCD board and replace it with a known working on. When doing so I found that the solder pad was physically not connected to the LCD board and that the trace leading to it was broken. It appears to have been overheated when assembled, or just poor quality PCB to begin with. Probably both... lol Once removed I found that the buzzer worked fine with 5V applied, so I scraped away the coating on the trace next to the buzzer's pin and soldered it back in making sure I had a good connection, and it works fine now. So just to verify, the pin for the buzzer is properly defined in the code, and the Anet full graphics display uses a piezoelectric buzzer, not a speaker.

keithluneau, 2018-05-03:
> If it can make different frequency tones, Marlin supports that with the SPEAKER option enabled.

Right! I just flashed again with `SPEAKER` Enabled and it does respond to different frequencies with M300. Thanks for the tip!