MarlinFirmware/Marlin #9808: Geeetech 3dTouch not retracting always. Help

Margus113, 2018-02-25:
Former FLSUN cube. Now upgraded to rail system and 3dtouch.
Tried a lot of setups etc. i dont know anymore, what now.
I will add configuration.h, can some smarter look, what i missing.
Sensor working, can perform all tests, but cant get work.
Sometimes, can do G28 ok, and G29 first few points, but after that,
senosor saty i upper position and nozzle rushing to bed.
[HELF.txt](https://github.com/MarlinFirmware/Marlin/files/1755905/HELF.txt)

psavva, 2018-02-25:
Can you run the selftest for 10 or 20 mins without failure?

I'd the pin loose when powering your bltouch clone down?

thinkyhead, 2018-02-25:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

Margus113, 2018-02-25:
Fater approx 1min self test, probe stay in upper pos and alarms.

thinkyhead, 2018-02-25:
Try enabling and increasing `BLTOUCH_DELAY`.
```cpp
#if ENABLED(BLTOUCH)
   //#define BLTOUCH_DELAY 375   // (ms) Enable and increase if needed
#endif
```

Margus113, 2018-02-25:
After 2 days of having love with that sensor, i finally get it to work.
Exact problem i cant say, but seems that were needed to adjust probe, from hex screw.
After adjusting /some random screwing.../ it not giving alert and working now.
Thanks

closed 2018-02-25 by Margus113

Bob-the-Kuhn, 2018-02-25:
What color is your 3D Touch?  Is it black or white?

Another user is having problems with a black one.  I haven't seen a black one until recently.