MarlinFirmware/Marlin #9811: Delta printer Dual hotend offsets - cannot set.

rhinorap, 2018-02-25:
------------------------------------------------------------
### Bug Report
- Description: Marlin 1.1.8 - Delta Printer DUAL hotend offset - CANNOT SET IN FW.
- Expected behaviour: to be able to set dual hotends to Y (-13.0, 13.0) as Delta printer 0,0 it the middle of the effector/bed (delta), not the edge of the bed (cartesian).
- Actual behaviour: FW assumes cartesian printer coordinates for hotends, meaning 0,0 is E0 and then E1,E2 etc are to one side of this. Delta printer 0,0 is the middle of the bed, not one side so setting E0 at a fixed 0,0 will not allow proper offsets.
Sanity check in compiler stops any progress.

Cartesian:

BED] (0,0)E0------>E1----->E2

Delta:

BED]------->E0-------0,0-----E1------->[BED

thinkyhead, 2018-02-28:
Marked as a feature request.

airbrett, 2020-04-25:
I'd be great if this could work. The only work around (that I've come up with) is leave E0 and E1 at 0,0 in Marlin and set the offsets in the slicer. This works but the compensation from bed leveling and (to a lesser extent) auto calibration are applied based on the X/Y position at the center of the effector, not the X/Y of the active nozzle.