MarlinFirmware/Marlin #9812: [bugfix-1.1.x] Firmware Retraction G10 S1 does nothing

agrif, 2018-02-25:
## Bug Report

I have tested this on 8f6a5d538fc1c9e57e27cc2a3c1d8d8f0b8c66b4 with the [attached configuration][configuration.zip].

  [configuration.zip]: https://github.com/MarlinFirmware/Marlin/files/1756536/configuration.zip

### Description

Issuing a `G10 S1` command does not actually retract at all, despite a positive defined `RETRACT_LENGTH_SWAP`. It also does not perform a z-hop, despite that also being enabled. The other firmware retract commands (`G10` and `G11`) work as expected. This makes it somewhat difficult to use multi-extruder files produced by slic3r, which emit `G10 S1` on toolchange and not `G10`.

### Expected Behavior

`G10 S1` should retract the distance defined by `RETRACT_LENGTH_SWAP`, or... at least some distance. It should also move the print head up by `RETRACT_ZLIFT`.

### Actual Behavior

`G10 S1` does not retract any distance, or move the print head at all. After issuing a `G10 S1`, issuing `G11` does not do anything either, so at least it's consistent.

GMagician, 2018-02-25:
In code I read a comment like this
    `// Allow G10 S1 only after G10`
don't know G10 purpose but maybe you have to generate a simple G10 before G10 S1

GMagician, 2018-02-25:
There is also a difference between 2.0 and 1.1...

look at Marlin_main.cpp, line 3452 you will see:
`fwretract.retracted_swap[active_extruder] = rs; // Use 'S' for swap, default to false`
try to put `//` in front, compile and check if it works

agrif, 2018-02-25:
When testing this earlier I saw that comment and tried `G10` followed by `G10 S1`. It retracted the normal amount on `G10`, but the `G10 S1` didn't do any additional retraction. My configuration should have a 5mm normal retract, and 7mm swap retract, and I didn't see that.

I tried commenting out the line you mentioned. After this change, `G10` will retract 5mm, and `G10 S1` retracts 7mm, but `G10` followed by `G10 S1` still only does 5mm. This might work as a short term fix to get me printing, but it doesn't seem to fix the underlying bug.

Studiodyne, 2018-02-26:
How many extruders ?
FwRetract made only for 2 or more

agrif, 2018-02-26:
Two extruders.

So the intended behavior of `G10` + `G10 S1` is to only retract the normal distance, not the swap distance? That seems odd to me, but if that's true then [uncommenting this line][line] as @GMagician mentioned seems to fix it. Someone who knows more than I do about the code should check to be sure that's an ok thing to remove.

  [line]: https://github.com/MarlinFirmware/Marlin/blob/8f6a5d538fc1c9e57e27cc2a3c1d8d8f0b8c66b4/Marlin/Marlin_main.cpp#L3452

Studiodyne, 2018-02-26:
The thirst to say , when you make G10 + G10S1 is ' WHY ? '
Marlin is not the doctor a BAD SLICER 
Marlin is not the cure for your bad gcode Files
G10 + G10S1 , first is not a 'normal gcode instruction ' , why G10 and G10S1 ? 
Marlin is to make ' good sense , and good clever instructions '
If i ask marlin to make ' bolognese pasta' i'm sure a strange behavior will appear

Studiodyne, 2018-02-26:
Make a G10S1 will long retract , and only a G11 must appear after that
Make a G10 , short retract , and only a G11 must appear after that

Marlin is made to prevent some bad gcode , but always to avoid big crash , like G10 S1 + G10 =G10S1 , because When G11 the long retract will be recovered and your print have strong chances to run normally
If MARLIN , after G10S1 + G10 ,  just recover the short retract , your print will have very long time before something will be extruder

Marlin not a doctor for bad gcodes files , or folkloric exotic instructions

The real topic is 
WHY NO ACTION ON G10S1
Make this for me , reset the printer , plug usb , start host , and send 
M207 F3000 S3 W50 Z0 (by lcd you can set manually : Feedrate + length + swap + hop )
M208 F1500 S0 W0 R1500 (same for recovery)
G10S1
and tell me is something's wrong 
👍

agrif, 2018-02-26:
I am away from my printer for the rest of the week, but will try this when I return this weekend.

In the meantime, I can say that on commit 8f6a5d538fc1c9e57e27cc2a3c1d8d8f0b8c66b4 when I reset everything and then run

    M207 S5 W7
    G10 S1

the extruder does nothing. Doing the same thing with only `G10` works fine.

Studiodyne, 2018-02-26:
Set the feedrates , to be sure , because if you have a feedrate of 0 the gear will never make a move

I'm the author of the improvements of FWretract
And i'm happy to see that someone use it and happy to see if somthing's wrong
I use it every day , and i love it
I have cleaned the code and make it to ensure all folkloric gcode files , to always give priority for a safe print
But , everybody knows , that marlin is not a doctor , the first things to ensure is a good gcode file 

Ok , i'm waiting some news

Studiodyne, 2018-02-26:
For zlift you have to set it
M207 F3000(feedrate=3000m/min but in lcd=50mm/s) S3(short retract)  W50(swap retract)  Z(=ZLIFT)
M208 F1500 S0 W0 R(recovery length) 1500 (same for recovery)

thinkyhead, 2018-02-27:
@Studiodyne — There's a default feedrate if none is set explicitly. And I'm shocked if Marlin allows a feedrate of 0 to be set, causing everything to freeze. That would be quite an oversight.

thinkyhead, 2018-02-27:
> But , everybody knows , that marlin is not a doctor , the first things to ensure is a good gcode file

Nevertheless, if there is something like `G10\nG10 S1` what would be the 'most correct' way to handle that. What would the user expectation be?

agrif, 2018-02-27:
My expectation was that `G10` means something like "put the filament away", and `G10 S1` means something like "put the filament really far away because we're going to change tools." Two `G10`s in a row probably should act like one, since if you already put it away, there's nothing to do on the second one. Same with two `G10 S1`s in a row.

I thought that issuing a `G10` then `G10 S1` would first retract the normal distance, then the additional distance needed to reach the swap retract distance. I can even imagine a use: first the slicer issues `G10` to retract quickly, then moves the nozzle off the print to issue the potentially much slower `G10 S1` before changing tools.

That's what I assumed the commands would do, anyway, especially after reading the comment in the source that mentions that a `G10 S1` must come after a `G10`.

Studiodyne, 2018-02-27:
@thinkyhead  yes it's true , but it's just exageration to explain the feedrate is important 😄 
 The way for a machine to translate G10/nG10S1 is that G10S1 always takes priority and print have chances to survive , and it's a gift of marlin to save the print. The only exception is for SLIC3R because the devs of slic3r never have deleted this ' non grata :smile ' G10 after G10S1 in their flavor .

Studiodyne, 2018-02-27:
@agrif 
Why retract 2mm , and a moment later , retract again ... totally incomprehensible 
You retract or not
Lol , you don't have a MARLIN issue lollllllllll
If you love to stack retraction or segmenting retraction
DISABLE FwRetract , and use the slicer 
You will have tons of possibilities 👍 
G1 Exxxx or G1 E-xxxxx  , you can stack , you can unload , anytime you want 
@agrif @thinkyhead  it's joke , it's for laugh !!! 
I help , but i help first to understand loll

@agrif 
Do the FwRetract works or not on your machine now  ?

agrif, 2018-02-27:
@Studiodyne still waiting till I get back to my machine to test. Just answering @thinkyhead in the meantime.

Studiodyne, 2018-02-27:
if G10 works , then set  the value for G10 ' 30mm' and see if all works
If a G10 of 30mm works then a G10S1 must work too
If not ... it's a config problem , check all basics and give me you ' adv.h'

thinkyhead, 2018-02-28:
> Last I tested, G10 S1 still does nothing on the bugfix-1.1.x branch.

Weird. Since this is @Studiodyne's code, I will leave it up to him to figure out why that is happening and provide a fix.

@Studiodyne — Do you think you can put more energy into problem-solving on this one?

Studiodyne, 2018-02-28:
@thinkyhead @agrif 

The bug is , fwretract of bugfix 1.1.x is not updated , the last update is on 2.0

To correct this , go to bugfix 1.1.x
line 3452 
 fwretract.retracted_swap[active_extruder] = rs; // Use 'S' for swap, default to false
and delete this 

and thinkyhead will make the fix for bugfix and 1.1.8
I have made the PR  for 2.0 , and it's thinkyhead that manage the transmition to other release

it's a good guy ... don't scream on him lolll
OK I STOP COCAIN

:kiss:

agrif, 2018-02-28:
Yes, removing [Marlin_main.cpp:3452][fix] does seem to fix `G10 S1`, thanks.

 [fix]: https://github.com/MarlinFirmware/Marlin/blob/8f6a5d538fc1c9e57e27cc2a3c1d8d8f0b8c66b4/Marlin/Marlin_main.cpp#L3452

Studiodyne, 2018-02-28:
@thinkyhead 
i've seen that a new line about Fwretract have been created , about G28 homing ...it's a good idea , but , the risk is the same : 
-You reset a retract status , but without reset the hop amount ' yes , a zlift can exist ' and you cannot access to it
-You have written for resetting G10 , but G10 S1 have the same risk ' must be written '

if and old G10 is of 3mm , and you delete it , the next G10 +G11 will go down of 6mm a breaks the glass

Suggestion: 
No idea

Studiodyne, 2018-02-28:
@thinkyhead 
If you want i make PR for all release i can , but before , take the TOOL MIGRATION CODE and make it in your name , i need to REFORK and i don't want to remake all my current PR
thks

Studiodyne, 2018-03-01:
@agrif , Your request here , have revealed a little issue , when homing , the reset of FwRetract is not made correctly , and to make it correctly we have to write a long code , it's why now , we have added information to inform FwRetract is not reset after homing , and if someone make a G28 after a zlift ,  the zlift exist after g28 and the first g11 

Thks for all
You can close :-) 
And send me your feedback about using Fw

closed 2018-03-03 by agrif

thinkyhead, 2018-03-04:
> …take the TOOL MIGRATION CODE and make it in your name…

Sorry about that. You can go ahead and pull it into your own new branch and start a new PR with it.

```
git remote add thinkyhead git@github.com:thinkyhead/Marlin.git
git fetch thinkyhead
git checkout thinkyhead/bf2_filament_load_menu -b my_filament_load_menu
git push --set-upstream origin my_filament_load_menu
```