MarlinFirmware/Marlin #9821: [ANET A8] X homing ignoring endstop

KALRONG, 2018-02-26:
### Bug Report
- Description: 90% of the time I home the X axis the endstop is ignored and the machine shows the X value of -379 causing the a lot of noise and mechanical issues as it tries to move over the endstop.
- Expected behaviour: The carriage to stop on the endstop or at least when it reaches de soft endstop at -33 (this actually works).
- Actual behaviour: 90% of the time I home the X axis the endstop is ignored and the machine shows the X value of -379 causing the a lot of noise and mechanical issues as it tries to move over the endstop.
- Steps to reproduce:
  - Home X axis

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1758697/Marlin.zip)

benlye, 2018-02-27:
Sounds like a bad endstop.  

You can use `M119` to get the endstop status.  I'd suggest manually triggering it and using `M119` to see if the software shows the correct state over a number of cycles.

Second idea - check if `ENDSTOP_INTERRUPTS_FEATURE` is enabled in Configuration.h and disable it if it is (also disable `ENDSTOPS_ALWAYS_ON_DEFAULT` in Configuration_adv.h if you disable `ENDSTOP_INTERRUPTS_FEATURE`).

KALRONG, 2018-03-05:
The endstop was working fine after several test, I did change the settings you mentioned and now its been working fine so far, thanks a lot :)

KALRONG, 2018-03-05:
No luck, just tried again today and is doing the same, its wokring like 65% of the time only, and now sometimes only jumps on the first touch and stops on the second

thinkyhead, 2018-03-09:
It sounds a lot like a wiring or floating pin issue. Apparently the Anet A8 uses either optical or Hall effect endstops (`*_ENDSTOP_INVERTING true`), since it uses the opposite state from mechanical switches (`*_ENDSTOP_INVERTING false`). Mechanical switches are normally-closed (NC), so they aren't affected by stray current or stray magnetic fields (only by loose connections). But since optical and Hall effect endstops are NO switches, they are more susceptible to induction, especially from nearby motor wires and PWM current. 

In future we'll have a DEBUG screen to show the state of the endstops in real-time for easier testing, but for now try this:
- Start from a successful XYZ homing…
- Do a move of X to the far end of the axis with `G1 X220 F300`.
- Do `M119` repeatedly during the move to see if any endstops change.
- Do a move of Y to the far end of the axis with `G1 Y220 F300`.
- Do `M119` repeatedly during the move to see if any endstops change.
- Do combined move of XY with `G1 X10 Y10`.
- Do `M119` repeatedly during the move to see if any endstops change.

Perhaps movement will cause a spurious change of endstop state.

Next, trigger the X but leave Y open. Check states with `M119`. Repeat a few times.
Now trigger the Y but leave X open. Check states with `M119`. Repeat a few times.

If none of this reveals anything, we can look further into it.

benlye, 2018-03-09:
>  Apparently the Anet A8 uses either optical or Hall effect endstops

Nope, good old fashioned mechanical switches, and pretty cheap ones at that.

Seems like at least a couple of people have had issues with the endstop pullup resistors.
https://www.thingiverse.com/groups/anet-a8-prusa-i3/forums/general/topic:20481

thinkyhead, 2018-03-11:
Hmm, it does sound like a hardware fault. As it turns out, I forgot that the endstops can be monitored in real-time (but it blocks other commands). You must enable `PINS_DEBUGGING` and then use the `M43 E` command.

Croug, 2018-06-24:
Just had this problem on mine, the cable running to the x axis endstop is slightly too small for the gantry to reach the top, so when it does it wears down on the connections to the switch. Check on the wires near the top and near the board, and anywhere they run along a corner and make sure the connection is still fine.

thinkyhead, 2018-06-30:
Closing due to inactivity (and not likely an actionable bug in Marlin).

closed 2018-06-30 by thinkyhead