MarlinFirmware/Marlin #9823: [Configuration] UBL Problem

Gigahertz77, 2018-02-26:
I don't know how to resolve this problem, step 1 ( G29 P1   ) after 58 test point, the probe come out of the bed.  I have try every number.
Bed 220x220

#define X_PROBE_OFFSET_FROM_EXTRUDER 38   // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 5   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.05   // Z offset: -below +above  [the nozzle]

  #define UBL_PROBE_PT_1_X 25     // Probing points for 3-Point leveling of the mesh
  #define UBL_PROBE_PT_1_Y 200
  #define UBL_PROBE_PT_2_X 25
  #define UBL_PROBE_PT_2_Y 30
  #define UBL_PROBE_PT_3_X 200
  #define UBL_PROBE_PT_3_Y 30
[image]

Roxy-3D, 2018-02-26:
You probably should shrink the effective bed size with an INSET....     And once everything stays on the bed, you can increase the INSET on the unaffected sides of the bed, one side at a time...

Gigahertz77, 2018-02-26:
this parameter? #define MESH_INSET 1              // Mesh inset margin on print area

Roxy-3D, 2018-02-26:
Yes.     But if that works...   In Configuration_adv.h you will be able to set the INSET for each side of the bed.      There is something else wrong with your settings.  But this will help you work around in and figure out the problem.

Studiodyne, 2018-02-26:
To proble too close to the edge will give issues in the future , a lag with the endstop or anything that make busy the machine , and the positions have 1mm less , it's why , apply a margin just to ensure a random 'endstoping'

gregspruce, 2018-02-26:
Can I have that inset, but then tell UBL to apply matrix points outside of the inset area so I can fill it in with G29 P2/P3? I guess the UBL raise/lower to infinity bug is fixed in one of the more recent commits to the bugfix branch but still...

Studiodyne, 2018-02-26:
No ... Ubl is not windows Xperience with 10 years old and 300 engineers lolll
Ubl is beautifull , but  , when someone doesn't know it , sometimes he wants something impossible.
For example in my situation , i have an outside purge bucket , and ubl can go on this area but no bed , the ubl don't set outside mesh points as the nearest , it's why @Roxy-3D have made a secure z height for off mesh points 
Try to improvise and find solution by creativity lollllllllllllll

thinkyhead, 2018-02-26:
> Bug of raise up , or raise down , is solved

Do you mean, "is solved" just for you, or "is solved" in the latest code?

thinkyhead, 2018-02-26:
@Gigahertz77 — The automated settings that determine the probing boundaries normally consider your probe offset, but for some reason this has been removed for UBL. The comment (below) says "UBL is careful not to probe outside the bed." However it still apparently tries to use the areas you can never reach. I believe the intent was that you would have to set those unreachable points manually after auto probing. However, in this case it looks like the points outside where your probe can reach are not, in fact, being left out.

@Roxy-3D — Can you have a look at UBL and see how it handles the probe boundaries? First of all, there is this block from `Conditionals_post.h`…

```cpp
      #if ENABLED(AUTO_BED_LEVELING_UBL)
        #define _MESH_MIN_X (max(X_MIN_BED + MESH_INSET, X_MIN_POS))  // UBL is careful not to probe off the bed.  It does not
        #define _MESH_MIN_Y (max(Y_MIN_BED + MESH_INSET, Y_MIN_POS))  // need *_PROBE_OFFSET_FROM_EXTRUDER in the mesh dimensions
        #define _MESH_MAX_X (min(X_MAX_BED - (MESH_INSET), X_MAX_POS))
        #define _MESH_MAX_Y (min(Y_MAX_BED - (MESH_INSET), Y_MAX_POS))
      #else
        #define _MESH_MIN_X (max(X_MIN_BED + MESH_INSET, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
        #define _MESH_MIN_Y (max(Y_MIN_BED + MESH_INSET, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
        #define _MESH_MAX_X (min(X_MAX_BED - (MESH_INSET), X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
        #define _MESH_MAX_Y (min(Y_MAX_BED - (MESH_INSET), Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
      #endif
```

And then in `UBL.h` it is definitely not considering the probe offset when deciding how large the mesh is or how far apart the grid lines are:
```cpp
#define MESH_X_DIST (float(MESH_MAX_X - (MESH_MIN_X)) / float(GRID_MAX_POINTS_X - 1))
#define MESH_Y_DIST (float(MESH_MAX_Y - (MESH_MIN_Y)) / float(GRID_MAX_POINTS_Y - 1))
```

Am I correct in my assumption that your plan was to allow people to specify the whole bed, even if their probe cannot reach it, and then have them do manual probing for those points?

thinkyhead, 2018-02-26:
A good rule of thumb, by the way, if your nozzle can move outside the bed, is to install your probe on the side of the nozzle where it will still be on the bed when the nozzle is outside.

Gigahertz77, 2018-02-26:
Not yet, because it's not my language, and I have a little problem to understand the right procedure.

Roxy-3D, 2018-02-27:
> And then in UBL.h it is definitely not considering the probe offset when deciding how large the mesh is or how far apart the grid lines are:

That is true.   With no INSET, and a X or Y_PROBE_OFFSET_FROM_EXTRUDER it is not possible to auto probe the entire bed.   But those points can be acquired and filled in easily with manual probing or with a G29 P3 (which does a smart fill).

thinkyhead, 2018-02-27:
So it sounds like the comparison that prevents the probe from going outside of the bed is potentially failing in `ubl.G29`. I'll look through the code and see if it's anything obvious.

thinkyhead, 2018-02-27:
Ah. Well here's the issue. There is no check. It just goes happily on with `probe_pt` whether the point can be reached or not. I suggest that it do the following:
```cpp
const float measured_z = position_is_reachable_by_probe(rawx, rawy)
                         ? probe_pt(rawx, rawy, stow_probe, g29_verbose_level)
                         : NAN;
```

thinkyhead, 2018-02-27:
Ah. Well here's the issue. There is no check. It just goes happily on with `probe_pt` whether the point can be reached or not. I suggest that it do the following:
```cpp
const float measured_z = position_is_reachable_by_probe(rawx, rawy)
                         ? probe_pt(rawx, rawy, stow_probe, g29_verbose_level)
                         : NAN;
```
Or, if it's better to leave the unreachable points unchanged…
```cpp
if (position_is_reachable_by_probe(rawx, rawy))
  z_values[location.x_index][location.y_index] = probe_pt(rawx, rawy, stow_probe, g29_verbose_level);
```
I guess the first option is better, since it then knows that the requested points need to be manually probed.

thinkyhead, 2018-02-27:
Possibly this test is now failing?
```cpp
        if (location.x_index >= 0) {    // mesh point found and is reachable by probe
```

thinkyhead, 2018-02-27:
So the logic of `find_closest_mesh_point_of_type` appears pretty sound. I can't see any reason why it would let an "unreachable" point through. So, maybe something is just amiss about @Gigahertz77's probe offset or probing boundaries.

thinkyhead, 2018-02-27:
So theoretically, you could put points off the bed if you wanted to. The probe couldn't reach them, but you could still manually set them. In that case, maybe it makes sense to skip probe points where neither the nozzle nor the probe can reach.

Roxy-3D, 2018-02-27:
In fact...   That is what happens with Delta's.   Each corner has a bunch of mesh points that are off the bed.

thinkyhead, 2018-02-27:
Indeed! That's why ABL operates the way it does. It inherits directly from the work by **jcrocholl**.

---
Delta's … _what?_ Delta's birthday?
Apostrophes are not needed to make a noun plural.
#grammarnazi 😈

Roxy-3D, 2018-02-27:
Or show possession.   The Delta owns those locations on the bed.

gloomyandy, 2018-02-27:
A setting I've found really useful is MIN_PROBE_X, MIN_PROBE_Y, MAX_PROBE_X, MAX_PROBE_Y. These settings allow you to override the defaults for the probable area without impacting the mesh size. So in my case I wanted to have a mesh that covers all of the bed (to allow me to print edge to edge), but to only automatically probe the area reachable by the probe (obviously) plus an extra margin because probing close to the edge of my bed gives odd results (the sensing area of the inductive probe is partly off the bed). Using the above settings with the following definitions:
```
// Min/Max probe area. We can only probe inside of the bed and really need to be 10mm away from an edge to get a good reading.
// Boundaries for Cartesian probing based on bed limits
#define MIN_PROBE_X (max(X_MIN_POS+10, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
#define MIN_PROBE_Y (max(Y_MIN_POS+10, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
#define MAX_PROBE_X (min(X_MAX_POS - 10, X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
#define MAX_PROBE_Y (min(Y_MAX_POS - 10, Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
```
does this for me and works very well. I get the centre part of the mesh probed automatically with a strip down each side which I either do a manual probe for or use auto-fill on. Note that this is not the same as using the UBL MESH_INSET which would inset the mesh by 10mm (or whatever), I have MESH_INSET set to 0 so the mesh covers my entire reachable area (which in my case is all of the bed).

ManuelMcLure, 2018-02-27:
Thank you, thank you, thank you! [MIN|MAX}_PROBE_[X|Y] is what I've been looking for to allow me to make the jump to UBL, since I have an inductive probe and need to keep the probe away from the edges of the bed. Assuming @Roxy-3D doesn't know any reason why this wouldn't work for my situation (want to use the whole bed but don't want to probe near the edges of the bed), I may give UBL another try this weekend.

Roxy-3D, 2018-02-28:
I think that should work...

boelle, 2019-02-20:
@Gigahertz77 problem solved?

Gigahertz77, 2019-02-21:
No, I quit, I have try 2 days without result.

boelle, 2019-02-21:
@Gigahertz77 maybe click the green button below?

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle