MarlinFirmware/Marlin #9824: [bugfx-2.0.x] M43 Output incorrect for ATmega1284p board

benlye, 2018-02-26:
When I run M43 on an Anet v1.0 board (ATmega1284p) the output for the pins is incorrect - the analog ADC pin appears as digital and the analog value appears against the wrong pin.  The value that appears agains pin 29 should be on pin 25.

The board is configured to use the standard Anet A8 5-button ADC display, with ADC_KEYPAD_PIN set to 1.

```
PIN:   0   Port: B0        E0_DIR_PIN                             protected 
PIN:   1   Port: B1        E0_STEP_PIN                            protected 
PIN:   2   Port: B2        Z_DIR_PIN                              protected 
PIN:   3   Port: B3        Z_STEP_PIN                             protected 
PIN:   4   Port: B4        AVR_SS_PIN                             protected 
.                          FAN_PIN                                protected 
.                          SS_PIN                                 protected 
PIN:   5   Port: B5        AVR_MOSI_PIN                           Output = 1  
.                          MOSI_PIN                               Output = 1
PIN:   6   Port: B6        AVR_MISO_PIN                           Input  = 1  
.                          MISO_PIN                               Input  = 1
PIN:   7   Port: B7        AVR_SCK_PIN                            Output = 0  
.                          SCK_PIN                                Output = 0
PIN:   8   Port: D0        RXD                                    Input  = 1  
PIN:   9   Port: D1        TXD                                    Input  = 0  
PIN:  10   Port: D2        LCD_PINS_D4                            Output = 0  
PIN:  11   Port: D3        LCD_PINS_D5                            Output = 0  
PIN:  12   Port: D4        HEATER_BED_PIN                         protected 
PIN:  13   Port: D5        HEATER_0_PIN                           protected 
PIN:  14   Port: D6        E0_ENABLE_PIN                          protected 
.                          X_ENABLE_PIN                           protected 
.                          Y_ENABLE_PIN                           protected 
PIN:  15   Port: D7        X_STEP_PIN                             protected 
PIN:  16   Port: C0        LCD_PINS_D6                            Output = 0  
PIN:  17   Port: C1        LCD_PINS_D7                            Output = 0  
PIN:  18   Port: C2        X_MIN_PIN                              protected 
.                          X_STOP_PIN                             protected 
PIN:  19   Port: C3        Y_MIN_PIN                              protected 
.                          Y_STOP_PIN                             protected 
PIN:  20   Port: C4        Z_MIN_PIN                              protected 
.                          Z_STOP_PIN                             protected 
PIN:  21   Port: C5        X_DIR_PIN                              protected 
PIN:  22   Port: C6        Y_STEP_PIN                             protected 
PIN:  23   Port: C7        Y_DIR_PIN                              protected 
PIN:  24   Port: A7        <unused/unknown>                       Input  = 0  
PIN:  25   Port: A6        ADC_KEYPAD_PIN                         Input  = 0  
PIN:  26   Port: A5        Z_ENABLE_PIN                           protected 
PIN:  27   Port: A4        SERVO0_PIN                             Input  = 0  
PIN:  28   Port: A3        LCD_PINS_RS                            Analog in =   279  
.                          LCD_SDSS                               Analog in =   275
PIN:  29   Port: A2        LCD_PINS_ENABLE                        Analog in =  1023  
PIN:  30   Port: A1        TEMP_BED_PIN                           protected 
.                          TEMP_BED_PIN                           protected 
PIN:  31   Port: A0        SDSS                                   Output = 1  
ok
```

Compare to the output from M43 on bugfix-1.1.x:
```
PIN:   0   Port: B0        E0_DIR_PIN                  protected 
PIN:   1   Port: B1        E0_STEP_PIN                 protected 
PIN:   2   Port: B2        Z_DIR_PIN                   protected 
PIN:   3   Port: B3        Z_STEP_PIN                  protected 
PIN:   4   Port: B4        AVR_SS_PIN                  protected 
.                          FAN_PIN                     protected 
.                          SS_PIN                      protected 
PIN:   5   Port: B5        AVR_MOSI_PIN                Output = 1  
.                          MOSI_PIN                    Output = 1
PIN:   6   Port: B6        AVR_MISO_PIN                Input  = 1  
.                          MISO_PIN                    Input  = 1
PIN:   7   Port: B7        AVR_SCK_PIN                 Output = 0  
.                          SCK_PIN                     Output = 0
PIN:   8   Port: D0        RXD                         Input  = 1  
PIN:   9   Port: D1        TXD                         Input  = 0  
PIN:  10   Port: D2        LCD_PINS_D4                 Output = 0  
PIN:  11   Port: D3        LCD_PINS_D5                 Output = 0  
PIN:  12   Port: D4        HEATER_BED_PIN              protected 
PIN:  13   Port: D5        HEATER_0_PIN                protected 
PIN:  14   Port: D6        E0_ENABLE_PIN               protected 
.                          X_ENABLE_PIN                protected 
.                          Y_ENABLE_PIN                protected 
PIN:  15   Port: D7        X_STEP_PIN                  protected 
PIN:  16   Port: C0        LCD_PINS_D6                 Output = 0  
PIN:  17   Port: C1        LCD_PINS_D7                 Output = 0  
PIN:  18   Port: C2        X_MIN_PIN                   protected 
.                          X_STOP_PIN                  protected 
PIN:  19   Port: C3        Y_MIN_PIN                   protected 
.                          Y_STOP_PIN                  protected 
PIN:  20   Port: C4        Z_MIN_PIN                   protected 
.                          Z_STOP_PIN                  protected 
PIN:  21   Port: C5        X_DIR_PIN                   protected 
PIN:  22   Port: C6        Y_STEP_PIN                  protected 
PIN:  23   Port: C7        Y_DIR_PIN                   protected 
PIN:  24   Port: A7        <unused/unknown>            Input  = 0  
PIN:  25   Port: A6        ADC_KEYPAD_PIN              Analog in =  1020  
PIN:  26   Port: A5        Z_ENABLE_PIN                protected 
PIN:  27   Port: A4        SERVO0_PIN                  Input  = 0  
PIN:  28   Port: A3        LCD_PINS_RS                 Output = 1  
.                          LCD_SDSS                    Output = 1
PIN:  29   Port: A2        LCD_PINS_ENABLE             Output = 0  
PIN:  30   Port: A1        TEMP_BED_PIN                protected 
.                          TEMP_BED_PIN                protected 
PIN:  31   Port: A0        SDSS                        Output = 1  
ok
```

I've spent some time trying to figure it out but I'm stuck.

Incidentally, to get the output to line up correctly in bugfix-2.0.x I had to change `#define MULTI_NAME_PAD 14` to `#define MULTI_NAME_PAD 26` in pinsDebug_AVR_8_bit.h.

thinkyhead, 2018-02-27:
I believe that @Bob-the-Kuhn would say the `pinsDebug` feature is incomplete in 2.0.x and we have more work to do before it has parity with 1.1.x.

thinkyhead, 2018-02-27:
I just noticed that the `pinsDebugList.h` has gotten a bit corrupted since I last cleaned it up. Non-pin entries like "`NUM_DIGITAL`" have reappeared. Let me patch that up and we'll see if it helps at all.

thinkyhead, 2018-02-27:
Ok, so I've removed some duplicate entries from `pinsDebug_list.h` in both 1.1.x and 2.0.x, so download them and retry. I didn't do anything that would guarantee pin 29 / 25 is correct. I still have to compare the pin mappings for 1284p in both branches.

benlye, 2018-02-27:
New problems now 😉 

Errors when compiling for ATmega1284p (Anet):
```
In file included from Marlin\src\gcode\config\M43.cpp:29:0:
Marlin\src\gcode\config\../../inc/../HAL/HAL_AVR/fastio_644.h:54:21: error: 'DIO16' was not declared in this scope
#define SCL         DIO16
^
Marlin\src\gcode\config\../../pins/pinsDebug.h:63:64: note: in definition of macro '_ADD_PIN_2'
#define _ADD_PIN_2(ENTRY_NAME, NAME, IS_DIGITAL) { ENTRY_NAME, NAME, IS_DIGITAL },
^
MMarlin\src\gcode\config\../../pins/pinsDebug.h:65:44: note: in expansion of macro '_ADD_PIN'
#define REPORT_NAME_DIGITAL(COUNTER, NAME) _ADD_PIN(NAME, COUNTER, true)
^
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:621:3: note: in expansion of macro 'REPORT_NAME_DIGITAL'
REPORT_NAME_DIGITAL(__LINE__, SCL)
^
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:621:33: note: in expansion of macro 'SCL'
REPORT_NAME_DIGITAL(__LINE__, SCL)
^
Marlin\src\gcode\config\../../inc/../HAL/HAL_AVR/fastio_644.h:55:21: error: 'DIO17' was not declared in this scope
#define SDA         DIO17
^
Marlin\src\gcode\config\../../pins/pinsDebug.h:63:64: note: in definition of macro '_ADD_PIN_2'
#define _ADD_PIN_2(ENTRY_NAME, NAME, IS_DIGITAL) { ENTRY_NAME, NAME, IS_DIGITAL },
^
Marlin\src\gcode\config\../../pins/pinsDebug.h:65:44: note: in expansion of macro '_ADD_PIN'
#define REPORT_NAME_DIGITAL(COUNTER, NAME) _ADD_PIN(NAME, COUNTER, true)
^
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:627:3: note: in expansion of macro 'REPORT_NAME_DIGITAL'
REPORT_NAME_DIGITAL(__LINE__, SDA)
^
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:627:33: note: in expansion of macro 'SDA'
REPORT_NAME_DIGITAL(__LINE__, SDA)
^
In file included from Marlin\src\gcode\config\../../pins/../HAL/HAL_pinsDebug.h:27:0,
from Marlin\src\gcode\config\../../pins/pinsDebug.h:103,
from Marlin\src\gcode\config\M43.cpp:29:
Marlin\src\gcode\config\../../pins/../HAL/HAL_AVR/pinsDebug_AVR_8_bit.h:225:13: warning: 'void print_is_also_tied()' defined but not used [-Wunused-function]
static void print_is_also_tied() { SERIAL_PROTOCOLPGM(" is also tied to this pin"); SERIAL_PROTOCOL_SP(14); }
^
*** [.pioenvs\melzi_optiboot\src\src\gcode\config\M43.cpp.o] Error 1
```

Errors when compiling for ATmega2560 (RAMPS):
```
In file included from Marlin\src\gcode\config\M43.cpp:29:0:
Marlin\src\gcode\config\../../inc/../HAL/HAL_AVR/fastio_1280.h:50:21: error: 'DIO21' was not declared in this scope
#define SCL         DIO21
^
Marlin\src\gcode\config\../../pins/pinsDebug.h:63:64: note: in definition of macro '_ADD_PIN_2'
#define _ADD_PIN_2(ENTRY_NAME, NAME, IS_DIGITAL) { ENTRY_NAME, NAME, IS_DIGITAL },
^
Marlin\src\gcode\config\../../pins/pinsDebug.h:65:44: note: in expansion of macro '_ADD_PIN'
#define REPORT_NAME_DIGITAL(COUNTER, NAME) _ADD_PIN(NAME, COUNTER, true)
^
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:621:3: note: in expansion of macro 'REPORT_NAME_DIGITAL'
REPORT_NAME_DIGITAL(__LINE__, SCL)
^
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:621:33: note: in expansion of macro 'SCL'
REPORT_NAME_DIGITAL(__LINE__, SCL)
^
Marlin\src\gcode\config\../../inc/../HAL/HAL_AVR/fastio_1280.h:51:21: error: 'DIO20' was not declared in this scope
#define SDA         DIO20
^
Marlin\src\gcode\config\../../pins/pinsDebug.h:63:64: note: in definition of macro '_ADD_PIN_2'
#define _ADD_PIN_2(ENTRY_NAME, NAME, IS_DIGITAL) { ENTRY_NAME, NAME, IS_DIGITAL },
^
Marlin\src\gcode\config\../../pins/pinsDebug.h:65:44: note: in expansion of macro '_ADD_PIN'
#define REPORT_NAME_DIGITAL(COUNTER, NAME) _ADD_PIN(NAME, COUNTER, true)
^
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:627:3: note: in expansion of macro 'REPORT_NAME_DIGITAL'
REPORT_NAME_DIGITAL(__LINE__, SDA)
^
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:627:33: note: in expansion of macro 'SDA'
REPORT_NAME_DIGITAL(__LINE__, SDA)
^
*** [.pioenvs\megaatmega2560\src\src\gcode\config\M43.cpp.o] Error 1
```

These can be made to go away if I change these lines in `pinsDebug_list.h`:
```
#if defined(SCL) && SCL >= 0
#if defined(SDA) && SDA >= 0
```
Back to: 
```
#if defined(SCL) && SCL > 0
#if defined(SDA) && SDA > 0
```

But, I noticed that in https://github.com/MarlinFirmware/Marlin/commit/27f7e3fe6f8493ef7390dba9d5ab0ceebab38b6d a few other 'problem pins' (including `SCK`) were commented out.  Do `SDA` and `SCL` just need the same treatment?  I haven't got hardware to hand at the moment, but I can try it later to see how the output looks with those pins removed.

benlye, 2018-02-27:
Making some more progress on this...

The analog pin output doesn't show up because this macro doesn't work for the Atmega1284p:
`#define IS_ANALOG(P) ((P) >= analogInputToDigitalPin(0) && ((P) <= analogInputToDigitalPin(15) || (P) <= analogInputToDigitalPin(7)))`

The analog to digital pin mapping for the 1284p is this:
PA0 (AI 0 / D31)
PA1 (AI 1 / D30)
PA2 (AI 2 / D29)
PA3 (AI 3 / D28)
PA4 (AI 4 / D27)
PA5 (AI 5 / D26)
PA6 (AI 6 / D25)
PA7 (AI 7 / D24)

And the analogInputToDigitalPin macro is this:
`#define analogInputToDigitalPin(p) ((p < 8) ? 31 - (p): -1)`

So `analogInputToDigitalPin(0)` is 31, aka the maximum pin number, not the minimum.

In the end it's not that hard to fix, by using calls to `!GET_ARRAY_IS_DIGITAL(x)` instead of `IS_ANALOG(pin)` in pinsDebug.h on [L125](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pinsDebug.h#L125) and [L154](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pinsDebug.h#L154).

Note that the call on L154 is incorrctly using `pin` instead of `x`, which doesn't work.

So now I get this:
```
PIN:  24   Port: A7 (A-7)  TEMP_0_PIN                             protected 
PIN:  25   Port: A6 (A-6)  TEMP_BED_PIN                           protected 
PIN:  26   Port: A5        Z_ENABLE_PIN                           protected 
PIN:  27   Port: A4        SERVO0_PIN                             Input  = 0  
PIN:  28   Port: A3        LCD_PINS_RS                            Output = 1  
.                          LCD_SDSS                               Output = 1
PIN:  29   Port: A2        LCD_PINS_ENABLE                        Output = 0  
PIN:  30   Port: A1 (A-1)  ADC_KEYPAD_PIN                         Analog in =  1023  
PIN:  31   Port: A0        SDSS                                   Output = 1  
```

Incidentally, this is also broken in a similar but different fashion in bugfix-1.1.x.  Happy to provide a PR for both if wanted.

thinkyhead, 2018-02-28:
Thanks for looking into it more. It's been a busy day, and I expect, a busy week ahead. I'll check it out later on and probably get it merged tonight.

Bob-the-Kuhn, 2018-02-28:
Apparently it's been a **VERY** long time since anyone has tried to use M43 with 1284 type boards.

I've got bugfix-2.0.x Platformio working with the exception of SDA & SCL.  I'll get that working, then test with Arduino and then move onto bugfix-1.1.x

thinkyhead, 2018-02-28:
I'm not surprised we're seeing this more with all the Melzi-based printers appearing lately.

Bob-the-Kuhn, 2018-02-28:
See PR #9864 for the 2.0.x changes.

I have not been able to compile it using Arduinio.

Onto bugfix-1.1.x.

Bob-the-Kuhn, 2018-03-01:
PR #9869 has the bugfix-1.1.x changes.  It compiles for both Arduino & PlatformIO.

Bob-the-Kuhn, 2018-03-01:
Finally compiled with Arduino.  Didn't change anything,  Just tried 1.9.0, 1.8.4, 1.8.5 and 1.9.0 again and it compiled.

benlye, 2018-03-01:
Nice - I'll give both a go on my 1284p board later on today.

Thanks!

benlye, 2018-03-01:
@Bob-the-Kuhn bugfix-1.1.x is perfect (builds and works for me in either Arduino IDE or VS Code) but bugfix-2.0.x still has a problem - the value of the analog pins isn't shown:
```
PIN:  29   Port: A2 (A-2)  LCD_PINS_ENABLE                        Output = 0  
PIN:  30   Port: A1 (A-1)  ADC_KEYPAD_PIN                         Input  = 1  
PIN:  31   Port: A0 (A 0)  SDSS                                   Input  = 0  
```

In pinsDebug.h, L154 is currently this:
`(!GET_ARRAY_IS_DIGITAL(pin)) {`

And it needs to be changed to:
`(!GET_ARRAY_IS_DIGITAL(x)) {`

Which makes the output this:
```
PIN:  29   Port: A2 (A-2)  LCD_PINS_ENABLE                        Output = 0  
PIN:  30   Port: A1 (A-1)  ADC_KEYPAD_PIN                         Analog in =   692  
PIN:  31   Port: A0 (A 0)  SDSS                                   Input  = 1  
```

#9874 will address it.

FWIW, there is another problem that will only affect Anet owners using the Arduino IDE - the `analogInputToDigitalPin` is incorrect in the board definition.  Hopefully that will get fixed over there (https://github.com/SkyNet3D/anet-board/issues/14).  PlatformIO has the correct macro so is fine.

Thanks,

Ben

closed 2018-03-01 by benlye

TheSFReader, 2018-03-01:
While enabling Pins debugging, I had the same compilation problem on the RAMPS Platform (mega2560), just like benlye reported. Was the same fix applied for 2560 ?

benlye, 2018-03-01:
@TheSFReader Yes, it fixed the issue for my mega2560 as well.

Bob-the-Kuhn, 2018-03-01:
I missed the **(!GET_ARRAY_IS_DIGITAL(x)) {** fix.  I'll add it to my PR.

I wonder how long it's been like that.

Bob-the-Kuhn, 2018-03-01:
Looks like I'm just too slow.  It's all been taken care off.

benlye, 2018-03-01:
I found another case where the 1284p needs a special accomodation - the `DIGITAL_PIN_TO_ANALOG_PIN` macro also needs tweaking because the current one returns negative numbers for pins other than A0.  

```
PIN:  29   Port: A2 (A-2)  LCD_PINS_ENABLE                        Output = 0  
PIN:  30   Port: A1 (A-1)  ADC_KEYPAD_PIN                         Analog in =   692  
PIN:  31   Port: A0 (A 0)  SDSS                                   Input  = 1  
```
**Should** be:
```
PIN:  29   Port: A2 (A 2)  LCD_PINS_ENABLE                        Output = 0  
PIN:  30   Port: A1 (A 1)  ADC_KEYPAD_PIN                         Analog in =   692  
PIN:  31   Port: A0 (A 0)  SDSS                                   Input  = 1  
```
It's a simple fix and I'm just checking it against my 1284 and 2560 boards before submitting the PR.