MarlinFirmware/Marlin #9825: Resuming a print - Need IDEA - all are welcome

Studiodyne, 2018-02-26:
I had an idea about POWER OUTAGE - resuming
Please STOP!! if you want to repeat that this topic is already  and already and already
Read all , and after you will talk 

My new idea 

- We add a new Gcode in slicer , for each layer change , like M1000
- We add a little code that check a pin , in this pin we add a power outage contact information
- We add ups , with autonomy just to finish a layer 

Now , if power outage ....  the pin is on ,  if the M1000 appears , the carriage go on pause park , stop the heaters ,  decrease bed temp , and store the layer in EEPROM.
If the power is back , the print resume
If the power make too much time to be back , the UPS FALLs

In the situation of UPS FALLS
- We have a perfect finished layer totally planned
- We have the last layer known

We need to resume. AND IT'S NOW I NEED HELP

- To resume the z homing , it's easy , we make a manual z homing on the object and we use babystepping to perfect it .

BUT FOR X and Y .....  I need and idea in association with babystepping X an Y , to reproduce the exact original position of X and Y before pause
-Use babystep at a very low feedrate and try to make the best ON A SHORT TIME , but print will have trace .... 
-Use ultra high end endstop ' not sure , but may be possible to have all homing totally the same '
-Use jauge comparators , mounted on x and y  , and we make a procedure before printing , we go to x0 and y0 and we memorise the jauges mesurements , in a situation of resuming , we go to X0 Y0 and with babystep we adjust to retrieve old mesurments , and we can start the rest of the print 

Thks for contributing , if i solve the X/Y babystepping , Power outage/resuming will never be a dream !

Sebastianv650, 2018-02-26:
The Prusa MK3 is showing how power outtakes are handled with perfection. If you don't know that already, check youtube.

But the problematic difference between Prusa and us is, we don't know all the printers detail where the feature will be used on. For example Z might drop to the floor as soon as power is lost. The position for Z-homing is also non-trivial, if there is a very big object there might be no usable place at all (if homing is not done to Z-max). Therefore a solution would need a lot of configurable "switches" to fit at least most of the printers out there.

My personal preference: Solution 2 is the only one which is worth creating new code as Solution 1 can be done as is today with little effort. I would rely solely on homing precision, but maybe that's because I'm no fan of babysteps at all. In my opinion (and experience), every homing operation is much more precise than any user-depending babystepping could be.

Studiodyne, 2018-02-26:
Thks 👍  i go to see this

Studiodyne, 2018-02-26:
YEAH very good in video loll , but , in application  , the printer will make me out of laugh !!!!!!
1-) Trinamic drivers ??? Pushing the limits ?? FFF need time for printing , high speed make lamination , the curse of the speed is not made for these extruders , only special high pressure system can print quickly
2-) Temp bullshit and ambiant , mdrrr  a prusa i3 is not a machine for professional mdrrr , if you want a good printer , use sla printing with resin , FFF is not a technology of beauty
3-) THE RESUME SYSTEM , MDRRRRRRRRRRRRRRRRRRRRRR the nozzle stay of the object , you will have a blob or a hole or  THE NOZZLE GLUED , when you restart , with the object that pop off , you will have an eject object of the bed , this system works  i'm sure , but with tons of options ' if ... if... if.... if... ' for me , NOT WORKING in all cases , it's a joke
4-) The dual extruder is not for me , because my printer can print Flexible , and pressure must be setted , these gears are not made for all filaments , only rigid 
5-) The sensors for runout ' VERY GREAT ' i have searched an arduino module that makes it , i thank you to have informed me , now , i will search this sensor , i will use the 'open source code ' , and i will make it for my machine , i have alreaday ' aaron tunell sensor monitor ', but this system is TOP

Loll , i'm sure this printer will enjoy all dummies , not all long date makers

Sebastianv650, 2018-02-26:
Well, I don't know which video you picked out, but you are the first person I met which laughes about the MK3.
- Trinamic: They have nothing todo with pushing some limits. The main advantage of them is a quite new current control system combined with the possibility to detect missed steps - which is a small revolution in the 3D printing marked (for the non-prof. systems as you pointed out).
- The extra temp. sensor is only a small small improvement, thats true. But it has no disadvantages and costs maybe 5cent extra. So why not doing it?
- The resume system lifts the nozzle away from the print. As far as I know it also tries to move away in X, Y, but I'm not sure about that. So your print is safe, it will not melt a hole in it or be fused with the part. On the PEI surface Prusa is using, the parts will also not come completely loose when the bed is cold, at least as long as you are not trying to print some very exotic filaments.
In the end, it's the best which can be done without a UPS, and it works.
- The MK3 has a direct drive single nozzle, not sure what you mean with dual nozzle.
- Of course it's not too hard to build a printer with a filament sensor. But their version is not a simple switch, it can so some more..

But it's up to you if you like it or not.

Studiodyne, 2018-02-26:
I've a friend that have a CREATBOT D600 with battery UPS inside and a special firmware , and all resuming have fails ... I have many experience on power outage/resuming , and it's a feature that all companies hate to integrate because of the tons of conditions and situation needed to have sucess
It's why , I work on a system , that can make the more as possible of the impossible lol

Studiodyne, 2018-02-26:
You have made the last LIN ADVANCE .... and I've respect for this job
Help to have a creative idea , that can work on all machine , all mechanics
The idea , is the baby step , but , the true idea is to make a xy babystep before starting the print to have a perfect resume

For the code , make all the need process to home on the object and more , not a problem
The true problem is to find the way to set x/y to perfection without making sad outlines on the object

thinkyhead, 2018-02-26:
> The resume system lifts the nozzle away from the print. As far as I know it also tries to move away in X, Y, but I'm not sure about that.

MK3 always re-homes XY when resuming, whether from power outage or bumping the carriage. Of course, not on filament outage.

thinkyhead, 2018-02-26:
Otherwise, as usual, **TL;DR**.

Studiodyne, 2018-02-26:
@thinkyhead  @Sebastianv650 
Look the sensor !!!!!!!!! What is this arduino module ??

Studiodyne, 2018-02-26:
@thinkyhead  as i see on the video , the carriage stay on the object , no batteries or ups , to go in pause park ,  this solution is dramatic , risk of jamming/englued , burn the object , melt the object , or make a mountain blob , that will make strong collision
MK3 have little ups to go outside of the object ?

Studiodyne, 2018-02-26:
https://steemit.com/printing3d/@boucaron/prusa-mk3-filament-detector-short-study

I2c ....

thinkyhead, 2018-02-26:
Duplicate of #2085

closed 2018-02-26 by thinkyhead