MarlinFirmware/Marlin #9826: Subsequent G29 J commands produce wrong mesh

klazzera, 2018-02-26:
### Bug Report
- Description: Subsequent "G29 J" commands when UBL is activated on the firmware configuration causes to mesh go awry

- Expected behaviour: I would expect each G29 J command to result the same adjusted mesh since the machine is physically stable between commands. I expect this command to behave more like Z homing.

- Actual behaviour: As far as i understand, each G29 J command adjusts the mesh in the ram. if another 3-point leveling was already done before one, it adjusts over the already adjusted mesh. This causes wrong meshes, failed prints and in extreme cases(repeating a few times) physical crash and stall in the machine.

- Steps to reproduce:
I first printout the current mesh using G29 T1 command:

```
SENT: G29 T1
RECEIVED: 
RECEIVED: Bed Topography Report for CSV:
RECEIVED: 
SENT: M105
RECEIVED: 0.625,0.367,0.312,0.433,0.710,0.913,1.039,1.124
RECEIVED: 0.427,0.193,0.123,0.250,0.557,0.822,0.966,0.988
RECEIVED: 0.374,0.182,0.159,0.293,0.562,0.798,0.879,0.893
RECEIVED: 0.449,0.295,0.350,0.454,0.642,0.853,0.874,0.845
RECEIVED: 0.465,0.602,0.436,0.517,0.684,0.796,0.814,0.808
RECEIVED: 0.528,0.449,0.473,0.547,0.650,0.754,0.831,0.938
RECEIVED: 0.495,0.365,0.345,0.419,0.527,0.681,0.814,0.976
RECEIVED: 0.383,0.395,0.425,0.527,0.671,0.820,0.968,1.185
RECEIVED: ok
```

Then i issue multiple G29 J's and another G29 T1 after them:

```
SENT: G29 J
SENT: G29 J
SENT: G29 J
SENT: G29 J
SENT: G29 J
SENT: G29 T1
```

after a few probing is completed and others failed:

```
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: echo:busy: processing
busy: processing
RECEIVED: ok
RECEIVED: Error:Probing failed
RECEIVED: Error:Attempt to probe off the bed.
RECEIVED: ok
RECEIVED: Error:Probing failed
RECEIVED: Error:Attempt to probe off the bed.
RECEIVED: ok
RECEIVED: 
RECEIVED: Bed Topography Report for CSV:
RECEIVED: 
RECEIVED: 15.048,13.088,12.741,13.807,16.122,17.850,18.953,19.731
RECEIVED: 13.920,12.148,11.689,12.804,15.357,17.573,18.826,19.104
RECEIVED: 13.955,12.521,12.436,13.602,15.855,17.845,18.586,18.801
RECEIVED: 15.015,13.881,14.421,15.349,16.952,18.743,19.008,18.874
RECEIVED: 15.600,16.790,15.568,16.309,17.749,18.740,18.980,19.034
RECEIVED: 16.560,16.025,16.316,17.006,17.934,18.862,19.578,20.531
RECEIVED: 16.757,15.810,15.751,16.441,17.407,18.735,19.900,21.291
RECEIVED: 16.317,16.508,16.848,17.764,19.017,20.307,21.585,23.426
RECEIVED: ok
```

The actual problem with that behaviour is that if i include "G29 J" in the startup of my g-code, printing parts one after another causes problems.

I need to run this command before each print because i have dual Z motors which go out of sync when steppers aren't powered because i use ball screws which have much less friction than metric threaded rods. And i also have only one Z max switch and one Z min probe, i can't really use dual Z sync which requires dual Z end stops. Theoretically syncing Z motors with one single probe should be possible with probing the bed from two extremes of the axis between the Z screws(X for my machine), but i couldn't find such a feature in Marlin. And mesh probing before each print takes a really long time.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1758920/Marlin.zip)

klazzera, 2018-02-26:
i guess i have 1.1.8 according to the readme.md file in the firmware folder

gloomyandy, 2018-02-26:
Hi, I saw similar problems. This is not an official fix, but it has worked for me. Try changing the following line in file ubl_G29.cpp at line 911 change from:
`
      z_values[i][j] += z_tmp - d;
`

to
`
          //z_values[i][j] += z_tmp - d;
          z_values[i][j] = z_tmp + d;
`

Please post to say if the above improves things for you.

thinkyhead, 2018-02-26:
I believe this is already patched in the bugfix branches. So, here's the canned blurb…

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

gloomyandy, 2018-02-26:
I've tested the bugfix-1.1.x and I think the problem still remains. However I think that the solution is slightly different. In bugfix-1.1.x changing ubl_G29.cpp line 1676 from:
  z_values[i][j] += z_tmp - lsf_results.D;
to
 z_values[i][j] = z_tmp - lsf_results.D;

gives results for me that are stable across multiple G29 J operations. I have been intending to post this information for a few days, but unfortunately other things got in the way. It would be good to have someone confirm my findings. I should say I have only tested the G29 J 3 point probing, not any of the other cases that use this common code.

Oh and I think that there is also a small bug in the diagnostic code for the new 3 point G29 J code. The setting of the debug variable z1 is not consistent with the way that z2 and z3 are set (z1 is being set to the probed value after correction while z2 and z3 are set to the original probe values. This means that the diagnostic checks for the bed mesh correction code produce incorrect output for the first probed point.

thinkyhead, 2018-02-26:
This is immensely helpful. You can stop being gloomy now!
We're one huge step closer to being able to release 1.1.9.

closed 2018-02-26 by thinkyhead

klazzera, 2018-03-09:
Thanks gloomyandy, your fix works! but only kinda... the code change makes sense since each operation seem to be adding on top of previous one in the original code. But now, the bed values converge into a correct mesh matrix in a few iterations, like if original mesh data from G29 P1 full probe operation was incorrect. This happens even if i issue the G29 J right after G29 P1, not giving a chance to the motors to go powered off and cause a tilt. After the G29 J operation i'm able to print with a decent leveling but still this behaviour is weird.

i'll post another issue after i properly define this issue with enough information.

commit ddfddae4 (2018-02-26): Fix subsequent `G29 J` commands

Fix #9826. Solution by @gloomyandy
commit 7bdb98c9 (2018-02-26): Fix #9826. Solution by @gloomyandy