MarlinFirmware/Marlin #9831: [2.0.x] Proposed workaround for STM32 enum pins

thinkyhead, 2018-02-26:
**Problem:** Because pins are defined for the STM32F1 platform with an `enum` the preprocessor can't see them. This breaks some of our indirection and causes compilation to fail.

**Solution:** Define the pin numbers in the HAL in the usual manner but with an underscore prefix. When we need to refer to the integer value of (_e.g._) `PA0` inside of a preprocessor `#if` expression such as `#if BEEPER_PIN == PA0`, wrap the pin name in a conversion macro.

### Proof of concept
```cpp
#include <iostream>
#include <string>
#include <math.h>

using namespace std;

#define _CAT(a, ...) a ## __VA_ARGS__
#define _PIN(P) _CAT(_, P)
#define PIN_EXISTS(P) (defined(P ##_PIN) && P ##_PIN >= 0)

enum { PA0, PA1, PA2, PA3, PA4 }; // constexpr also works fine

#define _PA0 0
#define _PA1 1
#define _PA2 2
#define _PA3 3
#define _PA4 4

#define HEATED_BED_PIN PA3

int main(int argc, char const *argv[]) {

  #if PIN_EXISTS(HEATED_BED)
    cout << "HEATED_BED_PIN=" << int(_PIN(HEATED_BED_PIN)) << endl;
  #else
    cout << "HEATED_BED_PIN undefined" << endl;
  #endif

  return 0;
}
```
CC: @MarlinFirmware/32bit-maintainers

victorpv, 2018-03-02:
In that case, we could just use the pin names with the underscore in Marlin, and we only need to add the list of pins in the HAL, i.e.
```
...
#define HEATED_BED_PIN _PA3

int main(int argc, char const *argv[]) {

  #if PIN_EXISTS(HEATED_BED)
    cout << "HEATED_BED_PIN=" << HEATED_BED_PIN << endl;
  #else
    cout << "HEATED_BED_PIN undefined" << endl;
  #endif

  return 0;
}
```

victorpv, 2018-03-02:
Either solution would work for most boards, except for the Maple ones, since those don't number the pins port sequency (historical reasons due to the way leaflabs did)

thinkyhead, 2018-03-05:
Well, we've incorporated the big blob o' defines and that solved the pins part of the equation. My proposed solution is now redundant. If there's a difference for Maple, we'll just have to handle it as a special case.

closed 2018-03-05 by thinkyhead