MarlinFirmware/Marlin #9833: [bugfix_2.0.x] Radds display back button inop

bwgeez02, 2018-02-27:
Back button doesn't work on Radds lcd. Pin #define BTN_BACK        71.

In all menus when back button is pressed the lcd should return to the previous menu.

Nothing happens when the back button is pressed. Not even the audible beep like the other buttons. In order to return to the previous menu the previous menu must be selected with the encoder and the encoder then pressed. 

Not a huge deal, but a small annoyance after getting used to using the back button.

thinkyhead, 2018-02-27:
In what fork of Marlin did you find that it worked?

bwgeez02, 2018-02-27:
I've been using Wurstnase's Marlin4due.

thinkyhead, 2018-02-27:
Thanks for pointing that way. We've just started adding 32-bit support and hadn't implemented the `BACK_BTN` yet. It will be merged into the project shortly.

closed 2018-02-27 by thinkyhead

AnHardt, 2018-02-27:
Marlins menu system was designed for using as few resources as possible. About the most valuable resources are pins. The menu system was made for requiring 3 inputs (Up, Down, Select). This decision required the 'back menu items'.

Seen from that viewpoint the RADDS panel has a additional useless back button, like some other 4/5 button panels.

To keep the concepts clean, we'd have to drop the 'back menu items', when we have a 'back button'.

thinkyhead, 2018-02-27:
> To keep the concepts clean, we'd have to drop the 'back menu items', when we have a 'back button'.

That sounds like a reasonable thing to do, and would recover some flash space. But I wonder if some people prefer to ignore that button and just use the encoder. My feeling is that sometimes they use the encoder button and sometimes use the back button, depending on how far down in the menu they are and how impatient they are. Marlin4Due implements it as a simple "back" but of course it could do anything.
- Return to Status
- Babystepping
- Move Axis (Z)
- Pause Print
- …