MarlinFirmware/Marlin #9839: [FR] Sensorless Homing for Core XY (XZ and YZ)

HendrikSP, 2018-02-27:
Hi,

This is my first post here, so let me thank you all for the effort you put in Marlin!

I have build a Core XY printer that uses TMC2130 Stepper Drivers. I already got sensorless homing working on X and Y with small modifications to the firmware. My feature request applies for Core XZ and YZ as well, but I will describe it below for XY only to keep this focused.

The main problem is that on Core XY printer you need to set both motors (X and Y aka A and B) up for the X or Y homing move.

I did that on 1.1.8 and noticed that the TMC2130 code has a bit changed in the bugfix 1.1. branch and that sensorless homing for XZ and YZ is now supported as well.

My simple code change for 1.1.8 is quite simple:

```cpp
   #if ENABLED(SENSORLESS_HOMING)
     #if ENABLED(X_IS_TMC2130)
+      #if ENABLED(COREXY) && ENABLED(Y_IS_TMC2130)
+      if (axis == X_AXIS || axis == Y_AXIS) {
+      #else
       if (axis == X_AXIS) {
+      #endif
         currentX = stepperX.getCurrent();
         tmc_sensorless_homing(stepperX);
         stepperX.setCurrent(X_HOMING_CURRENT, R_SENSE, HOLD_MULTIPLIER);
       }
     #endif
```

So the Y motor is setup for X homing as well. I did a similar change for the Y homing and X motor and after homing the reconfiguration to "normal" operation of both motors in the same fashion.

However when I thought about to adapt it for the bugfix branch and create a PR, I noticed that the code will become ugly quickly since I would have to consider XZ and YZ as well.  So before writing/changing code that will be hard to read and mostly untested (I have neither a XZ nor a YZ printer), I would like to ask for opinions/suggests on that.

thinkyhead, 2018-02-27:
Well, since you have it working for XY and can't test other configurations without some clever stepper and endstop swapping, you can leave it up to me to adapt it to the others based on your example. It shouldn't be too hard. It's basically the same principle but adapted in a more general way. I think it will make an interesting challenge. I'll let you know when there's something to test.

HendrikSP, 2018-02-27:
Thank you.

The point is that I'm completely ignorant of the implications on Z (levelling, homing) on a CoreXZ/CoreYZ printer. I not sure if even the sensorless Z homing makes sense on such a configuration: Will the Z distance constant enough after each homing?

tcm0116, 2018-02-27:
>Will the Z distance constant enough after each homing?

In my testing, I've found that it's pretty consistent. The only issue is that you can end up with a bit of an offset due to the fact the the stallGuard has an accuracy of one full step. On my Delta machine, with 200 steps/revolution motors and a 100 steps/mm resolution, I see about a 1mm offset between the reported home position and the actual home position. However, this offset seems to be consistent, so it shouldn't be much of an issue.

thinkyhead, 2018-03-01:
The two PRs above should do the trick. They add a generalized function to enable sensorless homing for an axis (if it has it), taking account of Core / H-bot Kinematics, and makes sure that sensorless homing is configured the same way for both combined axes.

HendrikSP, 2018-03-01:
Thanks a lot, I reviewed (with my still limited overall knowledge on the Marlin code base) both PRs and left some comments.

HendrikSP, 2018-03-01:
I've seen you merged the PRs 👍 . I will test 1.1.x on my printer and close this issue if I find no problems, ok?

thinkyhead, 2018-03-01:
Thanks for catching the typos. I threw it together on my way out the door and who knows if I would have caught them when I got home. Thanks for prompting this change. I think it's a good general improvement and makes the code a little cleaner on top of making it more compatible.

HendrikSP, 2018-03-01:
I have used the change on my Core XY. It works. Thanks again. Closing.

closed 2018-03-01 by HendrikSP