MarlinFirmware/Marlin #9845: 3d touch probe geeetech help

vMeph, 2018-02-27:
i have seen a couple issues about 3d touch probe geeetech chinese
but so far i cant seem to figuer out what im doing wrong
im using ramps 1.4
i have the 3d touch 3 wire conected to the servo 0 pins ,+5v, gnd, pin11 and the 2 wire conected to the zmin pins
what im experience is if i move by hand the proble the light will turn on and will turn off, but if i use the comand M119 to see pins states it says is alway open..
the probe  conections side if i remove the 3 wire plug the probe will snap,(showing me its alive)
but i just cant get it to work the probe is always lose like there is no strenght
im not using external pullups on the wiring

here is my configuration file:
https://hastebin.com/azojupepum.cpp

if someone could help me out figuer out whats wrong would be awsome

Bob-the-Kuhn, 2018-02-27:
It may be working correctly.  

Your configuration looks good.

M119 should always return open unless you're in the BLTouch test mode.

There's a servo probe test for the BLTouch type of probes.  To run it do the following:
1. Enable **pins_debugging** in **Configuration_adv.h**
2. Enter the command **M43 S** on the host.
3. Follow the prompts on the host screen.

vMeph, 2018-02-28:
Thanks @Bob-the-Kuhn  the results shows there is no trigger evean knowing lights turn on and off, at this point no idea whats going on, start to wonder if is a material defect grrr

Bob-the-Kuhn, 2018-02-28:
Double check your endstop cable and which endstop it's plugged into.  Maybe you have the ground and signal wires swapped.

Studiodyne, 2018-03-02:
Place a wire on the dual wire ' contact ' of 3d touch , make a z homing , when you contact the wires , the z stop , then , the sensor is down if not , it's circuit 
If it's circuit , verify presence of 5v ' pull up resistor '  , if not , verify your pulling setting for the pin used , an cables

vMeph, 2018-03-04:
got it to work, it resolve the problem by using a external 5v to power the probe and it seems to have fixed

thanks for help

closed 2018-03-04 by vMeph